Koenkk/zigbee2mqtt #13929: Aubess Mini Smart Switch Zigbee 3.0, "Unsupported", not working

topreece, 2022-09-09:
### What happened?

Aubess Mini Smart Switch Zigbee 3.0 shows as "Unsupported" and is not interviewing correctly hence not working.
Im using a SONOFF ZB Dongle -E coordinator (i have like 20 sensors working without a hitch with it)

Link: https://www.aliexpress.com/item/1005003637851837.html?spm=a2g0o.order_list.0.0.4dda1802oyL3ti

The pictures display a WHD02 as selling unit, but when you recieve the item it is not labeled as WHD02, only a generic name "Mini Smart Switch" (i don't know if this is a "cheaper" alternative to a WHD02 -which is supported by zigbee2mqtt- or there is something different in the firmware)
[image]
[image]
[image]
[image]
[image]

### What did you expect to happen?

I expect to connect and expose his attributes as intended

### How to reproduce it (minimal and precise)

Buy one and try to connect it to zigbee2mqtt 

### Zigbee2MQTT version

 1.27.2-1

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF ZB Dongle -E

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9531811/log.txt)

Here you can check i tried to join and delete the device several times without success.

Johan219, 2022-09-27:
I am having the same problem, hope the device can be supported

Caillema, 2022-10-21:
This problem is still exist.
This device is not recognized by Zigbee2MQTT (Conbee 2 dongle, same problem than described above). 
Can it be added please ?

Koenkk, 2022-10-21:
Can you provide the data/database.db entry of this device?

Caillema, 2022-10-21:
Sorry I'm a newbie. How should I found that ?
Thanks a lot for your support.

Caillema, 2022-10-21:
Ok. I think you need that (extract of my database.db) :
{"id":24,"type":"Router","ieeeAddr":"0xa4c138331a1497de","nwkAddr":49351,"manufId":4417,"manufName":"_TZ3000_xkap8wtb","powerSource":"Mains (single phase)","modelId":"TS000F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,10,57344,1794,2820,4096,6280],"outClusterList":[25],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"32768":0,"tuyaBacklightMode":1,"onOff":0,"moesStartUpOnOff":2,"onTime":0,"offWaitTime":0}},"genBasic":{"attributes":{"modelId":"TS000F","manufacturerName":"_TZ3000_xkap8wtb","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":241,"rmsCurrent":78,"activePower":0}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}},"manuSpecificTuya_3":{"attributes":{"switchType":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1666352032109,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-10-22:
`_TZ3000_xkap8wtb` will be supported in the next release (1 November) and is already supported in the dev branch (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Caillema, 2022-10-24:
Perfect, thank you so much for your update and your reactivity !
I will wait and test it with the next release in November.

callacomp, 2022-10-25:
Just got one of these also, Aubess again but only one switch point (Sw) on the right (two Neutral Nin instead). In the meantime, you can use this as an external converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS000F', manufacturerName: '_TZ3000_xkap8wtb'}],
    model: 'WHD02',
    vendor: 'TuYa',
    whiteLabel: [{vendor: 'TuYa', model: 'iHSW02'}],
    description: 'Wall switch module',
    toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior, tz.tuya_switch_type]),
    fromZigbee: extend.switch().fromZigbee.concat([fz.moes_power_on_behavior, fz.tuya_switch_type]),
    exposes: extend.switch().exposes.concat([e.power_on_behavior(), e.switch_type_2()]),
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);},
};

module.exports = definition;
```

Caillema, 2022-11-02:
> Just got one of these also, Aubess again but only one switch point (Sw) on the right (two Neutral Nin instead). In the meantime, you can use this as an external converter:

Perfect, device supported like a WDH02. Switch_type, Power_on_behavior and Linkquality work.
But this device could also provide the entities for Power, Current, Voltage and Energy.
How can I catch them?

Koenkk, 2022-11-02:
@Caillema does the power monitoring work with the following external converter?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS000F', ['_TZ3000_xkap8wtb']),
    model: 'TS000F_power',
    description: 'Switch with power monitoring',
    vendor: 'TuYa',
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.ignore_basic_report],
    toZigbee: [tz.on_off],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'haElectricalMeasurement', 'seMetering']);
        await reporting.rmsVoltage(endpoint, {change: 5});
        await reporting.rmsCurrent(endpoint, {change: 50});
        await reporting.activePower(endpoint, {change: 10});
        await reporting.currentSummDelivered(endpoint);
        endpoint.saveClusterAttributeKeyValue('haElectricalMeasurement', {acCurrentDivisor: 1000, acCurrentMultiplier: 1});
        endpoint.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
        device.save();
    },
    exposes: [e.switch(), e.power(), e.current(), e.voltage().withAccess(ea.STATE), e.energy()],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

Caillema, 2022-11-03:
> @Caillema does the power monitoring work with the following external converter?
>
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = [...]

z2m do not restart after these modifications. Perhaps comes syntaxe in the configuration.yaml, where I am that mistakes :

```
duplicated mapping key (4:1)

 1 | external_converters:
 2 |   - ext_converter.js
 3 | 
 4 | external_converters: []
-----^
 5 | devices:
 6 |   - devices.yaml
```

sdalee, 2022-11-03:
I tried the external converter above. Now the power and energy meter is displayed. The power might be a good value (load on LED light), the current is not good because it shows 83 amps continuously(even if I turn the switch off) and the switch setting options are gone (like 'power_outage_memory'), only the switch on/off is left.

Koenkk, 2022-11-03:
@sdalee I think these device doesn't support current. Can you provide the data/database.db entry of your device?

Caillema, 2022-11-03:
> I tried the external converter above. Now the voltage, current, power and energy meter is displayed. The power and voltage might be a good value (load is a LED light), the current is not good because it shows 83 amps continuously (even if I turn the switch off) and the switch setting options are gone (like 'power_outage_memory'), only the switch on/off is left.

Confirmed.
[image]

Caillema, 2022-11-03:
> @sdalee I think these device doesn't support current. Can you provide the data/database.db entry of your device?

Devices #24 & #26 : 
```
{"id":24,"type":"Router","ieeeAddr":"0xa4c138331a1497de","nwkAddr":49351,"manufId":4417,"manufName":"_TZ3000_xkap8wtb","powerSource":"Mains (single phase)","modelId":"TS000F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,10,57344,1794,2820,4096,6280],"outClusterList":[25],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"32768":0,"tuyaBacklightMode":1,"onOff":0,"moesStartUpOnOff":1,"onTime":0,"offWaitTime":0}},"genBasic":{"attributes":{"modelId":"TS000F","manufacturerName":"_TZ3000_xkap8wtb","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":90,"rmsCurrent":95,"activePower":0}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}},"manuSpecificTuya_3":{"attributes":{"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff05f8c2","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1667375592894,"defaultSendRequestWhen":"immediate"}
{"id":26,"type":"Router","ieeeAddr":"0xa4c138825b731585","nwkAddr":62575,"manufId":4417,"manufName":"_TZ3000_xkap8wtb","powerSource":"Mains (single phase)","modelId":"TS000F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,10,57344,1794,2820,4096,6280],"outClusterList":[25],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"tuyaBacklightMode":1,"moesStartUpOnOff":1,"onTime":0,"offWaitTime":0}},"genBasic":{"attributes":{"modelId":"TS000F","manufacturerName":"_TZ3000_xkap8wtb","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":240,"rmsCurrent":91,"activePower":0}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}},"manuSpecificTuya_3":{"attributes":{"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff05f8c2","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1667501438384,"defaultSendRequestWhen":"immediate"}
```

Koenkk, 2022-11-04:
Great, added the device!

Changes will be available in the dev branch tomorrow. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-04 by Koenkk

sdalee, 2022-11-05:
Hi, 

I installed the new Edge some minutes ago , but the part of power and energy measurement is missing.
I see now, that this topic and my original topic include two different Devices.  The device of this topic is a single switch, while my topic is a switch with power monitoring [1455](https://github.com/Koenkk/zigbee2mqtt/issues/14555). 

The db row of switch with monitoring :
{"id":107,"type":"Router","ieeeAddr":"0xa4c138e469f1a381","nwkAddr":13553,"manufId":4417,"manufName":"_TZ3000_xkap8wtb","powerSource":"Mains (single phase)","modelId":"TS000F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,10,57344,1794,2820,4096,6280],"outClusterList":[25],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genOnOff":{"attributes":{"32768":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"onOff":0,"onTime":0,"offWaitTime":0}},"haElectricalMeasurement":{"attributes":{"rmsVoltage":232,"rmsCurrent":66,"activePower":0}},"genBasic":{"attributes":{"modelId":"TS000F","manufacturerName":"_TZ3000_xkap8wtb","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"seMetering":{"attributes":{"currentSummDelivered":[0,0]}},"manuSpecificTuya_3":{"attributes":{"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00216004eb","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1667665508400,"defaultSendRequestWhen":"immediate"}

Koenkk, 2022-11-06:
@sdalee but it does work with https://github.com/Koenkk/zigbee2mqtt/issues/13929#issuecomment-1301081046 ?

sdalee, 2022-11-06:
> @sdalee but it does work with [#13929 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/13929#issuecomment-1301081046) ?

Well, If I use external converter, I will get the result in [https://github.com/Koenkk/zigbee2mqtt/issues/13929#issuecomment-1302558440](comment). (current and energy wrong, 'power_on_behavior' and 'swithc_type'  don't appear).

If I use zigbee2MQTT add-on, i will get the following result (after permit join):
[image]
I've tried the 'power on behavior'. It works. But as you see the power monitoring functions don't appear. 

If I use zigbee2MQTT Edge add-on, i will get the following result (after permit join):
[image]
In this case, the energy doesn't work and the 'power_on_behavior' and 'swithc_type'  don't appear.

sdalee, 2022-11-07:
Thanks, Koenkk, it works perfectly, only the energy meter doesn't work, but I think that's the fault of the device.

sdalee, 2022-11-08:
@Koenkk: As you can see [https://github.com/Koenkk/zigbee2mqtt/issues/13929#issuecomment-1304917165](my screenshot) : energy: Null kwh

Koenkk, 2022-11-09:
@sdalee does it provide a value when going to z2m frontend -> device -> dev console -> fill in like below (endpoint may differ) -> read?

[image]

Koenkk, 2022-11-09:
Great, it supports energy. Can you provide the debug log when starting z2m + 5 minutes with the following external converter? https://gist.github.com/Koenkk/550c7c2b486173a8c019bc8ba764fad8

sdalee, 2022-11-09:
How can I share the relatively large log? Should I copy it into this thread?

Domotixz, 2022-11-10:
How did you connect them to zigbee2mqtt. I have 5 of this exact same switches but z2m doesn't seem to recognize them.

Domotixz, 2022-11-10:
I have bought 5 of these switches. When adding them to z2m (not dev) i get a device with only 1 switch and no power metering. Is that expected behavior?

Koenkk, 2022-11-10:
@sdalee can you confirm that the converter is loaded correctly? I don't see the logging. The model under About should be `TS000F_power_TEMP`

sdalee, 2022-11-10:
@Koenkk Yesterday I add the converter in Z2M GUI -> Extensions -> + -> Save. Today I add the extension in the configuration.yaml and now the is correct: 
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/9983849/z2m.log)

I don't know why the extension worked yesterday.

@Domotixz: Koenkk is working on it now.

Koenkk, 2022-11-11:
It shows errors I see, I updated https://gist.github.com/Koenkk/550c7c2b486173a8c019bc8ba764fad8 to fix it.

sdalee, 2022-11-11:
Unfortunately, energy metering is still not working::
[image]

New Log:

 
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/9992262/z2m.log)

Domotixz, 2022-11-11:
[image]

i only get this. No power or energy metering.

Koenkk, 2022-11-12:
@sdalee 

- are you sure the converter is loaded correctly?
- Can you provide me the debug log starting from the point where z2m is started with the following converter: https://gist.github.com/Koenkk/550c7c2b486173a8c019bc8ba764fad8? (the current logs start while it is already running). 

@Domotixz without an external converter that is expected, we are working on it.

sdalee, 2022-11-12:
@Koenkk 

The whole log: 
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/9995704/z2m.log)

My configuration.yaml file first 2 raw:
external_converters: 
  - extension.js

Certainly, I copied your converter into extension.js in zigbee2mqtt directory, next to the configuration.yaml

Koenkk, 2022-11-13:
did you restart z2m after copying the converter? In your log I see: 

```
info  2022-11-12 19:45:14: Device '0xa4c138e469f1a381' is supported, identified as: TuYa Switch with power monitoring (TS000F_power_TEMP)
```

But this should be `TS000F_power_TEMP_1` (https://gist.github.com/Koenkk/550c7c2b486173a8c019bc8ba764fad8#file-ts000f_power-js-L47)

sdalee, 2022-11-13:
Yes, of course (more than once), but when I will arrive home, I will restart the host.

sdalee, 2022-11-13:
If I see well, now the right extension is loading:
 2022-11-13 18:40:10: Device '0xa4c138e469f1a381' is supported, identified as: TuYa Switch with power monitoring (TS000F_power_TEMP_1)

The result is same: 
energy
Sum of consumed energy:   Null kWh
Log: [z2m_TS000F_power_TEMP_1.log](https://github.com/Koenkk/zigbee2mqtt/files/9997898/z2m_TS000F_power_TEMP_1.log)

Log:

sdalee, 2022-11-13:
Now (18:53) it can see: N/A kWh

Koenkk, 2022-11-14:
I see an interesting error in the log:

```
error 2022-11-13 18:40:24: Failed to configure '0xa4c138e469f1a381', attempt 2 (Error: ConfigureReporting 0xa4c138e469f1a381/1 haElectricalMeasurement([{"attribute":"activePower","minimumReportInterval":5,"maximumReportInterval":3600,"reportableChange":10}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'INVALID_DATA_TYPE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.configureReporting (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:694:22)
    at Object.activePower (/app/node_modules/zigbee-herdsman-converters/lib/reporting.js:203:9)
    at Object.configure (/app/dist/util/externally-loaded.js:58:9)
    at Configure.configure (/app/lib/extension/configure.ts:115:13))
```

Can you provide the logging again with the updated https://gist.github.com/Koenkk/550c7c2b486173a8c019bc8ba764fad8 ?

sdalee, 2022-11-14:
With the new converter the result:
[image]

The log: 
[temp2.log](https://github.com/Koenkk/zigbee2mqtt/files/10007004/temp2.log)

sdalee, 2022-11-14:
Enery appeared:

[image]

Domotixz, 2022-11-15:
[image]

Switching to dev branche gives me problems with IKEA motion detector. Can this be related to the recent changes?

Koenkk, 2022-11-15:
Awesome! Let me know if it keeps working.

Domotixz, 2022-11-15:
How can i get this to work too? Do i need to remove them and then add them again? Or should i wait till it will be implemented in the master branch? What do you suggest?

Domotixz, 2022-11-15:
Yes, i did that but to no avail.

sdalee, 2022-11-15:
@Domotixz 
First force removes the switch from z2m. Then do Koenkk wrote two comments earlier. (certainly the configuration.yaml of z2m, not configuration.yaml of home assistant). Then restart z2m and rejoin the switch.

First force removes the switch from the z2m. Then do what Koenkk wrote two comments earlier (of course you have to use z2m configuration.yaml, not home assistant configuration.yaml). Then restart z2m and reconnect the switch.

Domotixz, 2022-11-15:
I did exactly that but it still is paired as the 'standard'  device without the energy metering. I'm not using it with HA but with domoticz. But for now i'm testing it in the dev branch (commit e66fe853). I removed it, (forced) and then added it again. But no energy metering.

Domotixz, 2022-11-15:
The manufacturer code is _TZ3000_46t1rvdu. I think that is the problem. Right?

Domotixz, 2022-11-15:
To answer my own question. The switch is defined as Wall Switch Module (TS0001) in the tuya.js file and not as a TS0001_power module. I guess it just doesn't have the feature to measure the power. Unfortunately. However. Thanks all for your replies and reactions.

sdalee, 2022-11-16:
Well, I've tested this aubess switch for 25 hours. The test load was a media wall (Router + Set Top Box + TV + Xbox X) The reference meter was a Blitzwolf smart socket with power metering.
The result: the Blitzwol measured 2.06 kWh and the Aubess 1.98 kWh. The difference is 4%.

@Koenkk: Thank you very much for the long process of helping us.

It's a bit off: I'm an electrical engineer, not a programmer. I like programming (as an amateur) and have done some java scripts in the Node-Red function node (Node-Red next to Hassio). Unfortunately, the description "support new device" on the Z2M website is too short for me.  Can you recommend me some good videos or some good descriptions on the web how to add a new device to Z2M?

Koenkk, 2022-11-17:
@sdalee to confirm, with the latest version of https://gist.github.com/Koenkk/550c7c2b486173a8c019bc8ba764fad8 , everything is working and all electrical measurement values update accordingly?

sdalee, 2022-11-17:
Yes, it's worked well for two days.
Reference Blitzwolf measured 3.62 kWh and the Aubess 3,5 kWh. The difference is 3,3%.
A screenshot:
[image]

Caillema, 2022-11-18:
> @sdalee to confirm, with the latest version of https://gist.github.com/Koenkk/550c7c2b486173a8c019bc8ba764fad8 , everything is working and all electrical measurement values update accordingly?

This update on dev branch (edge), will be integrated in the next release (December?) of Zigbee2MQQT?

Koenkk, 2022-11-20:
> will be integrated in the next release (December?) of Zigbee2MQTT?

Yes!

sdalee, 2022-11-21:
@Koenkk Thank you for your help!

sdalee, 2022-11-28:
Well,
The measured energy consumption has been 20 kWh. The difference between the two power meters is ~3%,  so I think it's perfect. 
I have only 1 problem.  The ZigBee communication is too frequent (see log 'zigbee2mqtt/Prb_Aubess_EnergyMeterSwitch'). Can we set something to make communication less frequent?
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/10106951/log.txt)

HENN0, 2022-12-05:
I got a bit differen Aubess switch, it should do power measurements but its not shown.
 2022-12-05 18:31:25: MQTT publish: topic 'zigbee2mqtt/0xa4c1383b0d3d5958', payload '{"device":{"applicationVersion":70,"dateCode":"","friendlyName":"0xa4c1383b0d3d5958","hardwareVersion":1,"ieeeAddr":"0xa4c1383b0d3d5958","manufacturerID":4417,"manufacturerName":"_TZ3000_46t1rvdu","model":"WHD02","networkAddress":53306,"powerSource":"Mains (single phase)","stackVersion":0,"type":"Router","zclVersion":3},"last_seen":"2022-12-05T18:31:25+01:00","linkquality":60,"power_on_behavior":"previous","state":"ON","switch_type":"toggle"}'

I am on dev branch and did update today.
With the external convertes z2m dosnt start anymore.

elmocito, 2023-02-14:
Hey @Koenkk,

I have a similar problem with this device:
https://de.aliexpress.com/item/1005005122047276.html?spm=a2g0o.order_list.order_list_main.5.1c615c5fpOe6hE&gatewayAdapt=glo2deu

This is the result:
[image]

Shall I create a separate ticket?

Koenkk, 2023-02-14:
@elmocito you are probably not using the latest z2m version, try z2m 1.30.1.

jmartincsek, 2023-02-21:
Hi,

I have also bought a couple of this device. I try to follow the instructions, but no luck. I don't see anything it in the logs. May I ask for some suggestion what is the process to add this device to zigbee2mqtt,

Thank you for your help

Regards

Janos

jmartincsek, 2023-02-21:
Sorry for not adding enough information:
I have the Aubess Mini Smart Switch Zigbee 3.0
https://www.aliexpress.com/item/1005003637851837.html?spm=a2g0o.order_list.0.0.4dda1802oyL3ti

I created the ext_converter.js file and I have have added  the lines  
external_converters:
  - ext_converter.js
to configuration.yaml:

thanks

regards

Janos

[2 more replies]