Koenkk/zigbee2mqtt #13932: Failed join with QBKG24LM. Failed to interview '...', device has not successfully been paired

shootnicks, 2022-09-09:
### What happened?

Hi guys.
My old "Texas Instruments CC2531" died and I bought a new "SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P". Performed a new Z2M installation and connected the devices. All devices connected correctly, but QBKG24LM (I have 2 of them, and the same problem with both) is not. It appears in the list of devices, but writes that "Unsupported".
I thought the problem was "SON OFF Zigbee 3.0 USB Dongle Plus ZBDongle-P". Therefore, I updated the firmware. it didn't help.
I tried to connect QBKG24LM to Xiaomi Gateway with ZiGate, but it's the same there: Failed to interview '...', device has not successfully been paired

### What did you expect to happen?

Device '...' is supported, identified as: Aqara D1 2 gang smart wall switch (with neutral wire) (QBKG24LM)

### How to reproduce it (minimal and precise)

Try to connect QBKG24LM.

### Zigbee2MQTT version

1.22.0 and 1.27.2

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9532651/log.txt)

shootnicks, 2022-09-13:
Hello everyone. Can someone take a look at my problem!? Please! :)

Koenkk, 2022-09-14:
Can you provide the herdsman debug log when pairing the device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

shootnicks, 2022-09-14:
@Koenkk, yes, of course.

I hope this is what you asked for. According to your link, it is not clear what needed to be done. :)
[STDOUT.txt](https://github.com/Koenkk/zigbee2mqtt/files/9566001/STDOUT.txt)

shootnicks, 2022-09-14:
Device: 0x00158d00028c212a

Koenkk, 2022-09-15:
It seems the device does not connect at all, can you try pairing the device close to the coordinator?

shootnicks, 2022-09-15:
I tried to do so.  Placed the coordinator 50 cm from the device.  Both QBKG24LM behave this way.  Even though they worked great before.  I've tried unplugging devices for a few hours to reset them.
I have no more ideas what to try on my end.

Koenkk, 2022-09-16:
Can you try to reflash your coordinator and try again?

YuriyPrivezentsev, 2022-10-23:
Hi, I have pretty much the same problem: both of the devices are not connected. The interview starts and never ends.