Koenkk/zigbee2mqtt #13944: TS130F_dual - state unknown and invert_cover functionality reversed as compared to documentation

convicte, 2022-09-09:
### What happened?

Half of the issue is mostly identical to what has been worked on in the PR (below) for the single cover model of TS130F.
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4609

1) The covers _left and _right fail to report state:
[image]

2) The cover lists state OPEN when in the fully down position and CLOSED when fully opened UP.

2) Second part of the issue pertains to ```invert_cover``` which when set to false behaves as it should for true and vice versa—described in detail here and below:  https://github.com/Koenkk/zigbee-herdsman-converters/pull/4609#issuecomment-1241742933

In conclusion, there are 3 issues:

TS130F_dual - fails to report the state for each cover entity in HA
TS130F_dual has the invert_cover functionality switched compared to default (false gives the results described in the docs for true and vice versa). [This](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4609) PR fixes it for the TS130F, will it also do it for TS130F_dual?
TS130F_dual has the OPEN/CLOSED issue happening for it as well, which was fixed by [this ](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4609)PR for TS130F , so I wonder if it will also fix it for TS130F_dual

### What did you expect to happen?

1) Cover _left and _right reporting state open/closed in HA for the cover entity
2) The cover to give correct state (open/close) when fully open or closed - identical to this [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4609)
3)  When ```cover_invert: false``` 
setting position 100 fully opens the cover
settings position 0 fully closes the cover
Currently, it's the opposite

The last time no.1 and no.2 worked was in version 1.26.0.
Issue no.3 never worked correctly for me.

### How to reproduce it (minimal and precise)

https://github.com/Koenkk/zigbee-herdsman-converters/pull/4609#issuecomment-1241872741
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4609#issuecomment-1241923780
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4609#issuecomment-1241941492

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

 20211217

### Adapter

Egony Stick V4 - CC2652 variant

Koenkk, 2022-09-10:
Pushed the `coverInverted` fix, can you check if it fixes this issue?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

convicte, 2022-09-11:
I just realized you added another PR in this area specifically!

Sorry for the confusion—I am still pretty new to Git and all the messages and code tracking.

I'll try to test the DEV branch and report back.
If there is a 'step by step' somewhere, for people transitioning between stable and dev branches, I would love a link.

Sorry again for being high maintenance!!

convicte, 2022-09-11:
@Koenkk, sorry to report, but the changes available in EDGE only corrected issue no.3

Problems 1 and 2 persist:
[image]

In addition, the issues for TS130F (since shutter module) also were not resolved with the above-mentioned PR.

[image]
```cover_inverse: false```, shutter fully closed, ```motor_reversal: OFF```, so the state should be closed.
I'll report it also in the PR for consistency.

Koenkk, 2022-09-12:
I see now, found the issue, fix is available in the latest dev!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

convicte, 2022-09-12:
I'll uninstall and reinstall DEV and see what's what.

convicte, 2022-09-12:
@Koenkk sorry to report, no change:

[image]

1) State for each entity is still not reported
2) The state in attributes for _left and _right is correct
3) ```inverse_cover``` is working correctly

In short, most issues are resolved-just the state is not being published.

Koenkk, 2022-09-12:
Can you provide me the debug logging when starting z2m?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

McGiverGim, 2022-09-12:
Maybe you updated too soon? The changes usually takes some hours to be available.

McGiverGim, 2022-09-12:
@Koenkk I think your changes are being overwrited by this line a little later in the code:

https://github.com/Koenkk/zigbee2mqtt/blob/809f5d3ed14d909d79a576825ecd76cab734c989/lib/extension/homeassistant.ts#L417

If it does not expose `motor_state` and `running` it will enter in this else part.

convicte, 2022-09-12:
> Can you provide me the debug logging when starting z2m?
> 
> See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

I believe you'd be interested in this section, where the specific dev version commit is published:

```yaml
info  2022-09-12 12:21:09: Logging to console and directory: '/config/zigbee2mqtt/log/2022-09-12.12-21-09' filename: log.txt
info  2022-09-12 12:21:09: Starting Zigbee2MQTT version 1.27.2-dev (commit #125ca61)
info  2022-09-12 12:21:09: Starting zigbee-herdsman (0.14.54)
info  2022-09-12 12:21:09: zigbee-herdsman started (resumed)
info  2022-09-12 12:21:09: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20211217,"transportrev":2},"type":"zStack3x0"}'
info  2022-09-12 12:21:09: Set transmit power to '20'
```

Koenkk, 2022-09-13:
@McGiverGim you are right, pushed another fix

convicte, 2022-09-13:
@Koenkk, not sure if I pulled the Edge add-on too quickly before your latest change took effect, but unfortunately there is no difference in this version:

```yaml

info  2022-09-13 09:56:55: Logging to console and directory: '/config/zigbee2mqtt/log/2022-09-13.09-56-55' filename: log.txt
info  2022-09-13 09:56:55: Starting Zigbee2MQTT version 1.27.2-dev (commit #809f5d3)
info  2022-09-13 09:56:55: Starting zigbee-herdsman (0.14.54)
info  2022-09-13 09:56:55: zigbee-herdsman started (resumed)
info  2022-09-13 09:56:55: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20211217,"transportrev":2},"type":"zStack3x0"}'

```

Please let me know if the version I have is the one with the change or if there is anything I should do?

McGiverGim, 2022-09-13:
You pulled it too soon. 809f5d3 is from yesterday.

convicte, 2022-09-13:
@Koenkk - success!!

[image]

stefanuytterhoeven, 2022-09-13:
Sorry to interrupt. I see "success" by convicte :)
Any idea when we can expect a release? I don't know when you release... monthly? After an amount of corrections,....
It's an annoying bug, that's all.
Anyway, you all are doing a great job! Thx

McGiverGim, 2022-09-13:
The release is monthly, it will be released at beginning of october. If you need it before, install the edge version.

Koenkk, 2022-09-14:
@convicte awesome, time to close?

stefanuytterhoeven, 2022-09-14:
@McGiverGim thanks. It's a no blocking bug for me, so I'll wait 
Thanks!

convicte, 2022-09-14:
@Koenkk - Yessir!

All three issues seem to be resolved for the _dual device

1) State for each entity is correct, as indicated above.
2) The state in attributes for _left and _right is correct
3) ```inverse_cover``` is working correctly (when false 100 is open and 0 is closed)

closed 2022-09-14 by convicte

stefanuytterhoeven, 2022-09-14:
Thanks for testing!