Koenkk/zigbee2mqtt #13945: Aqara roller shade companion E1 - Unknown Key Entries in Log

leacho73, 2022-09-09:
### What happened?

I have 5 x Aqara roller shade companion E1 devices - Z2M shows them as supported devices listing them as lumi.curtain.acn002
.
They are very intermittent whereas the rest of the network is very stable.

When trying to directly interact with the devices through the Z2M UI - I get the following message:

Publish 'set' state to 'Lounge Blinds' failed: '#Error Write 0xGUID/1' genAnalogueOutput({"85":{"value":0;"type":57}},["sendWhen":"immediate","timeout" failed (Data request failed with error: 'No network route' (205))'

I've noticed also, at a similar time Z2M appears to crash and restart - with nothing written to the logs to say why its restarted.

### What did you expect to happen?

Expected Z2M to operate normally as per other devices on the network.

### How to reproduce it (minimal and precise)

Inside Z2M - click on the device - select 'Exposes' and try and adjust the blind - the error appears in a red box in the top right hand corner.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

smlight-zigbee-lan-adapter-slzb-05en

leacho73, 2022-09-21:
Just a quick update on this - I've now removed the Aqara Shade Companions from Z2M - and the random reboots have stopped - so looks like they were causing the issue - still none the wiser as to why however!

closed 2022-10-30 by github-actions[bot]

mamrai1, 2023-04-12:
zha handler:
----
"""Aqara Roller Shade Driver E1 device."""
from __future__ import annotations

from typing import Any

from zigpy import types as t
from zigpy.profiles import zha
from zigpy.zcl import foundation
from zigpy.zcl.clusters.closures import WindowCovering
from zigpy.zcl.clusters.general import (
    Alarms,
    AnalogOutput,
    Basic,
    DeviceTemperature,
    GreenPowerProxy,
    Groups,
    Identify,
    MultistateOutput,
    OnOff,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.manufacturer_specific import ManufacturerSpecificCluster

from zhaquirks import Bus, CustomCluster, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.xiaomi import LUMI, BasicCluster, XiaomiCluster, XiaomiCustomDevice

PRESENT_VALUE = 0x0055
CURRENT_POSITION_LIFT_PERCENTAGE = 0x0008
GO_TO_LIFT_PERCENTAGE = 0x0005
DOWN_CLOSE = 0x0001
UP_OPEN = 0x0000
STOP = 0x0002

class XiaomiAqaraRollerE1(XiaomiCluster, ManufacturerSpecificCluster):
    """Xiaomi mfg cluster implementation specific for E1 Roller."""

    cluster_id = 0xFCC0

    attributes = XiaomiCluster.attributes.copy()
    attributes.update(
        {
            0x0400: ("reverse_direction", t.Bool, True),
            0x0402: ("positions_stored", t.Bool, True),
            0x0407: ("store_position", t.uint8_t, True),
            0x0408: ("speed", t.uint8_t, True),
            0x0409: ("charging", t.uint8_t, True),
            0x00F7: ("aqara_attributes", t.LVBytes, True),
        }
    )

class AnalogOutputRollerE1(CustomCluster, AnalogOutput):
    """Analog output cluster, only used to relay current_value to WindowCovering."""

    cluster_id = AnalogOutput.cluster_id

    def __init__(self, *args: Any, **kwargs: Any) -> None:
        """Init."""
[... 201 lines cut ...]
                    MultistateOutput.cluster_id,
                    OnOff.cluster_id,
                    Scenes.cluster_id,
                    WindowCovering.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                    Time.cluster_id,
                ],
            },
            # <SizePrefixedSimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=0,
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }
---