Koenkk/zigbee2mqtt #13947: Smart LED Light Strip from LIDL - Effects don't work

senarvi, 2022-09-09:
### What happened?

I've bought four of these Livarno Lux LED strips from LIDL. Two of them I bought a couple of years ago and two of them I bought now. They are recognized as "Livarno Lux smart LED light strip 2.5m (HG06104A)". I never managed to get the effects work. The other commands such as "color", "brightness", "color_temp_move" work. I think "transition" doesn't have any effect. Whenever I publish "effect" with a valid effect name (e.g. '{"effect": "breathe"}') the lights will quickly fade out and fade in. It does the same thing with any valid value ("blink", "breathe", "okay", "channel_change", "finish_effect", "stop_effect"). If I give an invalid value, it does nothing. Has anyone got the effects working with LIDL LEDs?

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20211217

### Adapter

ZigStar LAN Gateway

evlo, 2022-09-09:
Not working for me also, I confirm that, I don't use effects, so I did not complain.

senarvi, 2022-09-10:
> Not working for me also, I confirm that, I don't use effects, so I did not complain.

Thanks @evlo . Just wanted to check if I'm doing something wrong.

Koenkk, 2022-09-10:
>  the lights will quickly fade out and fade in. It does the same thing with any valid value ("blink", "breathe", "okay", "channel_change", "finish_effect", "stop_effect"). If I give an invalid value, it does nothing. Has anyone got the effects working with LIDL LEDs?

This means effects are not supported by the firmware of the device.

senarvi, 2022-09-12:
> This means effects are not supported by the firmware of the device.

Ok. I guess they sell the devices with different firmware in different countries.

dirkessl, 2022-09-20:
Interesting. on mine not even the colors or color temp works.

closed 2022-10-30 by github-actions[bot]