Koenkk/zigbee2mqtt #13954: Tuya TS0002_switch_module não aciona canais individualmente

andregoncalvespires, 2022-09-10:
### What happened?

Hello everyone.

I already use several devices model [Tuya TS0002_switch_module](https://www.zigbee2mqtt.io/devices/TS0002_switch_module.html#tuya-ts0002_switch_module) and they all work perfectly. Right now, I'm replacing the latest wifi devices with a few more of this model. And to my surprise, I'm encountering strange behavior. After pairing on the Z2M, when I turn on one of the channels, the other is also turned on. I imagined that it might be a need to update the firmware, but when pairing them in a tuya hub, there is no update and the operation is perfect.

This problem only happens for new devices, the ones that were already on the Z2M continue to work perfectly.

Can anyone save me?

### What did you expect to happen?

That the operation of the channels were independent

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

Home Assistant 2022.9.1

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

### Debug log

No data

andregoncalvespires, 2022-09-10:
An important information, the problem also happens with ZHA

MattWestb, 2022-09-11:
The device is very likely need "tuya magic spell" being  casted on it for working normal (like most tuya devices updated firmware).

I have one 4 gang version bayed for 7 month ago and its working OK (in ZHA) but very likely its having older firmware as your new one.

Koenkk, 2022-09-11:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-11 by Koenkk

andregoncalvespires, 2022-09-11:
Great! This job deserves a Coffee!

gericho, 2022-09-11:
Sorry my tuya meter on with new firmware (Ver.IVAP) gives this warnings, could you check please. Thank you!

I'm using HASSOS EDGE Plugin

```
2022-09-11 19:13:59zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #16 with data {"dp":16,"datatype":1,"data":{"type":"Buffer","data":[1]}}
warning 2022-09-11 19:14:00zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #6 with data {"dp":6,"datatype":0,"data":{"type":"Buffer","data":[0,0,0,182,0,0,0,40,3,35,0,3,214,9,3]}}
info 2022-09-11 19:14:03MQTT publish: topic 'zigbee2mqtt/PowerMeter', payload '{"energy":43.18,"linkquality":72,"power":182,"state":"ON"}'
info 2022-09-11 19:14:04MQTT publish: topic 'zigbee2mqtt/PowerMeter', payload '{"energy":43.19,"linkquality":76,"power":182,"state":"ON"}'
warning 2022-09-11 19:14:05zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
info 2022-09-11 19:14:06MQTT publish: topic 'zigbee2mqtt/PowerMeter', payload '{"energy":43.19,"linkquality":76,"power":182,"state":"ON"}'
warning 2022-09-11 19:14:07zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,2,55]}}
warning 2022-09-11 19:14:08zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #110 with data {"dp":110,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,39]}}
warning 2022-09-11 19:14:09zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #111 with data {"dp":111,"datatype":2,"data":{"type":"Buffer","data":[0,0,3,36]}}
warning 2022-09-11 19:14:11zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #9 with data {"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}
warning 2022-09-11 19:14:12zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #17 with data {"dp":17,"datatype":0,"data":{"type":"Buffer","data":[0,0,0,0,0,0,0,0,0,0,0,100,0,1,14,0,0,170,0,0,0,0,0,10,0,0,0,0,0,0]}}
warning 2022-09-11 19:14:12zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #18 with data {"dp":18,"datatype":0,"data":{"type":"Buffer","data":[0,5,0,100,0,5,0,30,0,60,0,0,0,0,0,0,0]}}
warning 2022-09-11 19:14:14zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #105 with data {"dp":105,"datatype":2,"data":{"type":"Buffer","data":[0,0,19,135]}}
```
[image]

tomekk3108, 2022-09-27:
Thank you! When can we expect it in official version?

Koenkk, 2022-09-28:
1 October!