Koenkk/zigbee2mqtt #13956: [New device support]:

sebastian-tennant, 2022-09-11:
### Link

https://www.aliexpress.com/item/1005003041905466.html?spm=a2g0o.order_list.0.0.2faf1802o1fmK9

### Database entry

{"id":21,"type":"EndDevice","ieeeAddr":"0xa4c1381f0f95dba1","nwkAddr":19327,"manufId":4417,"manufName":"_TZ3000_mg4dy6z6","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x00212effff05da84","zoneState":1,"currentZoneSensitivityLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1662843589411,"defaultSendRequestWhen":"immediate"}

### Comments

This appears to be a new manufacturer or firmware version of the already supported [TS0202](https://www.zigbee2mqtt.io/devices/TS0202.html) but is coming up as unsupported.

Koenkk, 2022-09-11:
Please check https://github.com/Koenkk/zigbee2mqtt/issues/9757#issuecomment-1242909728

Koenkk, 2022-09-12:
`_TZ3000_mg4dy6z6` is supported now in the dev branch.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-12 by Koenkk