Koenkk/zigbee2mqtt #13962: [New device support]: TS0601

Schlauchkaefer, 2022-09-11:
### Link

https://www.amazon.de/gp/product/B08GN52SWS/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

### Database entry

{"id":7,"type":"EndDevice","ieeeAddr":"0x540f57fffeec8c08","nwkAddr":13851,"manufId":4098,"manufName":"_TZE200_i48qyn9s","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"�[�*\u0013!j�*\u0012\"j�*\u0012","65506":27,"appVersion":64,"modelId":"TS0601","manufacturerName":"_TZE200_i48qyn9s","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00212effff09269e","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00212effff09269e","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00212effff09269e","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff09269e","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00212effff09269e","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1662885734380,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to follow the supporting new Tuya device page but got stuck at understanding Tuya data points.

This is from the log after using the externel converter:
zigbee-herdsman-converters:tuya_data_point_dump: Received Tuya DataPoint #34 from 0x540f57fffeec8c08 with raw data '{"dp":34,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,149]}}': type='commandDataReport', datatype='value', value='149', known DP# usage: []

On the Device is a Typ printed: ESS-HK-TRV-6102

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_i48qyn9s'
        },
    ],
    model: 'ess-hk-trv-6102',
    vendor: 'wesmartify',
    description: 'Thermostatic radiator valve',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;
```

Schlauchkaefer, 2022-09-17:
I am using Home Assistant with the Zigbee2MQTT Addon, so i think it is noch possible to edit the three Files that are referenced in the Add new Tuya Devices Manual.

So i need a solution out of the Box or even an external Converter, i think.

schluese, 2022-09-26:
I need support too

fr4nk3n5731n, 2022-10-10:
I'll try to help with this, I've got a tuya gateway on order to the requried data points as described in the documentation, I'll keep you posted.

fr4nk3n5731n, 2022-10-12:
@konvex1337 wanna play guinea pig for my external converter? I can't get my thermostat to pair to Z2M anymore but I did manage to map out all the datapoints

fr4nk3n5731n, 2022-10-12:
https://gist.github.com/fr4nk3n5731n/6ec787defb62f1bb17ae1a74eec5b7f5 I basically took the code from the Lidl Silvercrest Thermostat (because it looks the same) and changed the datapoint mapping

Schlauchkaefer, 2022-10-12:
I tried to use the converter. But i think im still doing something wrong.
I can pair the thermostate, but after that i get endless messages:

```
debug 2022-10-12 21:24:42Received Zigbee message from '0x540f57fffeec8c08', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":0}' from endpoint 1 with groupID null
warning 2022-10-12 21:24:42Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
```

fr4nk3n5731n, 2022-10-12:
the thermostat doesn't have a endpoint 1 as far as I can tell, and we can iron that out later if it's only that 1 datapoint that doesn't work. I'd argue timesync doesn't matter as much as setting the heat and mode does when you can schedule the heating with Home Assisstant or something similar. At least not initially.
So does it show the current temperature and can you set the heating temperature and mode?

Schlauchkaefer, 2022-10-12:
I got so many red boxes i nearly wasnt able to remove the device, so i didnt realy take a closer look. But i will pair it again and will see, what it shows. Just to show you what i mean here a nother snippet of my log.

```
warn  2022-10-12 21:20:09: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
warn  2022-10-12 21:20:09: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
debug 2022-10-12 21:20:09: Received Zigbee message from '0x540f57fffeec8c08', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,220],"type":"Buffer"},"datatype":2,"dp":24}],"seq":6144}' from endpoint 1 with groupID null
warn  2022-10-12 21:20:09: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
warn  2022-10-12 21:20:09: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
debug 2022-10-12 21:20:09: Received Zigbee message from '0x540f57fffeec8c08', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":2}],"seq":512}' from endpoint 1 with groupID null
warn  2022-10-12 21:20:09: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
warn  2022-10-12 21:20:09: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
debug 2022-10-12 21:20:09: Received Zigbee message from '0x540f57fffeec8c08', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0],"type":"Buffer"},"datatype":1,"dp":30}],"seq":7680}' from endpoint 1 with groupID null
warn  2022-10-12 21:20:09: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
warn  2022-10-12 21:20:09: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
debug 2022-10-12 21:20:09: Received Zigbee message from '0x540f57fffeec8c08', type 'attributeReport', cluster 'genBasic', data '{"65506":27,"appVersion":64}' from endpoint 1 with groupID null
warn  2022-10-12 21:20:09: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
warn  2022-10-12 21:20:09: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
debug 2022-10-12 21:20:09: Received Zigbee message from '0x540f57fffeec8c08', type 'attributeReport', cluster 'genBasic', data '{"65503":"\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0011\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0012"}' from endpoint 1 with groupID null
warn  2022-10-12 21:20:09: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
warn  2022-10-12 21:20:09: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
debug 2022-10-12 21:20:10: Received Zigbee message from '0x540f57fffeec8c08', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,151],"type":"Buffer"},"datatype":2,"dp":34}],"seq":8704}' from endpoint 1 with groupID null
warn  2022-10-12 21:20:10: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
warn  2022-10-12 21:20:10: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
debug 2022-10-12 21:20:10: Received Zigbee message from '0x540f57fffeec8c08', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,42],"type":"Buffer"},"datatype":2,"dp":101}],"seq":25856}' from endpoint 1 with groupID null
warn  2022-10-12 21:20:10: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
warn  2022-10-12 21:20:10: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
debug 2022-10-12 21:20:10: Received Zigbee message from '0x540f57fffeec8c08', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,42],"type":"Buffer"},"datatype":2,"dp":101}],"seq":25856}' from endpoint 1 with groupID null
warn  2022-10-12 21:20:10: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
warn  2022-10-12 21:20:10: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
debug 2022-10-12 21:20:10: Received Zigbee message from '0x540f57fffeec8c08', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,34],"type":"Buffer"},"datatype":2,"dp":102}],"seq":26112}' from endpoint 1 with groupID null
warn  2022-10-12 21:20:10: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
warn  2022-10-12 21:20:10: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
debug 2022-10-12 21:20:10: Received Zigbee message from '0x540f57fffeec8c08', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID null
[... cut ...]
warn  2022-10-12 21:20:54: Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_i48qyn9s'
warn  2022-10-12 21:20:54: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
debug 2022-10-12 21:20:55: Received Zigbee message from 'Licht Vorzimmer', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":386083096,"imageType":37132,"manufacturerCode":4509}' from endpoint 1 with groupID null
```

Schlauchkaefer, 2022-10-12:
My bad. Long day. I copied the converter into a file, but i missed to copy the file to the Pi......

fr4nk3n5731n, 2022-10-12:
no worries

fr4nk3n5731n, 2022-10-12:
I got it paired to another zigbee coordinator and switching between auto, manuel and holiday mode (kinda) works, so yay, a small win.

Schlauchkaefer, 2022-10-12:
I think im done for today. I cant get anything to work. I think i will try again at the weekend.

[image]
[image]
[image]
[image]

closed 2023-01-12 by github-actions[bot]