Koenkk/zigbee2mqtt #13964: [ezsp] Ikea Smartplug not joining, errors out with "status":"failed"},"type":"device_interview"}'

tomm1ed, 2022-09-11:
### What happened?

Silvercrest Zigbee Gateway rooted through instruction at https://zigbee.blakadder.com/Lidl_TYGWZ-01.html
zigbee2mqtt running in docker

Trying to add Ikea Smart Plug (https://www.zigbee2mqtt.io/devices/E1603_E1702_E1708.html) to zigbee2mqtt. Device does pair but end up as 'Unsupported'. Said Smart Plug can connect to the Ikea Gateway and was updated to the newest Firmware (2.3.089) using the Ikea Home Smart app prior to trying to connect to zigbee2mqtt. Both the Lidl Gateway and the Plug have been power cycled but the Interview keeps failing.

zigbee2mqtt log:
2022-09-11 11:09:27Device '0x588e81fffe6aef16' joined
info 2022-09-11 11:09:27MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x588e81fffe6aef16","ieee_address":"0x588e81fffe6aef16"},"type":"device_joined"}'
info 2022-09-11 11:09:27MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x588e81fffe6aef16"},"type":"device_connected"}'
info 2022-09-11 11:09:27Starting interview of '0x588e81fffe6aef16'
info 2022-09-11 11:09:27MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x588e81fffe6aef16","ieee_address":"0x588e81fffe6aef16","status":"started"},"type":"device_interview"}'
info 2022-09-11 11:09:27MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x588e81fffe6aef16"},"type":"pairing"}'
error 2022-09-11 11:12:27Failed to interview '0x588e81fffe6aef16', device has not successfully been paired
info 2022-09-11 11:12:27MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x588e81fffe6aef16","ieee_address":"0x588e81fffe6aef16","status":"failed"},"type":"device_interview"}'
info 2022-09-11 11:12:27MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x588e81fffe6aef16"},"type":"pairing"}'

### What did you expect to happen?

The plug pairs successfully

### How to reproduce it (minimal and precise)

Put zigbee2mqtt in pairing mode
Push the reset button on the plug
Wait

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

6.7.8.0 build 373

### Adapter

Silvercrest Zigbee Gateway Model TYGWZ-01 manufactured by Lidl

### Debug log

[zigbee2mqtt-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/9542165/zigbee2mqtt-debug.log)

Koenkk, 2022-09-12:
It seems the device doesn't connect properly. I will need to have a sniff to debug this further, can you provide one when trying to pair the device?

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

tomm1ed, 2022-09-12:
> It seems the device doesn't connect properly. I will need to have a sniff to debug this further, can you provide one when trying to pair the device?
> 
> https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

Hi Koen, thanks for the swift reply. I will do this next week as I will be traveling this week. Hopefully we can get to the bottom of this

sujithg9, 2022-09-14:
@Koenkk @tomm1ed I might be having a similar issue with my Ikea motion sensor which I have listed under the below issue and I have also attached some debug logging in there is that helps.

tomm1ed, 2022-09-19:
> It seems the device doesn't connect properly. I will need to have a sniff to debug this further, can you provide one when trying to pair the device?
> 
> https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

Hi Koen,
 
As promised the capture:
[Zigbee-capture.zip](https://github.com/Koenkk/zigbee2mqtt/files/9597061/Zigbee-capture.zip)

Koenkk, 2022-09-19:
@tomm1ed today a pairing improvement has been merged for EZSP adapters; can you try it? (https://github.com/Koenkk/zigbee-herdsman/pull/580). 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

If it doesn't help; please provide me a sniff again.

tomm1ed, 2022-09-20:
@Koenkk updated to 1.27.2-dev commit: 9b4a03c and the plug doesn't even register for inclusion
When going back to 1.27.2 commit: 7dc48fb the plug once more joins but the plug of course again ends up as interview failed/unsupported

Here's the capture:
[zigbee-9b4a03c.zip](https://github.com/Koenkk/zigbee2mqtt/files/9608324/zigbee-9b4a03c.zip)

kirovilya, 2022-09-20:
@tomm1ed please wait for merge PR https://github.com/Koenkk/zigbee-herdsman/pull/581 and then check again

kirovilya, 2022-09-20:
@tomm1ed I think - today, later

Koenkk, 2022-09-20:
It is in now.

tomm1ed, 2022-09-21:
> It is in now.

Thank you @Koenkk (and @kirovilya of course), will need some time to retest. WIll let you know ASAP

samuelthng, 2022-09-26:
Hello, wanted to check if anyone else with pairing issues faced this:
- Repeated trustCenterJoin frame
- No action taken by z2m after

Not sure if it's related to the pairing issue but wanted to get some opinions and eyes on this.
https://github.com/Koenkk/zigbee-herdsman/blame/85c277f82204be6d7419df396896645da549def6/src/adapter/ezsp/driver/driver.ts#L298

```
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,6,0): 9600010500
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:uart --> [66d421a9512ab85c7e]
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart <-- [67d4a1a9512a20c47e]
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,7,0): 67d4a1a9512a20c47e
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 9680010500
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false}
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:uart <-- [77d4b1a9702a9900fa7d387d38a5126d53ed9d4e27ab59f77e]
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,7,0): 77d4b1a9702a9900fa1818a5126d53ed9d4e27ab59f77e0000
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 96900124008cb2a38c5280b838c1a401000000b439
2022-09-26T15:06:02.610Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_id_":36,"_cls_":"trustCenterJoinHandler","_isRequest_":false,"newNodeId":45708,"newNodeEui64":[164,193,56,184,128,82,140,163],"status":1,"policyDecision":0,"parentOfNewNodeId":0}
2022-09-26T15:06:06.196Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2022-09-26T15:06:06.196Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2022-09-26T15:06:06.197Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2022-09-26T15:06:06.197Z zigbee-herdsman:adapter:ezsp:uart --> DATA (7,0,0): 9700010500
2022-09-26T15:06:06.197Z zigbee-herdsman:adapter:ezsp:uart --> [70d521a9512a85687e]
2022-09-26T15:06:06.197Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (0)
2022-09-26T15:06:06.201Z zigbee-herdsman:adapter:ezsp:uart <-- [00d5a1a9512a1dcc7e]
2022-09-26T15:06:06.201Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,0,0): 00d5a1a9512a1dcc7e
2022-09-26T15:06:06.201Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2022-09-26T15:06:06.201Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2022-09-26T15:06:06.202Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 9780010500
2022-09-26T15:06:06.202Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false}
2022-09-26T15:06:06.202Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (0) success
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:uart <-- [10d5b1a9702a9900fa7d387d38a5126d53ed9d4e27ab643a7e]
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,0,0): 10d5b1a9702a9900fa1818a5126d53ed9d4e27ab643a7e0000
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 97900124008cb2a38c5280b838c1a40100000089f4
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_id_":36,"_cls_":"trustCenterJoinHandler","_isRequest_":false,"newNodeId":45708,"newNodeEui64":[164,193,56,184,128,82,140,163],"status":1,"policyDecision":0,"parentOfNewNodeId":0}2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,6,0): 9600010500
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:uart --> [66d421a9512ab85c7e]
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart <-- [67d4a1a9512a20c47e]
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,7,0): 67d4a1a9512a20c47e
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 9680010500
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false}
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:uart <-- [77d4b1a9702a9900fa7d387d38a5126d53ed9d4e27ab59f77e]
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,7,0): 77d4b1a9702a9900fa1818a5126d53ed9d4e27ab59f77e0000
[... cut ...]
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 97900124008cb2a38c5280b838c1a40100000089f4
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_id_":36,"_cls_":"trustCenterJoinHandler","_isRequest_":false,"newNodeId":45708,"newNodeEui64":[164,193,56,184,128,82,140,163],"status":1,"policyDecision":0,"parentOfNewNodeId":0}
```

kirovilya, 2022-09-26:
@samuelthng are you using the latest dev version of z2m? can you attach the full herdsman log with attempts to connect the device? maybe you can still make a sniff of this whole process?

infabo, 2022-09-27:
I experienced the same pairing issues with an IKEA shortcut button. While that button was pairing super-butter-smooth with home-assistant ZHA, it is one of the super-duper-troublesomest devices to pair at zigbee2mqtt.

It starts pairing and after about a minute a pairing failed message is logged.

So I now discovered these github issues and I am now using latest-dev docker image. Now I do not even see/get a "starting interview" message anymore. Nothing happens. So current dev seems to be non-working regarding EZSP adapters.

infabo, 2022-09-27:
> Not sure if it's related to the pairing issue but wanted to get some opinions and eyes on this. https://github.com/Koenkk/zigbee-herdsman/blame/85c277f82204be6d7419df396896645da549def6/src/adapter/ezsp/driver/driver.ts#L298

Seems to be related to my "this does nothing"-observation 🤣

samuelthng, 2022-09-28:
> @samuelthng are you using the latest dev version of z2m? can you attach the full herdsman log with attempts to connect the device? maybe you can still make a sniff of this whole process?

Yes using latest version, I'll try to create the log and sniff. My network is pretty noisy so not sure how to filter out the unrequired data. Will target to do so this weekend due to schedule.

samuelthng, 2022-09-29:
Hey @kirovilya, sadly I don't think I'll be able to provide a sniff without an extended downtime on my setup. I've ordered a conbee and it's on it's way to replace my ezsp hub. Once that arrives I'll do the sniff.

Element2, 2022-09-29:
The latest dev modifications (Update Herdsman to 0.14.61) fixed the problem for me. The dev now pairs and interviews successful.
And after a forced remove again pairs and interviews successful.

infabo, 2022-09-29:
I can confirm. Most recent `latest-dev` docker image resolves the issue for me. IKEA shortcut button successfully pairing now (and again).

TTR6214, 2022-09-30:
I also confirm, now I'm able to pair again device when I use the last Zigbee2MQTT Edge in Home Assistant.

samuelthng, 2022-09-30:
Confirmed as well, all previously removed devices now pairing properly. Thank you for fixing this kirovilya and koenkk. Not tagging because don't wanna ping needlessly.

tomm1ed, 2022-09-30:
OMG awesome guys. Sorry for not getting back earlier but on build 1.27.2-dev commit: b54e299 the Tradfri plug now also works :)

tomm1ed, 2022-09-30:
As this is fixed many thanks to @Koenkk and @kirovilya for taking the time to respond and fix the issue

closed 2022-09-30 by tomm1ed