Koenkk/zigbee2mqtt #13965: No MQTT disconnect reason message when disconnection happens after successful start

Impact123, 2022-09-11:
### What happened?

When the intial connection to the MQTT server fails you get a message like this
```
MQTT failed to connect: getaddrinfo ENOTFOUND mqtt
```
When the connection fails **after** the Z2M was already started you only get this
```
Not connected to MQTT server!
```
When I change this line
```js
this.client.on('error', (err) => reject(err));
```
to
```js
this.client.on('error', (err) => console.log(err));
```
inside [mqtt.ts](https://github.com/Koenkk/zigbee2mqtt/blob/b05b4adf57856fb63efabab26f90585ecc649817/lib/mqtt.ts#L86)
I can get a message like this
```js
Error: getaddrinfo ENOTFOUND mqtt
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (node:dns:109:26) {
  errno: -3008,
  code: 'ENOTFOUND',
  syscall: 'getaddrinfo',
  hostname: 'mqtt'
}
```
so it's clearly possible to provide more information.

### What did you expect to happen?

I expected to get the reason the disconnection happened similar to when it does when it first starts

### How to reproduce it (minimal and precise)

Start Zigbee2MQTT. Shut down the MQTT server.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

 20211115

### Adapter

CC2531

Koenkk, 2022-09-12:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

kleijnweb, 2022-09-12:
@Koenkk So now you're no longer logging the reason at startup?

Koenkk, 2022-09-12:
It is always logged now: https://github.com/Koenkk/zigbee2mqtt/commit/125ca61812a3fd47834943265777bfdabff600d1#

kleijnweb, 2022-09-12:
Ah ok so now also on connect the error reason is logged by the `error` event handler.