Koenkk/zigbee2mqtt #13966: GWA1531_Shutter error in settings specific invert_cover

TRuHa83, 2022-09-11:
### What happened?

Doesn't display status correctly. When the shutter is fully open it shows 100% which is correct and the buttons work correctly. the problem is that in the specific settings it is in **True**. I want them to be in **False**

> Specific settings according to Frontend
> Inverts the cover position, false: open=100,close=0, true: open=0,close=100.

This causes an erroneous state of the blind in Home Assistant even though the buttons work correctly. What's going on? I'm going crazy with the problem.

[image]

[image]

### What did you expect to happen?

With the setting that is set, it should be, in 100% state it should be closed.

### Zigbee2MQTT version

1.27.2 commit: 7dc48fb

### Adapter firmware version

20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle

Koenkk, 2022-09-12:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['GWA1531_Shutter'],
    model: 'GWA1531',
    description: 'Shutter actuator',
    vendor: 'Gewiss',
    fromZigbee: [fz.cover_position_tilt, fz.ignore_basic_report],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    meta: {coverInverted: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
        await reporting.currentPositionLiftPercentage(endpoint);
    },
    exposes: [e.cover_position()],
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

bilbon0bud, 2022-09-12:
Hello

I have the same issue with tuya TS130F.

Best regards

bilbon0bud, 2022-09-12:
i edit the converter to match my zigbee reference and it work almost...
in HA i have the good state, but now HA don't give me the good button anymore.
When my cover is open i can just move the slider or click open...

TRuHa83, 2022-09-12:
> ¿Podría verificar si el problema se solucionó con el siguiente convertidor externo:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     zigbeeModel: ['GWA1531_Shutter'],
>     model: 'GWA1531',
>     description: 'Shutter actuator',
>     vendor: 'Gewiss',
>     fromZigbee: [fz.cover_position_tilt, fz.ignore_basic_report],
>     toZigbee: [tz.cover_state, tz.cover_position_tilt],
>     meta: {coverInverted: true},
>     configure: async (device, coordinatorEndpoint, logger) => {
>         const endpoint = device.getEndpoint(1);
>         await reporting.bind(endpoint, coordinatorEndpoint, ['closuresWindowCovering']);
>         await reporting.currentPositionLiftPercentage(endpoint);
>     },
>     exposes: [e.cover_position()],
> };
> 
> module.exports = definition;
> ```
> 
>     * guardar esto como archivo junto a `configuration.yaml`como `ext_converter.js`
> 
>     * añádelo a `configuration.yaml`:
> 
> 
> ```yaml
> external_converters:
>   - ext_converter.js
> ```
> 
>     * start z2m, check if issue is fixed

Yes now!! It works perfect. You are a crack. Thank you very much, you do a great job.

closed 2022-09-12 by TRuHa83

Koenkk, 2022-09-13:
@TRuHa83 great pushed the fix

@bilbon0bud the TS130F issue has already been fixed in the latest dev

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)