Koenkk/zigbee2mqtt #13970: Aqara smart wall switch H1 EU (WS-EUK03) not working with group scenes

outlying, 2022-09-11:
### What happened?

https://youtube.com/shorts/sKDWExGhoV4

I have a simple group that includes Philips Hue bulb and Aqara smart wall switch, I defined two scenes for that group:

1. on - where both are turned on
2. off - where both are turned off

When I recall scenes only the bulb reacts, wall switch remains in the same state it was before the scene recall. 

See video I linked above

### What did you expect to happen?

I would expect wall switches to work with group+scenes 

### How to reproduce it (minimal and precise)

As mentioned above, create a simple group with those wall switches and try to apply scenes

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

0x26780700

### Adapter

Conbee II

### Debug log

Zigbee2MQTT:info  2022-09-11 18:05:13: Successfully recalled group scene
Zigbee2MQTT:info  2022-09-11 18:05:13: MQTT publish: topic 'zigbee2mqtt/kitchen_lights_mood', payload '{"brightness":29,"color_mode":"color_temp","color_temp":454,"state":"ON"}'
Zigbee2MQTT:info  2022-09-11 18:05:13: MQTT publish: topic 'zigbee2mqtt/kitchen_light_mood_1', payload '{"brightness":29,"color_mode":"color_temp","color_temp":454,"color_temp_startup":454,"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-11 18:05:13: MQTT publish: topic 'zigbee2mqtt/corridor_lights', payload '{"color_mode":"xy","state":"ON"}'
Zigbee2MQTT:info  2022-09-11 18:05:13: MQTT publish: topic 'zigbee2mqtt/corridor_switch_light_1', payload '{"action":null,"color_mode":"xy","consumption":0.12,"current":0,"device_temperature":38,"energy":0.12,"flip_indicator_light":"OFF","led_disabled_night":null,"linkquality":255,"operation_mode":"decoupled","power":0,"power_outage_count":5,"power_outage_memory":null,"state":"ON","voltage":228.2}'
Zigbee2MQTT:info  2022-09-11 18:05:13: MQTT publish: topic 'zigbee2mqtt/testing', payload '{"brightness":29,"color_mode":"color_temp","color_temp":454,"state":"ON"}'
Zigbee2MQTT:info  2022-09-11 18:05:13: MQTT publish: topic 'zigbee2mqtt/testing', payload '{"brightness":29,"color_mode":"xy","color_temp":454,"state":"ON"}'
Zigbee2MQTT:info  2022-09-11 18:05:13: MQTT publish: topic 'zigbee2mqtt/living_room_kitchen_lights', payload '{"brightness":29,"color_mode":"color_temp","color_temp":454,"state":"ON"}'
Zigbee2MQTT:info  2022-09-11 18:05:16: Successfully recalled group scene
Zigbee2MQTT:info  2022-09-11 18:05:16: MQTT publish: topic 'zigbee2mqtt/kitchen_lights_mood', payload '{"brightness":29,"color_mode":"color_temp","color_temp":454,"state":"OFF"}'
Zigbee2MQTT:info  2022-09-11 18:05:16: MQTT publish: topic 'zigbee2mqtt/kitchen_light_mood_1', payload '{"brightness":29,"color_mode":"color_temp","color_temp":454,"color_temp_startup":454,"linkquality":255,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2022-09-11 18:05:16: MQTT publish: topic 'zigbee2mqtt/corridor_lights', payload '{"color_mode":"xy","state":"OFF"}'
Zigbee2MQTT:info  2022-09-11 18:05:16: MQTT publish: topic 'zigbee2mqtt/corridor_switch_light_1', payload '{"action":null,"color_mode":"xy","consumption":0.12,"current":0,"device_temperature":38,"energy":0.12,"flip_indicator_light":"OFF","led_disabled_night":null,"linkquality":255,"operation_mode":"decoupled","power":0,"power_outage_count":5,"power_outage_memory":null,"state":"OFF","voltage":228.2}'
Zigbee2MQTT:info  2022-09-11 18:05:16: MQTT publish: topic 'zigbee2mqtt/testing', payload '{"brightness":29,"color_mode":"color_temp","color_temp":454,"state":"OFF"}'
Zigbee2MQTT:info  2022-09-11 18:05:16: MQTT publish: topic 'zigbee2mqtt/testing', payload '{"brightness":29,"color_mode":"xy","color_temp":454,"state":"OFF"}'

Koenkk, 2022-09-12:
Seems to be not supported by the switch.

outlying, 2022-09-12:
Are scenes provided by software or hardware? My understanding was it is a feature provided by Zigbee2MQTT

Koenkk, 2022-09-13:
It is a Zigbee feature and executed on the device side. E.g. z2m says to the device restore scene 0, the device is then responsible for doing this.

closed 2022-10-21 by github-actions[bot]