Koenkk/zigbee2mqtt #13971: [New device support]:

gericho, 2022-09-11:
### Link

https://it.aliexpress.com/item/1005002409588154.html?spm=a2g0o.order_list.0.0.18753696sIeqOV&gatewayAdapt=glo2ita

### Database entry

{"id":33,"type":"Router","ieeeAddr":"0x84b4dbfffeb5f542","nwkAddr":6663,"manufId":4098,"manufName":"_TZE200_bkkmqmyo","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"#  *i%  *i)  *i   *i&  *iK  *f\u001d  *i$  *i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_bkkmqmyo","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c9f1e4","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1662917944594,"defaultSendRequestWhen":"immediate"}

### Comments

Get some functionality on EDGE put for example no voltage and no switch works. Manufacturer said with the IVAP versione (please see pictures) they add more features. I can pair it with the EDGE but receiving below errors, I have another exactly same counter bought 2 months ago without IVAP writing on it that works flawlessly.

I'm using HASSOS EDGE Plugin

Thank you!

```
2022-09-11 19:13:59zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #16 with data {"dp":16,"datatype":1,"data":{"type":"Buffer","data":[1]}}
warning 2022-09-11 19:14:00zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #6 with data {"dp":6,"datatype":0,"data":{"type":"Buffer","data":[0,0,0,182,0,0,0,40,3,35,0,3,214,9,3]}}
info 2022-09-11 19:14:03MQTT publish: topic 'zigbee2mqtt/PowerMeter', payload '{"energy":43.18,"linkquality":72,"power":182,"state":"ON"}'
info 2022-09-11 19:14:04MQTT publish: topic 'zigbee2mqtt/PowerMeter', payload '{"energy":43.19,"linkquality":76,"power":182,"state":"ON"}'
warning 2022-09-11 19:14:05zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #102 with data {"dp":102,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,0]}}
info 2022-09-11 19:14:06MQTT publish: topic 'zigbee2mqtt/PowerMeter', payload '{"energy":43.19,"linkquality":76,"power":182,"state":"ON"}'
warning 2022-09-11 19:14:07zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #109 with data {"dp":109,"datatype":2,"data":{"type":"Buffer","data":[0,0,2,55]}}
warning 2022-09-11 19:14:08zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #110 with data {"dp":110,"datatype":2,"data":{"type":"Buffer","data":[0,0,0,39]}}
warning 2022-09-11 19:14:09zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #111 with data {"dp":111,"datatype":2,"data":{"type":"Buffer","data":[0,0,3,36]}}
warning 2022-09-11 19:14:11zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #9 with data {"dp":9,"datatype":5,"data":{"type":"Buffer","data":[0]}}
warning 2022-09-11 19:14:12zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #17 with data {"dp":17,"datatype":0,"data":{"type":"Buffer","data":[0,0,0,0,0,0,0,0,0,0,0,100,0,1,14,0,0,170,0,0,0,0,0,10,0,0,0,0,0,0]}}
warning 2022-09-11 19:14:12zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #18 with data {"dp":18,"datatype":0,"data":{"type":"Buffer","data":[0,5,0,100,0,5,0,30,0,60,0,0,0,0,0,0,0]}}
warning 2022-09-11 19:14:14zigbee-herdsman-converters:TuyaDinRailSwitch: NOT RECOGNIZED DP #105 with data {"dp":105,"datatype":2,"data":{"type":"Buffer","data":[0,0,19,135]}}

DSC_1443
```
[image]

gericho, 2022-09-24:
Updated to last EDGE since days, still no luck 👎

Koenkk, 2022-09-25:
Lets continue in https://github.com/Koenkk/zigbee2mqtt/issues/13767

closed 2022-09-25 by Koenkk