Koenkk/zigbee2mqtt #13973: Tuya TS0201 Reporting Humidity off by factor of 10

fmartinou, 2022-09-11:
### What happened?

Hi,

Below my first issue on this **amazing** project 🤩 !

I bought those [Moes temperature sensors](https://www.aliexpress.com/item/1005004517035659.html?spm=a2g0o.order_list.0.0.60615e5bmEU44v&gatewayAdapt=glo2fra); the hardware is the `Hangzhou Kangbazi  KCTW1Z`.

All sensors are reporting fine **except the humidity which is reporting the correct value divided by 10**.
(e.g. 5.1% instead of 51%)

[image]

Zigbee2mqtt recognizes them as [`Tuya TS0201`](https://www.zigbee2mqtt.io/devices/TS0201.html#tuya-ts0201).

[This issue seems pretty similar to this one](https://github.com/Koenkk/zigbee2mqtt/issues/13328).

Or maybe those devices should be recognized as [`Tuya KCTW1Z`](https://www.zigbee2mqtt.io/devices/KCTW1Z.html)?

Thank you very much for your suppport!

### What did you expect to happen?

Humidity value reported 10 times more than currently.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

 6.10.3.0 build 297

### Adapter

ITead Sonoff Zigbee 3.0 USB Dongle Plus model ZBDongle-E

### Debug log

```
Debug 2022-09-11 22:13:57Received Zigbee message from '0xa4c138548b0ee91e', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":578}' from endpoint 1 with groupID 0
Info 2022-09-11 22:13:57MQTT publish: topic 'zigbee2mqtt/0xa4c138548b0ee91e', payload '{"battery":91,"humidity":5.78,"linkquality":184,"temperature":24.78,"voltage":2900}'
Debug 2022-09-11 22:14:17Received Zigbee message from '0xa4c138548b0ee91e', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":516}' from endpoint 1 with groupID 0
Info 2022-09-11 22:14:17MQTT publish: topic 'zigbee2mqtt/0xa4c138548b0ee91e', payload '{"battery":91,"humidity":5.16,"linkquality":168,"temperature":24.78,"voltage":2900}'
Debug 2022-09-11 22:15:42Received Zigbee message from '0xa4c138548b0ee91e', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2509}' from endpoint 1 with groupID 0
Info 2022-09-11 22:15:42MQTT publish: topic 'zigbee2mqtt/0xa4c138548b0ee91e', payload '{"battery":91,"humidity":5.16,"linkquality":168,"temperature":25.09,"voltage":2900}'
Debug 2022-09-11 22:21:36Received Zigbee message from '0xa4c138548b0ee91e', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":182,"batteryVoltage":29}' from endpoint 1 with groupID 0
Info 2022-09-11 22:21:36MQTT publish: topic 'zigbee2mqtt/0xa4c138548b0ee91e', payload '{"battery":91,"humidity":5.16,"linkquality":176,"temperature":25.09,"voltage":2900}'
Debug 2022-09-11 22:32:52Received Zigbee message from '0xa4c138548b0ee91e', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2509}' from endpoint 1 with groupID 0
Info 2022-09-11 22:32:52MQTT publish: topic 'zigbee2mqtt/0xa4c138548b0ee91e', payload '{"battery":91,"humidity":5.16,"linkquality":176,"temperature":25.09,"voltage":2900}'
Debug 2022-09-11 22:32:53Received Zigbee message from '0xa4c138548b0ee91e', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":485}' from endpoint 1 with groupID 0
Info 2022-09-11 22:32:53MQTT publish: topic 'zigbee2mqtt/0xa4c138548b0ee91e', payload '{"battery":91,"humidity":4.85,"linkquality":180,"temperature":25.09,"voltage":2900}'
```

Koenkk, 2022-09-12:
This has been fixed in https://github.com/Koenkk/zigbee-herdsman-converters/issues/4614. Fix is currently only available in the dev branch.  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) and will be in the next release on 1 October.

closed 2022-09-12 by Koenkk

fmartinou, 2022-09-12:
Thank you for your quick answer! 👍