Koenkk/zigbee2mqtt #13980: TuYa TS130F moving direction inverted

Skysurfer-14, 2022-09-12:
### What happened?

The indicated moving direction (UP, DOWN, STOP) seems to be inverted. I am checking the values on the detail page of zigbee2mqtt and with MQTT Explorer. There are the following reasons for my assumption:

1. The moving direction is inverted if I look on the electrical connection of the pushbutton
2. The moving direction is inverted if I look to the output of the device (brown motor wire should be activated on UP)
3. And its inverted if I look to the OPEN/CLOSE state after moving UP or DOWN. After moving UP the indicated state is CLOSE.

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 USB Dongle

convicte, 2022-09-12:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/4609

There are at minimum 5 issues already for similar behaviors.

Skysurfer-14, 2022-09-12:
I suspect that the other errors in this context are consequential errors from the UP/DOWN problem.

convicte, 2022-09-12:
It's, at least in my opinion, unlikely, since there are several devices under the same product code umbrella, and most of them work OK.

Mine are these:
[image]

At least for me, the arrows on the device are resulting in the expected behavior (L1 for UP (open), when the ```motor_reversal: OFF```, and the L2 for DOWN (closed)).

Until 1.27.X everything was fine, beside the ```cover_invert``` showing reversed 0 and 100 positions.

Koenkk, 2022-09-13:
This issue has already been fixed in the latest dev.  (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-09-13 by Koenkk

hardwareadictos, 2024-03-22:
Seems that the issue is happening again (¿?)

Skysurfer-14, 2024-03-22:
I see no issue with the current version 1.36.0-1 of Zigbee2MQTT.

hardwareadictos, 2024-03-22:
> I see no issue with the current version 1.36.0-1 of Zigbee2MQTT.

I will check then. In my case my curtains are inverted on 3 different devices I installed yesterday. Seems that "invert cover" set to true doesn't work or it's being ignored.