Koenkk/zigbee2mqtt #13981: Simulated brightness entity doesn't appear in Home Assistant

YodaDaCoda, 2022-09-12:
### What happened?

I set up `simulated_brightness` on my [E1744](https://www.zigbee2mqtt.io/devices/E1744.html) & observed the brightness attribute changing via the State tab in the webui as I turned the remote. I went to Home Assistant to use the new brightness on an automation and couldn't find it.

### What did you expect to happen?

I expected to find a brightness entity attached to the device in Home Assistant.

### How to reproduce it (minimal and precise)

Pair E1744 device. Observe it is added to Home assistant. Configure `simulated_brightness` in z2m. Observe the brightness changes within the state as the remote is rotated. Note the brightness does not appear in Home Assistant.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

zzh!

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9547854/log.txt)

ModuCZ, 2022-09-24:
I have experienced the same issue.  Brightness value is visible in MQTT Info inside HA.

YodaDaCoda, 2022-10-31:
Issue persists.

closed 2022-12-09 by github-actions[bot]

AndreKR, 2022-12-30:
The issue is still valid.

thefelixno, 2024-06-11:
had to activate legacy entity attributes for the ha-integrations, now works for me

```yml
homeassistant:
  # Optional: Home Assistant legacy entity attributes, (default: shown below), when enabled:
  # Zigbee2MQTT will send additional states as attributes with each entity. For example,
  # A temperature & humidity sensor will have 2 entities for the temperature and
  # humidity, with this setting enabled both entities will also have
  # an temperature and humidity attribute.
  # Note: Disabling this option, requires a Home Assistant restart
  legacy_entity_attributes: true
```

rrakso, 2024-07-24:
I can confirm this. I had (somehow) set `legacy_entity_attributes` to `false`, when changed to true it works very well!