Koenkk/zigbee2mqtt #13982: Add invert open/close commands on cover for Smartwings WM25L-Z

digiblur, 2022-09-12:
### Is your feature request related to a problem? Please describe

The states are correct but it seems in some configurations these covers have the commands reversed where open actually closes it and close actually opens the blinds.  

### Describe the solution you'd like

Add Settings Specific option to reverse the commands much like you can with reversing the state.

### Describe alternatives you've considered

I've tried reversing the cover with the remote but this causes other issues with the states and the remote itself.

### Additional context

Nothing else to add, just reverse the commands being sent would be great!

digiblur, 2022-10-13:
Still needed mr bot

digiblur, 2022-11-14:
Doesn't look like anyone needs this feature Mr Bot so you're good to do your thing.  I'll work around it in other ways.

closed 2022-12-23 by github-actions[bot]

latez, 2023-02-04:
Any chance to resurrect this request, this is still badly needed!

digiblur, 2023-02-04:
Indeed! An awesome feature for all cover devices under settings specific. Invert cover state, commands and percents.

mikerenna, 2023-04-07:
This fixed an issue for me where the zigbee2mqtt commands were reversed. Not sure if this is the root cause of this item, but if so, this may help. 
[image]