Koenkk/zigbee2mqtt #13986: Heiman HS2WD-E siren - no switch in Home Assistant

fribse, 2022-09-12:
### What happened?

I have two Heiman [HS2WD-E](https://www.zigbee2mqtt.io/devices/HS2WD-E.html#heiman-hs2wd-e) sirens.
Looking in z2m they show up like this:
[image]
Which is great, but when I look at it in Home Assistant I just see this:
[image]
So no way of turning the siren on from within Home Assistant.

I can turn it on / off by sending MQTT messages to the siren, but it would be great if it showed normally in HA.

### What did you expect to happen?

I expected to see the siren, flash, flash frequency etc. to appear as switches and sliders in HA.

### How to reproduce it (minimal and precise)

Add the siren to z2m.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

TubesZB

lavinir, 2022-11-06:
I'm seeing the same thing. The device was previously paired with ZHA which created a dedicated Siren entity for this.

closed 2022-12-15 by github-actions[bot]

Home9096, 2023-01-22:
Facing the same issue, however I am not able to make it sound with any type of automation nor even get a signal. The only place I am able to make it sound is in z2m on expose. I have tried following guides and codes of other people making it work, still no luck from my side.

Would be great if anyone could help out or give a clear "how-to". :)

I have tried the following: 
[image]

Nothing really happens except that I get :
[image]

luixal, 2023-02-10:
Hi  @Home9096 , in order to make it work have a look at [this comment](https://github.com/Koenkk/zigbee2mqtt/issues/14424#issuecomment-1374786659) in issue #14424 

It doesn't fix z2m support for this device, but at least you can start/stop the siren.