Koenkk/zigbee2mqtt #13990: OSRAM 4052899926110 - Can't change color in HomeAssistant

impreza2k3, 2022-09-12:
### What happened?

When using HomeAssistant front end with Zigbee2MQTT and the OSRAM 4052899926110 the color wheel selector does not change the color of the lights. Using the temperature slider does change the color. 

Using the Zigbee2mqtt "Exposes" interface color_hs throws an error:  

Publish 'get' 'color' to 'light strip' failed: 'Error: Read 0x841826000005afea/3 lightingColorCtrl(["colorMode","currentX","currentY","enhancedCurrentHue","currentSaturation","colorTemperature"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

### What did you expect to happen?

Color wheel to change the colors.  Using Zigbee2MQTT color_HS does not work but color_xy does.

### How to reproduce it (minimal and precise)

Goto HomeAssistant and attempt to change the color using the color wheel selector

or

Goto Zigbee2MQTT and adjust the color_hs attribute and select refresh

### Zigbee2MQTT version

1.27.2-dev commit: 809f5d3

### Adapter firmware version

 0x26780700

### Adapter

ConBee2/RaspBee2

### Debug log

["zigbee2mqtt_0x841826000005afea"],"manufacturer":"OSRAM","model":"Flex RGBW (4052899926110)","name":"  light strip"},"enabled_by_default":false,"icon":"mdi:palette","json_attributes_topic":"zigbee2mqtt/  light strip","name":"  light strip effect","options":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"unique_id":"0x841826000005afea_effect_zigbee2mqtt"}'
debug 2022-09-12 13:08:22: Received MQTT message on 'homeassistant/sensor/0x841826000005afea/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x841826000005afea"],"manufacturer":"OSRAM","model":"Flex RGBW (4052899926110)","name":"  light strip"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/  light strip","name":"  light strip linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/  light strip","unique_id":"0x841826000005afea_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
debug 2022-09-12 13:08:22: Received MQTT message on 'homeassistant/sensor/0x841826000005afea/update_state/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x841826000005afea"],"manufacturer":"OSRAM","model":"Flex RGBW (4052899926110)","name":"  light strip"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/  light strip","name":"  light strip update state","state_topic":"zigbee2mqtt/  light strip","unique_id":"0x841826000005afea_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
debug 2022-09-12 13:08:22: Received MQTT message on 'homeassistant/binary_sensor/0x841826000005afea/update_available/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x841826000005afea"],"manufacturer":"OSRAM","model":"Flex RGBW (4052899926110)","name":"  light strip"},"device_class":"update","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/  light strip","name":"  light strip update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/  light strip","unique_id":"0x841826000005afea_update_available_zigbee2mqtt","value_template":"{{ value_json['update']['state'] == \"available\" }}"}'
debug 2022-09-12 13:08:25: Received MQTT message on 'zigbee2mqtt/  light strip/set' with data '{"color":{"x":0.6307692307692307,"y":0.3230769230769231}}'
debug 2022-09-12 13:08:25: Skipping state because of Home Assistant
debug 2022-09-12 13:08:25: Publishing 'set' 'color' to '  light strip'
debug 2022-09-12 13:08:25: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x841826000005afea!
info  2022-09-12 13:08:25: MQTT publish: topic 'zigbee2mqtt/  light strip', payload '{"brightness":15,"color":{"h":3,"hue":3,"s":96,"saturation":96,"x":0.6307692307692307,"y":0.3230769230769231},"color_mode":"xy","color_temp":319,"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'
debug 2022-09-12 13:08:26: Received MQTT message on 'zigbee2mqtt/  light strip/get' with data '{"color":""}'
debug 2022-09-12 13:08:26: Skipping state because of Home Assistant
debug 2022-09-12 13:08:26: Publishing get 'get' 'color' to '  light strip'
debug 2022-09-12 13:08:26: Received Zigbee message from '  light strip', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorMode":1,"colorTemperature":219,"currentSaturation":243,"currentX":41337,"currentY":21173}' from endpoint 3 with groupID null
debug 2022-09-12 13:08:26: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x841826000005afea!
info  2022-09-12 13:08:26: MQTT publish: topic 'zigbee2mqtt/  light strip', payload '{"brightness":15,"color":{"h":3,"hue":3,"s":96,"saturation":96,"x":0.6308,"y":0.3231},"color_mode":"xy","color_temp":319,"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'
error 2022-09-12 13:08:26: Publish 'get' 'color' to '  light strip' failed: 'Error: Read 0x841826000005afea/3 lightingColorCtrl(["colorMode","currentX","currentY","enhancedCurrentHue","currentSaturation","colorTemperature"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
debug 2022-09-12 13:08:26: Error: Read 0x841826000005afea/3 lightingColorCtrl(["colorMode","currentX","currentY","enhancedCurrentHue","currentSaturation","colorTemperature"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:317:28)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:480:22)
    at Object.convertGet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1128:13)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:273:21)
info  2022-09-12 13:08:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'color' to '  light strip' failed: 'Error: Read 0x841826000005afea/3 lightingColorCtrl([\"colorMode\",\"currentX\",\"currentY\",\"enhancedCurrentHue\",\"currentSaturation\",\"colorTemperature\"], {\"sendWhen\":\"immediate\",\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"  light strip"},"type":"zigbee_publish_error"}'
debug 2022-09-12 13:08:26: Received MQTT message on 'zigbee2mqtt/  light strip/set' with data '{"color":{"hue":34,"saturation":83}}'
debug 2022-09-12 13:08:26: Skipping state because of Home Assistant
debug 2022-09-12 13:08:26: Publishing 'set' 'color' to '  light strip'
debug 2022-09-12 13:08:27: Missing colorTempPhysicalMin and/or colorTempPhysicalMax for endpoint 0x841826000005afea!
info  2022-09-12 13:08:27: MQTT publish: topic 'zigbee2mqtt/  light strip', payload '{"brightness":15,"color":{"h":34,"hue":34,"s":83,"saturation":83,"x":0.4585,"y":0.4174},"color_mode":"hs","color_temp":361,"linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'
debug 2022-09-12 13:08:27: Received MQTT message on 'zigbee2mqtt/  light strip/get' with data '{"color":""}'
debug 2022-09-12 13:08:27: Skipping state because of Home Assistant
debug 2022-09-12 13:08:27: Publishing get 'get' 'color' to '  light strip'
[... cut ...] iled (no response received)'","meta":{"friendly_name":"  light strip"},"type":"zigbee_publish_error"}'
debug 2022-09-12 13:08:54: Received Zigbee message from 'Adam's Office Motion', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID null
info  2022-09-12 13:08:54: MQTT publish: topic 'zigbee2mqtt/Adam's Office Motion', payload '{"battery":null,"battery_low":false,"linkquality":255,"occupancy":true,"tamper":false}'

desertblade, 2022-09-13:
I had this issue too. Looking at the release log I believed it was related to the change for #13597. 

Edited osram.js in zigbee-herdsman-converters to remove the change (https://github.com/Koenkk/zigbee-herdsman-converters/commit/63436483078fa48960114c12299eb165be362f1b) 

Restarted zigbee2mqtt and color picker works great again in home assistant.

impreza2k3, 2022-09-13:
Thanks for the suggestion, tried restarting both HA and Zigbee2mqtt docker and no change on Home Assistant ver 2022.9.2

desertblade, 2022-09-13:
I manually edited osram.js to remove the change before I restarted zigbee2mqtt docker.

impreza2k3, 2022-09-13:
> 

Thanks for the gentle nudge to re-read what you posted. That also worked for me. The color wheel now works within Home Assistant after manually editing the osram.js.

CorePoint, 2022-09-26:
Just saw this issue because you mentioned mine here.

As I requested this change, I naturally tested it and for me it works perfectly, no change to before besides the improved color accuracy.

impreza2k3, 2022-09-26:
Mine shows:

Firmware build date
20160204N****
Manufacturer
[OSRAM](https://www.zigbee2mqtt.io/supported-devices/#v=OSRAM)
Model

Firmware version is blank
[4052899926110](https://www.zigbee2mqtt.io/devices/4052899926110.html#osram-4052899926110)

CorePoint, 2022-09-26:
Firmware build date for me is `20140331CNWT****` and that's the newest available through z2m OTA functionality.

Seems like there might be some hardware revision stuff going on?

Miloune, 2022-10-10:
Hello,
Updated Z2M to [1.28.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.0) commit: [03ba647d](https://github.com/Koenkk/zigbee2mqtt/commit/03ba647d) and then got near the same error:

I was sending json to MQTT to put on my light like this:
`{"brightness":1,"color":{"x":0.66,"y":0.281},"color_mode":"xy","color_temp":148,"state":"ON"}`

and since the last update, got this error: 
`No converter available for 'set' 'color_mode' (xy)`

So I updated my sending JSON to this and now it's working back:
`{"brightness":1,"color":{"x":0.66,"y":0.281},"state":"ON"}`

desertblade, 2022-10-20:
I believe I updated my firmware before OSRAM discontinued  support for their hub a few years back. I also believe they started to encrypt their firmware, so that is why the last one available in Z2M is from 2014.

My Firmware build date is 20160204N****.  I have not figured out a way to downgrade the firmware. I am not sure I would risk it since these are out of support,

For compatibility I would role back the change.

desertblade, 2022-10-20:
Created a PR in zigbee-herdsman-converters to not default to using preferHS and keeps supportsHS. 

This will make the color wheel work for both XY/HS people, while maybe not having the best color accuracy.   

This should continue to allow using hs_color in automations, for higher color accuracy. I have no way to verify this.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/4804