Koenkk/zigbee2mqtt #13991: Lixee Issue Values are wrong

nbufnoir, 2022-09-12:
### What happened?

Hello,

I've bought a lixee Zlinky device and I've got an issue with 2 values: "apparent_power" & "injected_v_a".

When those values fall to 0, Z2M uses the last value they had instead of sending 0 For example below InjectedVA

debug 2022-09-12 19:23:45: Received Zigbee message from 'Leexy_TIC', type 'readResponse', cluster 'liXeePrivate', data '{"**injectedVA":0**}' from endpoint 1 with groupID 0

info  2022-09-12 19:23:45: MQTT publish: topic 'zigbee2mqtt/Leexy_TIC', payload '{"active_enerfy_out_d01":9763,"active_power":312,"active_power_max":560,...

### What did you expect to happen?

I expected that the MQTT message sent by Z2M says that injectedVA is 0

### How to reproduce it (minimal and precise)

A lixee device with a production system, when you stop injecting power into the grid, injectedVA should fall to 0, instead it keeps the last non-null value

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

20220219

### Adapter

zStack3x0

### Debug log

debug 2022-09-12 18:51:46: Received Zigbee message from 'Leexy_TIC', type 'readResponse', cluster 'liXeePrivate', data '{"injectedVA":526}' from endpoint 1 with groupID 0
info  2022-09-12 18:51:46: MQTT publish: topic 'zigbee2mqtt/Leexy_TIC', payload '{"active_enerfy_out_d01":9763,"active_power":312,"active_power_max":560,"active_power_ph_b":320,"apparent_power":1235,"available_power":9,"average_rms_voltage_meas_period":242,"current_date":"E220912005303","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":10159,"current_summ_received":218,"current_tarif":"BASE","current_tier1_summ_delivered":10156,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3140,"injected_v_a":526,"injected_v_a_max_n":1840,"injected_v_a_max_n1":1530,"linkquality":51,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"812061085925","power_threshold":9,"reactive_power":8,"reactive_power_ph_b":-22926,"reactive_power_ph_c":-26832,"rms_current":2,"rms_voltage":243,"site_id":"19345296663764","software_revision":2,"status_register":"003A4301","total_reactive_power":13895,"update":{"state":"idle"},"update_available":false}'
debug 2022-09-12 18:51:46: Received Zigbee message from 'Leexy_TIC', type 'readResponse', cluster 'liXeePrivate', data '{"injectedVAMaxN":1840}' from endpoint 1 with groupID 0
info  2022-09-12 18:51:46: MQTT publish: topic 'zigbee2mqtt/Leexy_TIC', payload '{"active_enerfy_out_d01":9763,"active_power":312,"active_power_max":560,"active_power_ph_b":320,"apparent_power":1235,"available_power":9,"average_rms_voltage_meas_period":242,"current_date":"E220912005303","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":10159,"current_summ_received":218,"current_tarif":"BASE","current_tier1_summ_delivered":10156,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3140,"injected_v_a":526,"injected_v_a_max_n":1840,"injected_v_a_max_n1":1530,"linkquality":40,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"812061085925","power_threshold":9,"reactive_power":8,"reactive_power_ph_b":-22926,"reactive_power_ph_c":-26832,"rms_current":2,"rms_voltage":243,"site_id":"19345296663764","software_revision":2,"status_register":"003A4301","total_reactive_power":13895,"update":{"state":"idle"},"update_available":false}'
debug 2022-09-12 18:51:46: Received Zigbee message from 'Leexy_TIC', type 'readResponse', cluster 'liXeePrivate', data '{"injectedVAMaxN1":1530}' from endpoint 1 with groupID 0
info  2022-09-12 18:51:46: MQTT publish: topic 'zigbee2mqtt/Leexy_TIC', payload '{"active_enerfy_out_d01":9763,"active_power":312,"active_power_max":560,"active_power_ph_b":320,"apparent_power":1235,"available_power":9,"average_rms_voltage_meas_period":242,"current_date":"E220912005303","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":10159,"current_summ_received":218,"current_tarif":"BASE","current_tier1_summ_delivered":10156,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3140,"injected_v_a":526,"injected_v_a_max_n":1840,"injected_v_a_max_n1":1530,"linkquality":51,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"812061085925","power_threshold":9,"reactive_power":8,"reactive_power_ph_b":-22926,"reactive_power_ph_c":-26832,"rms_current":2,"rms_voltage":243,"site_id":"19345296663764","software_revision":2,"status_register":"003A4301","total_reactive_power":13895,"update":{"state":"idle"},"update_available":false}'
debug 2022-09-12 18:51:46: Received Zigbee message from 'Leexy_TIC', type 'readResponse', cluster 'haElectricalMeasurement', data '{"rmsCurrent":2}' from endpoint 1 with groupID 0
info  2022-09-12 18:51:46: MQTT publish: topic 'zigbee2mqtt/Leexy_TIC', payload '{"active_enerfy_out_d01":9763,"active_power":312,"active_power_max":560,"active_power_ph_b":320,"apparent_power":1235,"available_power":9,"average_rms_voltage_meas_period":242,"current_date":"E220912005303","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":10159,"current_summ_received":218,"current_tarif":"BASE","current_tier1_summ_delivered":10156,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3140,"injected_v_a":526,"injected_v_a_max_n":1840,"injected_v_a_max_n1":1530,"linkquality":54,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"812061085925","power_threshold":9,"reactive_power":8,"reactive_power_ph_b":-22926,"reactive_power_ph_c":-26832,"rms_current":2,"rms_voltage":243,"site_id":"19345296663764","software_revision":2,"status_register":"003A4301","total_reactive_power":13895,"update":{"state":"idle"},"update_available":false}'
debug 2022-09-12 18:51:46: Received Zigbee message from 'Leexy_TIC', type 'readResponse', cluster 'haElectricalMeasurement', data '{"rmsVoltage":243}' from endpoint 1 with groupID 0
info  2022-09-12 18:51:46: MQTT publish: topic 'zigbee2mqtt/Leexy_TIC', payload '{"active_enerfy_out_d01":9763,"active_power":312,"active_power_max":560,"active_power_ph_b":320,"apparent_power":1235,"available_power":9,"average_rms_voltage_meas_period":242,"current_date":"E220912005303","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":10159,"current_summ_received":218,"current_tarif":"BASE","current_tier1_summ_delivered":10156,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3140,"injected_v_a":526,"injected_v_a_max_n":1840,"injected_v_a_max_n1":1530,"linkquality":51,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"812061085925","power_threshold":9,"reactive_power":8,"reactive_power_ph_b":-22926,"reactive_power_ph_c":-26832,"rms_current":2,"rms_voltage":243,"site_id":"19345296663764","software_revision":2,"status_register":"003A4301","total_reactive_power":13895,"update":{"state":"idle"},"update_available":false}'
debug 2022-09-12 18:52:02: Received Zigbee message from 'Leexy_TIC', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,10159304],"currentSummReceived":[0,217896]}' from endpoint 1 with groupID 0
info  2022-09-12 18:52:02: MQTT publish: topic 'zigbee2mqtt/Leexy_TIC', payload '{"active_enerfy_out_d01":9763,"active_power":312,"active_power_max":560,"active_power_ph_b":320,"apparent_power":1235,"available_power":9,"average_rms_voltage_meas_period":242,"current_date":"E220912005303","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":10159,"current_summ_received":218,"current_tarif":"BASE","current_tier1_summ_delivered":10156,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3140,"injected_v_a":526,"injected_v_a_max_n":1840,"injected_v_a_max_n1":1530,"linkquality":40,"message1":"PAS DE MESSAGE","message2":"","meter_serial_number":"812061085925","power_threshold":9,"reactive_power":8,"reactive_power_ph_b":-22926,"reactive_power_ph_c":-26832,"rms_current":2,"rms_voltage":243,"site_id":"19345296663764","software_revision":2,"status_register":"003A4301","total_reactive_power":13895,"update":{"state":"idle"},"update_available":false}'
debug 2022-09-12 18:52:07: Received Zigbee message from 'Detecteur Cellier', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 2 with groupID 0
info  2022-09-12 18:52:07: MQTT publish: topic 'zigbee2mqtt/Detecteur Cellier', payload '{"battery":54,"illuminance":19999,"illuminance_lux":100,"led_indication":null,"linkquality":156,"motion_sensitivity":"high","occupancy":true,"occupancy_timeout":30,"temperature":24.76,"update":{"state":"idle"},"update_available":false}'
[... cut ...]
info  2022-09-12 18:53:46: MQTT publish: topic 'zigbee2mqtt/Leexy_TIC', payload '{"active_enerfy_out_d01":9763,"active_power":312,"active_power_max":560,"active_power_ph_b":320,"apparent_power":519,"available_power":9,"average_rms_voltage_meas_period":242,"current_date":"E220912005303","current_index_tarif":1,"current_price":"BASE","current_summ_delivered":10159,"current_summ_received":218,"current_tarif":"BASE","current_tier1_summ_delivered":10156,"current_tier2_summ_delivered":0,"drawn_v_a_max_n1":3140,"injected_v_a":526,"injected_v_a_max_n":1840,"injected_v_a_max_n1":1530,"linkquality":51,"message1":"PAS DE

arenoux, 2022-09-23:
Same issue here !
Would love to help that being fixed :)

Koenkk, 2022-09-23:
@vk496 / @SilentT-FR  could you check this?

BrunoSpy, 2022-10-03:
Just updated to 1.28.0 : problem still here.

ddelpire, 2022-10-03:
Same problem in 1.28.0. injectedVA should fall to 0, instead it keeps the last non-null value

hdeadvance01, 2022-10-08:
hello
I have same problem if I injected VA, sensor PAPP (zlinky_tic)  diplay anything but not 0VA
then linky electric meter diplay 0VA
HA version core-2022.9.7 and Zigbee2mqtt version:1.28.0-1

BrunoSpy, 2022-10-08:
This issue is really troublesome so I tried to look at the code but I don't know where to start.
Maybe there's a bug in https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lixee.js but I can't figure out where.

hdeadvance01, 2022-10-08:
sorry I am novice on HA and I am not competent to understand the code

hdeadvance01, 2022-10-08:
strange sensor PAPP never goes to zero see file:

[image]

BrunoSpy, 2022-10-08:
Maybe @vk496 can help ?

ddelpire, 2022-10-09:
This problem happens with Z2MQTT, with ZHA 0VA value is present. It seems like a problem with negative value (when auto-production is higher than tier deliver).

ddelpire, 2022-10-09:
@vk496  Problem of conversion number to string  in lib/extension/receive.ts ?

const converters = data.device.definition.fromZigbee.filter((c) => {
            const type = Array.isArray(c.type) ? c.type.includes(data.type) : c.type === data.type;
            return c.cluster === data.cluster && type;
        });

vk496, 2022-10-09:
Hello,

Sorry, difficult days and I can't take a look rn.

The problem is with conversion of data types (or precision with floats) for sure, but not there. You should take a look in the lixee.js

Have a nice day :)
Valentín

vk496, 2022-10-09:
Just reviewed the code from my phone. I think the bug may come from bad condition check. Specifically, those lines:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lixee.js#L51

```
if (msg.data[at] != null) {
```

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lixee.js#L107

```
if (val != null) {
```

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lixee.js#L158

```
if (val != null) {
```

I think those 3 lines should fix this bug and any future related. However, I'm not able to test if is really the cause. Any volunteer?

Have a nice day :)
Valentín

nbufnoir, 2022-10-10:
> Just reviewed the code and the issue description from my phone. I think the bug may come from bad condition check. Specifically, those lines:
> 
> https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lixee.js#L51
> 
> ```
> if (msg.data[at] != null) {
> ```
> 
> https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lixee.js#L107
> 
> ```
> if (val != null) {
> ```
> 
> https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/lixee.js#L158
> 
> ```
> if (val != null) {
> ```
> 
> I think those 3 lines should fix this bug and any future related. However, I'm not able to test if is really the cause. Any volunteer?
> 
> Have a nice day :) Valentín

I can test it, just have no clue how to modify those lines...

ddelpire, 2022-10-10:
Hi, i will try tonight . thanks !

BrunoSpy, 2022-10-10:
I patched it directly into the running docker container, restarted z2m but value is still wrong.
I'm not sure if the modifications have been taken into account by z2m.

vk496, 2022-10-10:
For applying it, you must use external converters. Refer to Z2M documentation for more info

ddelpire, 2022-10-10:
Hi, how keep persistance in modifications ?
I attach the z2mqtt container with command :
docker exec -it container_number bash

so in bash i modify with vi command lixee.js file but when i restart z2mqtt plugin in HA this container is destroyed and a new container is created and i loss modification.

vk496, 2022-10-10:
You mustn't edit the files manually. Because you just modify the source files, not the compiled binary that is being run inside the container.

As I said before, use external converters strategy, described in Z2M documentation.

Have a nice day :)
Valentín

vk496, 2022-10-10:
Some resources:

https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters

https://www.reddit.com/r/homeassistant/comments/rwg1ly/comment/hrcd1no/

ddelpire, 2022-10-11:
Hi, 

this weekend i added lixee.js in HA with file editor plugin and modified in config/zigbee2mqtt directory the configuration.yaml file with  theses lines ::

external_converters:
  - lixee.js

but Z2MQTT restart failed.

ddelpire, 2022-10-11:
Or another solution but it's not easy :

- Fork zigbee-herdsman-converters and update lixee.js
- Fork Zigbee2MQTT to integrate zigbee-herdsman-converters fork
- Build a new image docker Zigbee2MQTT with theses 2 forks

vk496, 2022-10-11:
Weird, did you checked the logs of Z2M to see why it's failed fo you?

Also, @Koenkk external converters should work for devices that already exist, right? Or the precedence is not defined/unfavorable for external converters when there is already one in the core?

vk496, 2022-10-11:
> Or another solution but it's not easy :\n\nFork zigbee-herdsman-converters and update lixee.js\nFork Zigbee2MQTT to integrate zigbee-herdsman-converters fork\nBuild a new image docker Zigbee2MQTT with theses 2 forks

Yes, that's the way to setup a local development project. You will need to make sure that dependencies are pointing to your local repository fork and build them before using somewhere else.

vk496, 2022-10-12:
> Can you tell me where are locations logs ? Thanks

If you use HA, probably in the plugin logs. You may be need to increase the debug of the logs to see more. https://www.zigbee2mqtt.io/guide/configuration/logging.html

arenoux, 2022-10-12:
Hey all ;)

I opened an issue on zigbee-herdsnam-converters and suggest a pull request to fix it.
This has just been approved and merged.

So next zigbee2mqtt version will fix this issue 😇😇😇

hdeadvance01, 2022-10-13:
hello
thank you
very good news

hdeadvance01, 2022-10-13:
Hello
yes very well I am a question:
How to reset sensor when the Z2M will fix.
I want to restart with good value 
[image]

because this value is wrong
thanks

nbufnoir, 2022-10-13:
> [image] I'm sure you guys will appreciate to see this screenshot :p

Thxs a lot for your work :)

hdeadvance01, 2022-11-02:
hello
I update zigbe2mqtt with 1.28.1-1 but I dont see 0 value this morning ??
it's normal ?? I reboot home assistant.
[image]

hdeadvance01, 2022-11-02:
but this problem seems fixed. list fixed with version 1.28.1-1 : 
I wait this afternoon if there is sun!?

[image]

hdeadvance01, 2022-11-02:
hello 
thanks I wait and see

nbufnoir, 2022-11-02:
Fixed !! great, thxs a lot :)

closed 2022-11-02 by nbufnoir

hdeadvance01, 2022-11-02:
I am impatient it's good 
thanks a lot :)
[image]