Koenkk/zigbee2mqtt #13993: [New device support]: Aqara Smart Radiator Thermostat E1 (SRTS-A01)

ccorderor, 2022-09-13:
### Link

https://www.aqara.com/eu/product/radiator-thermostat-e1

### Database entry

{"id":80,"type":"EndDevice","ieeeAddr":"0x54ef441000128356","nwkAddr":12635,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.airrtc.agl001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":769,"inClusterList":[0,1,3,64704,10,513],"outClusterList":[3,64704,513],"clusters":{"genBasic":{"attributes":{"appVersion":23,"modelId":"lumi.airrtc.agl001"}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[3,40,30,5,33,1,0,10,33,75,16,13,35,23,8,0,0,17,35,1,0,0,0,101,32,0,102,41,250,10,103,41,140,10,104,35,0,0,0,0,105,32,100,106,32,0]},"625":1,"626":0,"627":0,"628":1,"629":0,"630":{"type":"Buffer","data":[4,62,1,224,0,0,9,96,4,56,0,0,6,164,5,100,0,0,8,152,129,224,0,0,8,152]},"631":0,"633":500,"634":0,"635":1,"636":0,"637":0,"638":0,"640":0,"1034":100}},"hvacThermostat":{"attributes":{"localTemp":2810,"occupiedHeatingSetpoint":2700}},"genPowerCfg":{"attributes":{"batteryVoltage":33}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":23,"hwVersion":1,"dateCode":"Apr 25 2022","swBuildId":"0.0.0_0023","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1906590817},"lastSeen":1663048175699,"defaultSendRequestWhen":"immediate"}

### Comments

The device is unsupported. I've created a basic external converter and am able to manually control the heating setpoint, but there are still many custom aqaraOpple messages to be implemented.

For the moment, I'm sure the aqaraOpple message with key id 626 is the antifreeze option (value 2 is on, value 0 is off)

### External converter

```shell
const definition = {
    zigbeeModel: ['lumi.airrtc.agl001'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SRTS-A01', // Vendor model number, look on the device for a model number
    vendor: 'Xiaomi', // Vendor of the device (only used for documentation and startup logging)
    description: 'Aqara Smart Radiator Thermostat E1', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.thermostat, fz.aqara_opple, fz.battery],
    toZigbee: [tz.thermostat_local_temperature, tz.thermostat_occupied_heating_setpoint, tz.thermostat_control_sequence_of_operation,
            tz.thermostat_system_mode, tz.thermostat_keypad_lockout, tz.thermostat_occupied_heating_setpoint], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery_voltage(), exposes.climate()
            .withSetpoint('occupied_heating_setpoint', 5, 30, 0.5, ea.STATE_SET).withLocalTemperature(ea.STATE)
            .withSystemMode(['off', 'auto', 'heat'], ea.STATE_SET)
            .withRunningState(['idle', 'heat'], ea.STATE)], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genPowerCfg', ['batteryVoltage']);
        },
};
```

kirovilya, 2022-09-13:
I have already [added thermostat converters](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/xiaomi.js#L2257), but they are not complete. you can add/correct them as needed.

ccorderor, 2022-09-13:
> I have already [added thermostat converters](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/xiaomi.js#L2257), but they are not complete. you can add/correct them as needed.

Nice work!
I'll check it to see if you have already added the antifreeze and manual option.

Thanks :)

kirovilya, 2022-09-13:
I am named them 'away_preset_temperature' and 'preset'

stefan-schoene, 2022-09-13:
According to the product page, this thermostat can be linked to an external temperature sensor in the Aqara app. Is it possible to support this as well or is this allegedly based on some proprietary Aqara implementation?

kirovilya, 2022-09-13:
Yes, there is such a possibility.
The TH-sensor sends temperature data to the hub, and then the hub sends a control command to the thermostat. Only the command and data are proprietary :)
Example:
`aa:71:13:44:04:8a:05:41:10:00:15:8d:00:01:e4:21:8a:00:01:00:55:45:23:80:00`
[image]

stefan-schoene, 2022-09-13:
Thanks! Could you explain this part a little further:
`Only the command and data are proprietary :)`
Does this mean it's theoretically possible to link to a sensor but in reality due to proprietary commands it's not?

kirovilya, 2022-09-13:
I think if understand the data format, then we can use other sensors.
This is not quite a link - this is some kind of external control based on data from the sensor.

kirovilya, 2022-09-13:
I propose to think about how to parse the commands that the gateway sends to the thermostat: their structure and meaning.
```
send command "link sensor"
aa:71:31:44:00:70:02:41:2e:63:14:57:81:3d:04:54:ef:44:10:00:51:c6:68:00:15:8d:00:01:e4:21:8a:00:01:00:55:13:0a:02:00:00:64:04:ce:c2:b6:c8:00:00:00:00:00:01:3d:64:65
aa:71:34:44:01:6c:02:41:31:63:14:57:81:3d:05:54:ef:44:10:00:51:c6:68:00:15:8d:00:01:e4:21:8a:08:00:07:fd:16:0a:02:0a:c9:e8:b1:b8:d4:da:cf:df:c0:eb:00:00:00:00:00:01:3d:04:65
received 26.09* from TH sensor then send control
aa:71:13:44:02:8c:05:41:10:00:15:8d:00:01:e4:21:8a:00:01:00:55:45:23:10:00
aa:71:13:44:03:8b:05:41:10:00:15:8d:00:01:e4:21:8a:00:01:00:55:45:23:10:00
received 26.16* then send control
aa:71:13:44:04:8a:05:41:10:00:15:8d:00:01:e4:21:8a:00:01:00:55:45:23:80:00
aa:71:13:44:05:89:05:41:10:00:15:8d:00:01:e4:21:8a:08:00:07:fd:00:00:00:01 - ?
received 26.68* then send control 
aa:71:13:44:06:88:05:41:10:00:15:8d:00:01:e4:21:8a:00:01:00:55:45:26:c0:00
received 26.71*  then send control
aa:71:13:44:07:87:05:41:10:00:15:8d:00:01:e4:21:8a:00:01:00:55:45:26:f0:00
received 27.24* then send control
aa:71:13:44:08:86:05:41:10:00:15:8d:00:01:e4:21:8a:00:01:00:55:45:2a:40:00
send command "unlink sensor"
aa:71:1d:44:09:7b:04:41:1a:63:14:59:1d:3d:05:54:ef:44:10:00:51:c6:68:00:00:00:00:00:00:00:00:00:00:00:00
```
Partially I managed to understand the composition of the package.
For example:
`aa:71:13:44:02:8c:05:41:10:00:15:8d:00:01:e4:21:8a:00:01:00:55:45:23:10:00`
`aa:71:13:44:` - ?
`02:` - subsequence
`8c:` - backwards subsequence
`05:` - ? command ?
`41:` - type octetStr
`10:` - length 16
`00:15:8d:00:01:e4:21:8a:00:01:00:55:45:23:10:00` - ?

`aa:71:31:44:00:70:02:41:2e:63:14:57:81:3d:04:54:ef:44:10:00:51:c6:68:00:15:8d:00:01:e4:21:8a:00:01:00:55:13:0a:02:00:00:64:04:ce:c2:b6:c8:00:00:00:00:00:01:3d:64:65`
`aa:71:31:44:` - ?
`00:` - subsequence
`70:` - backwards subsequence
`02:` - ? command ?
`41:` - type octetStr
`2e:` - length 46
`63:14:57:81:3d:04:54:ef:44:10:00:51:c6:68:00:15:8d:00:01:e4:21:8a:00:01:00:55:13:0a:02:00:00:64:04:ce:c2:b6:c8:00:00:00:00:00:01:3d:64:65` - ?
here the second octetStr also has a familiar value - ieee temperature sensor address `54:ef:44:10:00:51:c6:68:`

0ip, 2022-09-13:
Thank you @kirovilya for your contribution! I'm using the thermostat with the latest dev branch. 
Please excuse my question. Do we know if `local_temperature` really corresponds to the measured temperature the thermostat uses to adjust the valve or rather the internal SoC temperature or something else? I'm asking because my reading is far off from other temp sensors in the room, knowing full well that the value at the radiator can of course differ, but for me the value makes no sense. After all, my device could also be broken :D

kirovilya, 2022-09-14:
@0ip as far as I understand, these are the readings of the internal temperature sensor, according to which the thermostat is adjusted

happyhammer7, 2022-09-18:
> I have already [added thermostat converters](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/xiaomi.js#L2257), but they are not complete. you can add/correct them as needed.

Hi I am trying to setup a converter for his device and having issues. Can you check my code to see if this will work please as I keep getting syntax errors. Thank you

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.airrtc.agl001'],
    model: 'SRTS-A01',
    vendor: 'Xiaomi',
    description: 'Aqara Smart Radiator Thermostat E1',
    fromZigbee: [fzLocal.aqara_trv, fz.thermostat],
    toZigbee: [tzLocal.aqara_trv, tz.thermostat_occupied_heating_setpoint],
    exposes: [e.switch().setAccess('state', ea.STATE_SET),
        exposes.climate().withSetpoint('occupied_heating_setpoint', 5, 30, 0.5)
            .withLocalTemperature(ea.STATE).withPreset(['manual', 'away', 'auto'], ea.STATE_SET),
        e.child_lock(), e.window_detection(), e.valve_detection(),
        e.away_preset_temperature(),
    },
};

module.exports = definition;`

trullock, 2022-09-20:
Just checking in to request this device be added. I'm happy to help test things - how do I switch my installation to the dev branch without losing all my existing devices/config? cheers

stefan-schoene, 2022-09-24:
Any news on this? I just received and installed my thermostat as well,  so I'd also help if possible/necessary.

lunatik98, 2022-09-27:
Thanks for your work.
Currently turning the device on or off from Home Assistant does not actually turn on or off the thermostat in Z2M. But it works fine from Z2M.

lunatik98, 2022-09-27:
> Just checking in to request this device be added. I'm happy to help test things - how do I switch my HA addon installation to the dev branch without losing all my existing devices/config? cheers

Just stop the normal Z2M, copy the settings from the addon to the dev version and start it :)

trullock, 2022-09-30:
@lunatik98 sorry, which files exactly need putting where? Cheers

trullock, 2022-10-01:
One issue I've found is that when you change the set point to either above or below the current temperature, and you hear the physical valve open/close, the valve state switch in HA doesn't update

trullock, 2022-10-02:
Voltage and LinkQuality also dont show up in HA

artem-sedykh, 2022-10-03:
`[aa:71]:[13]:44:2b:63:05:41:10:[00:15:8d:00:04:44:c8:24]:00:01:00:55:[45:19:60:00]`  
`[aa:71] ` &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;: const  
`[13] ` &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;: data offset  
`[00:15:8d:00:04:44:c8:24]`&nbsp;: zbee_nwk.src64(temperature sensor)
`[45:19:60:00]  ` &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;: temperature type float

artem-sedykh, 2022-10-03:
[image]
[Frame Length 79.txt](https://github.com/Koenkk/zigbee2mqtt/files/9698570/Frame.Length.79.txt)

trullock, 2022-10-03:
Not sure if this is useful, here's an unhandled message (247/0xf7) that I receive

```
 2022-10-03 20:30:42Received Zigbee message from 'Radiator', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[3,40,21,5,33,1,0,10,33,21,178,13,35,25,8,0,0,17,35,1,0,0,0,101,32,0,102,41,108,7,103,41,96,9,104,35,0,0,0,0,105,32,100,106,32,0],"type":"Buffer"}}' from endpoint 1 with groupID 0
```

artem-sedykh, 2022-10-03:
send command "link sensor"  

| const | data offset | const | subsequence | backwards subsequence | command | const | data length | data block                                                                                                                                                                                                                                                                                                                                                     |
|-------|-------------|-------|-------------|-----------------------|---------|-------|-------------|----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|
| aa:71 | 31          | 44    | 63          | 0d                    | 02      | 41    | 2e          | <table><tr><th>const</th><th>???</th><th>const</th><th>device address</th><th>sensor address</th><th>const</th><th>const</th></tr><tr><td>63</td><td>3b:38:fd</td><td>3d:04</td><td>54:ef:44:10:00:51:cf:e6</td><td>00:15:8d:00:04:44:c8:24</td><td>00:01:00:55</td><td>13:0a:02:00:00:64:04:ce:c2:b6:c8:00:00:00:00:00:01:3d:64:65</td></tr></table>          |
| aa:71 | 34          | 44    | 64          | 09                    | 02      | 41    | 31          | <table><tr><th>const</th><th>???</th><th>const</th><th>device address</th><th>sensor address</th><th>const</th><th>const</th></tr><tr><td>63</td><td>3b:38:fd</td><td>3d:05</td><td>54:ef:44:10:00:51:cf:e6</td><td>00:15:8d:00:04:44:c8:24</td><td>08:00:07:fd</td><td>16:0a:02:0a:c9:e8:b1:b8:d4:da:cf:df:c0:eb:00:00:00:00:00:01:3d:04:65</td></tr></table> |
| aa:71 | 12          | 44    | 65          | 29                    | 05      | 41    | 10          | <table><thead><tr><th>sensor address</th><th>const</th><th>temperature(float)</th></tr></thead><tbody><tr><td>00:15:8d:00:04:44:c8:24</td><td>00:01:00:55</td><td>45:1a:b0:00</td></tr></tbody></table>                                                                                                                                                        |

nickagian, 2022-10-04:
> One issue I've found is that when you change the set point to either above or below the current temperature, and you hear the physical valve open/close, the valve state switch doesn't update, meaning you cant know if its open or not (without making an assumption from the temperature difference)

May I ask which topic in MQTT you are using to check if the valve is open or closed? I don't seem to find this feature. I can only find the switch "Valve_detection", but I guess this is to enable/disable the feature of reading the valve state.

Is this generally supported by the device, to see valve state?

trullock, 2022-10-04:
> May I ask which topic in MQTT you are using to check if the valve is open or closed? I don't seem to find this feature. I can only find the switch "Valve_detection", but I guess this is to [...]

I'm not sure if this is a feature, but I'd expect it to be. I'm expecting the on/off switch to have its state updated if the valve determines it should be on/off automatically or through physical adjustment of the device. Is this not the case?

Valve Detection seems to be a badly named feature, and is really something else:

> You can enable/disable the Valve Detection function on the settings page. In order to prevent the prolonged low-temperature or high-temperature state of room temperature from becoming uncomfortable, the thermostat will remind the user of temperature control abnormality when it detects that the room temperature can’t reach the set temperature, after heating is started, for the following possible causes:
The thermostat is not installed correctly. It is suggested that you should remove the battery and put it back 30 sec later, then reinstall the thermostat and calibrate it again, and further wait for a period of time, and then observe whether the fault is eliminated.
The room temperature is significantly higher than the set temperature due to the incorrect temperature setting. It is suggested that you should re-set the temperature again, wait for a period of time, and then observe whether the fault is eliminated.
Valve failure. Please replace the valve, reinstall the thermostat, and carry out valve calibration.
The room is too large (the radiator is not matching) or there is too much ventilation in the room. It is suggested that you should replace the radiator or test whether the room door or window is opened or leaking air.
The inlet water pressure or inlet water temperature is too low. Please check the inlet water pressure and inlet water temperature.
Upon networking with the temperature sensors, no linkage is established with the temperature sensor in the same room, or the distance from the temperature sensor to the radiator is remote (it is recommended that the temperature sensor should be placed in the same room with the thermostat, for a distance of 2-3m from the thermostat).

stefan-schoene, 2022-10-05:
> Valve Detection seems to be a badly named feature, and is really something else:
> 
> 
> 
> > You can enable/disable the Valve Detection function on the settings page. In order to prevent the prolonged low-temperature or high-temperature state of room temperature from becoming uncomfortable, the thermostat will remind the user of temperature control abnormality when it detects that the room temperature can’t reach the set temperature, after heating is started, for the following possible causes:
> 
> The thermostat is not installed correctly. It is suggested that you should remove the battery and put it back 30 sec later, then reinstall the thermostat and calibrate it again, and further wait for a period of time, and then observe whether the fault is eliminated.
> 
> The room temperature is significantly higher than the set temperature due to the incorrect temperature setting. It is suggested that you should re-set the temperature again, wait for a period of time, and then observe whether the fault is eliminated.
> 
> Valve failure. Please replace the valve, reinstall the thermostat, and carry out valve calibration.
> 
> The room is too large (the radiator is not matching) or there is too much ventilation in the room. It is suggested that you should replace the radiator or test whether the room door or window is opened or leaking air.
> 
> The inlet water pressure or inlet water temperature is too low. Please check the inlet water pressure and inlet water temperature.
> 
> Upon networking with the temperature sensors, no linkage is established with the temperature sensor in the same room, or the distance from the temperature sensor to the radiator is remote (it is recommended that the temperature sensor should be placed in the same room with the thermostat, for a distance of 2-3m from the thermostat).
> 
> 
> 
> It might be helpful to either alude to this in the UI or otherwise completely rename it to something sensible

This function can show an icon on the thermostat screen in case one of the conditions above fires. The manual refers to this as "fault icon". So maybe this function / switch should rather be called something like "fault detection" in combination with further documentation (e.g. the conditions above) and that this enables this icon on the screen.

trullock, 2022-10-05:
> This function can show an icon on the thermostat screen in case one of the conditions above fires. The manual refers to this as "fault icon". So maybe this function / switch should rather be called [...]

I think if anything is done here it should be additive, e.g. call it "Fault (Valve) Detection", as renaming it alone creates a mismatch between feature names which is arguably more confusing - Whos to know theyre the same feature and not different features?

nickagian, 2022-10-05:
> > > One issue I've found is that when you change the set point to either above or below the current temperature, and you hear the physical valve open/close, the valve state switch doesn't update, meaning you cant know if its open or not (without making an assumption from the temperature difference)
> > 
> > 
> > May I ask which topic in MQTT you are using to check if the valve is open or closed? I don't seem to find this feature. I can only find the switch "Valve_detection", but I guess this is to enable/disable the feature of reading the valve state.
> > Is this generally supported by the device, to see valve state?
> 
> I'm not sure if this is a feature, but I'd expect it to be. I'm expecting the on/off switch to have its state updated if the valve determines it should be on/off automatically or through physical adjustment of the device. Is this not the case?
> 
> Valve Detection seems to be a badly named feature, and is really something else:
> 
> > You can enable/disable the Valve Detection function on the settings page. In order to prevent the prolonged low-temperature or high-temperature state of room temperature from becoming uncomfortable, the thermostat will remind the user of temperature control abnormality when it detects that the room temperature can’t reach the set temperature, after heating is started, for the following possible causes:
> > The thermostat is not installed correctly. It is suggested that you should remove the battery and put it back 30 sec later, then reinstall the thermostat and calibrate it again, and further wait for a period of time, and then observe whether the fault is eliminated.
> > The room temperature is significantly higher than the set temperature due to the incorrect temperature setting. It is suggested that you should re-set the temperature again, wait for a period of time, and then observe whether the fault is eliminated.
> > Valve failure. Please replace the valve, reinstall the thermostat, and carry out valve calibration.
> > The room is too large (the radiator is not matching) or there is too much ventilation in the room. It is suggested that you should replace the radiator or test whether the room door or window is opened or leaking air.
> > The inlet water pressure or inlet water temperature is too low. Please check the inlet water pressure and inlet water temperature.
> > Upon networking with the temperature sensors, no linkage is established with the temperature sensor in the same room, or the distance from the temperature sensor to the radiator is remote (it is recommended that the temperature sensor should be placed in the same room with the thermostat, for a distance of 2-3m from the thermostat).
> 
> It might be helpful to either alude to this in the UI or otherwise completely rename it to something sensible

I definitely agree with you. Something like is expected to be present from a smart thermostat, otherwise how can you monitor when it is operating or not?

artem-sedykh, 2022-10-09:
[image]

0ip, 2022-10-09:
Now that we know how the linking procedure to an external temperature sensor works, thanks to @duvholt and others, should/can it be integrated into Z2M, or should that be left to applications built on top of Z2M?

nickagian, 2022-10-09:
@0ip I would say that Z2M should expose this feature through MQTT topics. Isn't that common practice?

gdschut, 2022-10-09:
I have this TRV only reporting local_temperature at steps of 1.0 degree as 18.2, 19.2, 20.2 etc. Is this normal?

artem-sedykh, 2022-10-09:
> Now that we know how the linking procedure to an external temperature sensor works, thanks to @duvholt and others, should/can it be integrated into Z2M, or should that be left to applications built on top of Z2M?

I added the ability to switch to an external and internal temperature sensor, as well as sending temperature from an external sensor [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4754)

artem-sedykh, 2022-10-09:
**attribute: 0x00f7**
03:28:[1b]:05:21:01:00:0a:21:00:00:0d:23:19:08:00:00:11:23:01:00:00:00:65:20:00:66:29:[f2:08]:67:29:[2e:09]:68:23:[01:00:00:00]:69:20:64:6a:20:00

[1b] - ?
[13:01] - local_temperature type int16
[2e:09] - occupied_heating_setpoint type int16
[01:00:00:00] - valve_alarm 01:00:00:00 or 00:00:00:00 type float

duvholt, 2022-10-09:
Some notes about **0x00f7**:

It looks like this is a shared attribute that several Xiaomi devices use. 
It can be decoded using `buffer2DataObject` from `lib/xiaomi.js` which produces data like this:
```js
{
  '3': 30,
  '5': 1,
  '10': 56730,
  '13': 2073,
  '17': 1,
  '101': 0,
  '102': 2400,
  '103': 2550,
  '104': 1,
  '105': 100,
  '106': 0
}
```
Then there's a shared Xiaomi function to turn this into a payload: `numericAttributes2Payload`, but currently it seems like the only overlap for this device is `device_temperature` and `power_outage_count`.

After some trial and error this is what I've gathered so far:
**3**: Likely device temperature as described in `numericAttributes2Payload`. In my samples this is a value between 24 and 30. 
**5**: Likely power_outage_count. In all of my samples except one this is reported as 1.
**10**: Usually 56730 but I've 0 seen a couple of times.
**13**: ? Always 2073
**17**: ? Always 1
**101**: Preset (0x0272). 2 away, 1 auto, 0 manual
**102**: Local temp
**103**: Current setpoint
**104**: In my notes I wrote that it might be valve detection but valve alarm seems more likely. 
**105**:  ? Always 100
**106**:  ? Always 0

trullock, 2022-10-09:
> Some notes about **0x00f7**:
>
> It looks like this is a shared attribute that several Xiaomi devices use. It can be decoded using `buffer2DataObject` from `lib/xiaomi.js` which produces data like [...]

Any sign of a flag for the state of the valve?

duvholt, 2022-10-09:
> Any sign of a flag for the state of the valve?

No, doesn't look like the device reports any changes when adjusting the valve. I don't see any info about this in the Aqara Home app either.

duvholt, 2022-10-09:
I created a PR to improve the Home Assistant integration: Koenkk/zigbee-herdsman-converters/pull/4760

trullock, 2022-10-09:
> > Any sign of a flag for the state of the valve?
> 
> No, doesn't look like the device reports any changes when adjusting the valve. I don't see any info about this in the Aqara Home app either.

Does your PR fix this?

artem-sedykh, 2022-10-09:
Added scheduling support [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/4763)

duvholt, 2022-10-10:
> > > Any sign of a flag for the state of the valve?
> > 
> > 
> > No, doesn't look like the device reports any changes when adjusting the valve. I don't see any info about this in the Aqara Home app either.
> 
> Does your PR fix this?

No, not really. I just displays on/off (or actually heat/off) status in Home Assistant instead of always showing the device as off. 
Heat doesn't actually mean "is currently heating", but that the device will heat if below the current setpoint. 
Unless I've misunderstood, what you are asking for is RunningState which reports if the thermostat is currently heating or not and so far it doesn't seem like this device supports that.

trullock, 2022-10-10:
> what you are asking for is RunningState which reports if the thermostat is currently heating or not and so far it doesn't seem like this device supports that.

Good to know the proper terminology.

Surely it supports this? Glad I spent £50 on it.....

This can be inferred though right, as the valve opens when its >1 degree below the set point?

0ip, 2022-10-10:
Has anyone observed the time intervals at which the TRV synchronizes with an external T sensor? What do you think would be a reasonable interval?

IvanUpa, 2022-10-11:
Tell me how to add an external sensor? I updated the edge version, the switch internal\external appeared, but I did not find where to select the sensor.

gdschut, 2022-10-14:
> > I have this TRV only reporting local_temperature at steps of 1.0
> 
> The same happens here. I have two of them, and one is always ending with x.2°C and the other always with x.9°C
> 
> Here is a graph from Home Assistant with the temperature from both radiators from last week: [image] You see that is will always steps with a full degree.

Does anybody know if this happens due to how the TRV works? Or is it caused by the zigbee2mqtt implementation? Or perhaps because of the use of conbee II instead of the Aqara zigbee hub?

ooii, 2022-10-14:
> Has anyone observed the time intervals at which the TRV synchronizes with an external T sensor? What do you think would be a reasonable interval?

From what you say in your [next message](https://github.com/Koenkk/zigbee2mqtt/issues/13993#issuecomment-1274277630), I understand that you decide at which interval you push the information to the valve. That can be done each time the sensor reports a new temperature (which is dependent on the reporting threshold of the sensor).

0ip, 2022-10-14:
@ooii Initially, for inspiration, I would have liked to know how Aqara handles the timing of the synchronization. I had already thought to myself it may not be a good idea to publish every new measurement, if the reporting interval is rather short, to avoid unnecessary load on the network and increased power consumption for the TRV, handling all those messages.

My current procedure is to report only sufficiently large temperature changes immediately, but at most one minute apart, and to send the current temperature after one hour at the latest, even if it is unchanged. The latter may be unnecessary if there is no mechanism in the TRV to discard temperature reports that are too old. But we don't know that yet, I suppose.

duvholt, 2022-10-14:
From what I could gather Aqara just syncs the sensor every time a new temperature is reported. And since Aqara temperature sensors only report new values when there is a temperature/humidity/pressure change I don't think there's any mechanism to consider sensor values old. From what I can remember my TRV never reverted back to the local tempeature even when I wasn't sending any sensor updates for several hours.

Regarding power consumption I'm not sure yet how much it matters. All my TRVs report 100% battery in both Aqara Home and z2m, but I don't really trust that the battery info is correct. Would probably be interesting to test out different automations over time and see if we can notice a difference.

MrCamby, 2022-10-16:
Maybe I missed some important informations, but am I right that there is currently no way how I can combine the Aqara temperature and humidity sensor with the Aqara Thermostate e1? I thought it's possible to bind them together.

IvanUpa, 2022-10-16:
> Maybe I missed some important informations, but am I right that there is currently no way how I can combine the Aqara temperature and humidity sensor with the Aqara Thermostate e1? I thought it's possible to bind them together.

It is possible, with any temperature sensor. Just publishing a value in `{"Sensor_TEMP": <T ° C Float>}`

MrCamby, 2022-10-16:
Something like this? Or inside z2m?
[image]

IvanUpa, 2022-10-16:
No.
Topic: zigbee2mqtt/TRV_Hall/set/sensor_temp
Payload: 21.2

MrCamby, 2022-10-16:
@IvanUpa thank you for helping so far but I seem to still be a noob. After fixing an spelling error (IPhone adding an space at the end) I now get the following error
```
Zigbee2MQTT:error 2022-10-16 18:26:20: No converter available for 'sensor_temp' (30)
```

IvanUpa, 2022-10-16:
Like this:
[image]

MrCamby, 2022-10-16:
I´m still having the same error `No converter available for 'sensor_temp' (30)`
[image]
Do I need the external converter cause I tried setting it up, but I have no clue how without getting any errors.

Blayven, 2022-10-16:
You need to either install Zigbee2mqt Edge instead of Zigbee2MQTT or wait for Zigbee2MQTT to upgrade in November. Next, switch Sensor TRV to External.

Blayven, 2022-10-16:
Of course.
1. Stop Zigbee2MQTT
2. Install Zigbee2MQTT Edge
3. Copy a configuration from Zigbee2MQTT to Zigbee2MQT Edge
4) Launch Zigbee2MQTT Edge

MrCamby, 2022-10-16:
Finally! Thank you @Blayven for the troubleshooting!

Jtef, 2022-10-17:
Tested on Edge version and it works as desired. Thank you guys, you are awesome!

stefan-schoene, 2022-10-17:
I'm gonna wait on the public release but I'm just wondering, how often are you guys broadcasting new values? Every time the value changes more than x°C or just every x minutes?

MrCamby, 2022-10-17:
Currently I do update every little temp change. But I will change that sooner or later.

[185 more replies]

closed 2023-07-31 by github-actions[bot]