Koenkk/zigbee2mqtt #13999: ZIGBEE 3.0 USB DONGLE PLUS issue with z2m

lazarvgd, 2022-09-13:
### What happened?

I recently bought two devices ZIGBEE 3.0 USB DONGLE PLUS, from different seller. Fun part is that one is longer than another. I red the documentation and found that this devices should be supported.

The problem:
I run into the problem where z2m logs report that z2m cannot communicate with the device:

```
Zigbee2MQTT:error 2022-09-13 11:31:24: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyUSB0'
    at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
    at SerialPort._error (/app/node_modules/@serialport/stream/lib/index.js:198:14)
    at /app/node_modules/@serialport/stream/lib/index.js:242:12
```

I managed to flash bigger dongle with coordinator frimware and I was able to connect with z2m, but unfortunately every zigbee device that I have in the house can't pair or if it is paired it cannot be controlled.

Does anybody have any idea what is going on?
Can somebody help me to fix this?

thanks 

[image]

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20220219.hex

### Adapter

Texas Instruments CC2652/CC1352 chip

banditto9, 2022-09-13:
Let me put in a word, but I can be wrong. Probably the another stick (flashed as coordinator) is not empty - it seems to have something written in the NVRAM. In order to get config from the z2m it should have empty NVRAM AFAIK.  Try doing a backup first from both of your sticks (just in case) and clean an NVRAM for the one you are going to use. ZigStarGW-MT is the tool I do use for flashing sticks. You can even transfer NVRAM from one to another by doing a read to a file & write that file to another stick. Just be very careful not to erase the working one - do backups of both with friendly naming. The left one seems to be from Sonoff (P model) and it works fine both as coordinator and as a router as well - I use them and happy with the nice look and black metal finish.

As to the latest z2m I have another but similar issue when 2652P can't be found after restart - it just continuously searching for it. This happens not every time but it happens. Physical eject and insert works fine but for me this is a remote site and physical presence is not possible sometimes. I hope the next release will fix this somehow as 'absent' coordinator just make the whole zigbee network useless when there is no 'connect' with the coordinator. I'm afraid now of the future remote updates.

kirovilya, 2022-09-13:
@lazarvgd set `adapter: ezsp` in configuration for using ZBDongle-E

lazarvgd, 2022-09-14:
> @lazarvgd set `adapter: ezsp` in configuration for using ZBDongle-E

Thanks Ilija, I managed to make smaller stick to work, but the problem with pairing with devices still remains. Do you have any idea how can I fix this? Thanks for the response :)

lazarvgd, 2022-09-14:
> The shorter stick is the new "ZBDongle-_**E**_" based on the SI Labs EFR32MG21 chipset. It uses a different firmware.

I spent a week trying to fix this, I will try to update the firmware, thanks :)

lazarvgd, 2022-09-14:
> Let me put in a word, but I can be wrong. Probably the another stick (flashed as coordinator) is not empty - it seems to have something written in the NVRAM. In order to get config from the z2m it [...]

After setting the flag that Ilija mentioned I made smaller stick to work with z2m, but the problem still remains with pairing, it is strange. Do you maybe know why is this happening?

Thank you for your time and effort :) 

P.S. Will try to empty both sticks and install appropriate firmware.

lazarvgd, 2022-09-15:
I did not make these two devices to work with z2m, but I managed to setup them with zha and I can control devices.

It would be awesome if z2m can make these two adapters to work.

thanks for help :)

closed 2022-09-15 by lazarvgd