Koenkk/zigbee2mqtt #14001: Error during instalation - npm ci

krajewmar, 2022-09-13:
### What happened?

A lot of errors during instalation, see log. I programmed firmware CC2531_DEFAULT_20211115.zip I don't know exactyl Zigbee2MQTT version due to I'm unable to finish instalation without error.  It was cloned from githib today.

### How to reproduce it (minimal and precise)

I go simple trought install guide on this page. Adapter is seen as
usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B001CDD3214-if00 -> ../../ttyACM0
other
pi@rPi4BAP:/opt/zigbee2mqtt $ node --version
v18.9.0
pi@rPi4BAP:/opt/zigbee2mqtt $ npm --version
8.19.1

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20211115

### Adapter

CC2531 usb stick

### Debug log

[2022-09-13T14_44_12_596Z-debug-0.log](https://github.com/Koenkk/zigbee2mqtt/files/9558201/2022-09-13T14_44_12_596Z-debug-0.log)

krajewmar, 2022-09-13:
in current state, running Zigbee2mqtt is causing error
pi@rPi4BAP:~ $ cd /opt/zigbee2mqtt/
pi@rPi4BAP:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.27.2 start
> node index.js

node:internal/modules/cjs/loader:958
  throw err;
  ^

Error: Cannot find module 'core-js/features/object/from-entries'
Require stack:
- /opt/zigbee2mqtt/index.js
    at Module._resolveFilename (node:internal/modules/cjs/loader:955:15)
    at Module._load (node:internal/modules/cjs/loader:803:27)
    at Module.require (node:internal/modules/cjs/loader:1021:19)
    at require (node:internal/modules/cjs/helpers:103:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/index.js:1:1)
    at Module._compile (node:internal/modules/cjs/loader:1119:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1173:10)
    at Module.load (node:internal/modules/cjs/loader:997:32)
    at Module._load (node:internal/modules/cjs/loader:838:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:81:12) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ '/opt/zigbee2mqtt/index.js' ]
}

Node.js v18.9.0

Koenkk, 2022-09-14:
What is the output of `npm ci`? (executed in `/opt/zigbee2mqtt`)

krajewmar, 2022-09-14:
Output in file. I've added some message translation after //// 

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9569201/log.txt)

Koenkk, 2022-09-15:
Can you try downgrading your node version to 16 (from 18)

krajewmar, 2022-09-15:
After downgrade, same issue happen, as I think. Log in file
[log_16x.txt](https://github.com/Koenkk/zigbee2mqtt/files/9573272/log_16x.txt)

krajewmar, 2022-09-16:
Instllation on fresh rpi went fine.

Koenkk, 2022-09-19:
Ok great, not sure what the problem was but it seems system related.

closed 2022-09-19 by Koenkk