Koenkk/zigbee2mqtt #14003: State of IKEA lights remains OFF

paulz-gh, 2022-09-14:
### What happened?

Hi I recently notice the state of some/most of my IKEA lights remains OFF in the dashboard even when they are on.
The on/off switch in the dashboard or in the exposes page also remains off.
The switch does work and the lights do switch on/off in response to commands it seems it's just the state that's incorrect.

Looked back in the log and as far back as Sept 6 I could not find a line with state ON for this light.

I updated from 1.27.0 to 1.27.2 this morning but that does not resolve this issue. 

This happens for several different types of IKEA lights, both GU10 and E27.

I have several other devices like Aqara plugs and they don't show this problem.

Any idea
 Paul

### What did you expect to happen?

State to be shown correctly

### How to reproduce it (minimal and precise)

This is a snip from the log :
```text
Sep 14 08:29:17 pi4 npm[24393]: Zigbee2MQTT:debug 2022-09-14 08:29:17: Received MQTT message on 'zigbee2mqtt/LHalHuiskamer/set' with data '{"state":"ON"}'
Sep 14 08:29:17 pi4 npm[24393]: Zigbee2MQTT:debug 2022-09-14 08:29:17: Publishing 'set' 'state' to 'LHalHuiskamer'
Sep 14 08:29:17 pi4 npm[24393]: Zigbee2MQTT:info  2022-09-14 08:29:17: MQTT publish: topic 'zigbee2mqtt/LHalHuiskamer', payload '{"brightness":254,"last_seen":1663136957632,"level_config":{"on_level":"previous"},"linkquality":54,"power_on_behavior":"on","state":"OFF","update":{"state":"idle"},"update_available":false}'
```

State in the State tab for this light shows:
```json
{
    "brightness": 254,
    "last_seen": 1663138290027,
    "level_config": {
        "on_level": "previous"
    },
    "linkquality": 14,
    "power_on_behavior": "on",
    "state": "OFF",
    "update": {
        "state": "idle"
    },
    "update_available": false
}
```

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

Sonoff

### Debug log

Switched ON and Off LHalHuiskamer (TRADFRI bulb GU10 W 400lm) 

```
Sep 14 08:58:53 pi4 npm[24393]: Zigbee2MQTT:debug 2022-09-14 08:58:53: Received MQTT message on 'zigbee2mqtt/LHalHuiskamer/set' with data '{"state":"ON"}'
Sep 14 08:58:53 pi4 npm[24393]: Zigbee2MQTT:debug 2022-09-14 08:58:53: Publishing 'set' 'state' to 'LHalHuiskamer'
Sep 14 08:58:53 pi4 npm[24393]: Zigbee2MQTT:info  2022-09-14 08:58:53: MQTT publish: topic 'zigbee2mqtt/LHalHuiskamer', payload '{"brightness":254,"last_seen":1663138733129,"level_config":{"on_level":"previous"},"linkquality":29,"power_on_behavior":"on","state":"OFF","update":{"state":"idle"},"update_available":false}'
Sep 14 08:58:54 pi4 npm[24393]: Zigbee2MQTT:debug 2022-09-14 08:58:54: Received Zigbee message from 'OKoffie', type 'attributeReport', cluster 'aqaraOpple', data '{"229":{"data":[22,0,17,115,181,5,111,2,93,18,89,158,77,18,76,116,67,18,151,67,192,18,104,215,178,37,58,241,77,37,31,210,58,37,126,73,146,37,50,231,219,37,36,51,41,37,82,83,168,37,110,200,12,37,125,143,110,37,54,56,152,37,49,49,28,37,93,0,0,36,35],"type":"Buffer"}}' from endpoint 1 with groupID 0
Sep 14 08:58:54 pi4 npm[24393]: [85B blob data]
Sep 14 08:58:54 pi4 npm[24393]: [91B blob data]
Sep 14 08:58:54 pi4 npm[24393]: $#
Sep 14 08:58:54 pi4 npm[24393]: Zigbee2MQTT:debug 2022-09-14 08:58:54: lumi.plug.mmeu01: Processed data into payload {}
Sep 14 08:58:54 pi4 npm[24393]: Zigbee2MQTT:info  2022-09-14 08:58:54: MQTT publish: topic 'zigbee2mqtt/OKoffie', payload '{"auto_off":true,"consumer_connected":true,"consumption":16.59,"current":0.0026,"device_temperature":23,"energy":16.59,"illuminance":0,"illuminance_lux":0,"last_seen":1663138734392,"led_disabled_night":false,"linkquality":120,"overload_protection":2300,"power":0.63,"power_outage_count":3,"power_outage_memory":true,"state":"ON","temperature":20,"update":{"state":"idle"},"update_available":false,"voltage":239}'
Sep 14 08:58:54 pi4 npm[24393]: Zigbee2MQTT:debug 2022-09-14 08:58:54: Received MQTT message on 'zigbee2mqtt/LHalHuiskamer/set' with data '{"state":"OFF"}'
Sep 14 08:58:54 pi4 npm[24393]: Zigbee2MQTT:debug 2022-09-14 08:58:54: Publishing 'set' 'state' to 'LHalHuiskamer'
Sep 14 08:58:54 pi4 npm[24393]: Zigbee2MQTT:info  2022-09-14 08:58:54: MQTT publish: topic 'zigbee2mqtt/LHalHuiskamer', payload '{"brightness":254,"last_seen":1663138734839,"level_config":{"on_level":"previous"},"linkquality":32,"power_on_behavior":"on","state":"OFF","update":{"state":"idle"},"update_available":false}'
```

paulz-gh, 2022-09-14:
I just noticed that when I click on the button with the arrows left of the on/off switch the state does update correctly.

paulz-gh, 2022-09-14:
Looks like this might have been caused by setting Optimistic = false. Is that possible ?

Koenkk, 2022-09-15:
Yes, with optimistic == false you rely on state reporting of the device.

closed 2022-09-15 by paulz-gh