Koenkk/zigbee2mqtt #14005: [New device support]: Bosch Radiator thermostat II BTH-RA

schluese, 2022-09-14:
### Link

https://www.bosch-smarthome.com/at/de/produkte/geraete/heizkoerper-thermostat/

### Database entry

{"id":26,"type":"Unknown","ieeeAddr":"0xc4988600000b5098","nwkAddr":22537,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1663142316635,"defaultSendRequestWhen":"immediate"}

### Comments

For this new Bosch Radiator thermostat it seems no support yet

danieledwardgeorgehitchcock, 2022-09-15:
Hi,

You might want to check out #13780 - Some new support has been added to the Development build of Z2M to allow for device install codes which are required for binding Bosch devices correctly.

Once you have added the install code, and re-added the device, you should get a better database entry to post in here

KRGhost, 2022-10-08:
Here is the database entry after adding the install code
`{
   "id":2,
   "type":"EndDevice",
   "ieeeAddr":"0xc4988600000b8b5c",
   "nwkAddr":14692,
   "manufId":4617,
   "manufName":"BOSCH",
   "powerSource":"Battery",
   "modelId":"RBSH-TRV0-ZB-EU",
   "epList":[
      1
   ],
   "endpoints":{
      "1":{
         "profId":260,
         "epId":1,
         "devId":769,
         "inClusterList":[
            0,
            1,
            3,
            4,
            32,
            513,
            516,
            2821
         ],
         "outClusterList":[
            10,
            25
         ],
         "clusters":{
            "genPollCtrl":{
               "attributes":{
                  "checkinInterval":6480
               }
            }
         },
         "binds":[
            {
               "cluster":32,
               "type":"endpoint",
               "deviceIeeeAddress":"0x00124b0026b72652",
               "endpointID":1
            }
         ],
         "configuredReportings":[
            
         ],
         "meta":{
            
         }
      }
   },
   "appVersion":1,
   "stackVersion":5,
   "hwVersion":1,
   "dateCode":"20220627",
   "swBuildId":"3.02.05",
   "zclVersion":8,
   "interviewCompleted":true,
   "meta":{
      
   },
   "lastSeen":1665238528218,
   "defaultSendRequestWhen":"active"
}`

itkama, 2022-10-25:
@danieledwardgeorgehitchcock Are there any more infos one can gather? I just got this thermostat today and was able to successfully pair it via the dev branch, but could not do much more, since the device is unknown for now. I tried doing an external converter file - but the device does not seem to be sending any info I could work with.

itkama, 2022-11-11:
I have some "progress" I think. 
Debug 2022-11-11 16:44:42Received Zigbee message from 'Manuel Thermostat', type 'attributeReport', cluster 'hvacThermostat', data '{"localTemp":1970}' from endpoint 1 with groupID 0

If 1970 means 19,70°C (which could be about right) I just managed to get a temperature reading. I just don't know how 😅

mmattel, 2022-11-27:
I think this can be closed as implemented, see referenced issue including the reference to the integration PR https://github.com/Koenkk/zigbee2mqtt/issues/14926

closed 2022-11-27 by Koenkk