Koenkk/zigbee2mqtt #14015: zigbee2mqtt/FRIENDLY_NAME/get - No converter available for 'state' ("")

bakman2, 2022-09-15:
### What happened?

Getting the state of a device results in error.

Publish 
topic: `zigbee2mqtt/<FRIENDLY_NAME>/get`  (`zigbee2mqtt/rooms/hal/switch/deurbel/get`)
payload: `{"state": ""}`

Result:

```
No converter available for 'state' ("")
```

Are there any prerequisites for getting the state ? 

### What did you expect to happen?

I expected the same output as is shown in the front-end exposes/state tab:

```
{
    "battery": 53,
    "device_temperature": 21,
    "last_seen": "2022-09-15T04:38:21.035Z",
    "linkquality": 83,
    "power_outage_count": 26,
    "voltage": 3035
}
```

### How to reproduce it (minimal and precise)

Publish 
topic: `zigbee2mqtt/<FRIENDLY_NAME>/get`  (`zigbee2mqtt/rooms/hal/switch/deurbel/get`)
payload: `{"state":''}`

### Zigbee2MQTT version

1.27.0 commit: a9b88082

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 Plus

### Debug log

Same output as above:

```
No converter available for 'state' ("")
```

sjorge, 2022-09-15:
Does the device have a `state` attribute? You can only get attribute for which the device has a toZigbee convertor.
So you for example probably can't grab e.g. temperature for a light bulb.

Also of note, not all attributes support getting. You can usually spot those that do because they have the blue refresh icon on the frontend.

Let's look at one of my TRV:
[image]

Most of these attribute support because they have the blue icon, but linkquality and battery do not.

But then if you compare this to one of my temperature sensor, it doesn't support get at all
[image]

bakman2, 2022-09-15:
> Most of these attribute support because they have the blue icon, but linkquality and battery do not.
Indeed, there are no blue icons, i understand you can (usually) not get the state of battery powered devices. 

> But then if you compare this to one of my temperature sensor, it doesn't support get at all

But yet, the front-end can display the values, these are the values I want to get as well (even if they are not realtime). 

Those values are also in the "state" tab. Where does it get it from ? I assume it updates it everytime a value changes, these values should be exposed on some topic as well (maybe together with the `'last_seen` property).

sjorge, 2022-09-15:
> Indeed, there are no blue icons, i understand you can (usually) not get the state of battery powered devices.

Then you can't use a `/get` request over mqtt to retrieve the data as that is not what that is for.

> But yet, the front-end does display the values, these are the values I want to get as well (even if they are not realtime).

You can get this data another way, but note that the wsapi and mqtt don't work in exactly the same way, for wsapi you can just get the device path and it returns the cached data (if any). This obviously not how mqtt works, but if you enabled `retain` the data gets published on subscribe to a topic.

So if retain is enabled you can dynamically subscribe and unsubscribe to the device topic `zigbee2mqtt/device` to trigger the broker to send the retained information. (newest node-red's mqtt node supports dynamic subscriptions)

bakman2, 2022-09-16:
> You can get this data another way, but note that the wsapi and mqtt don't work in exactly the same way, for wsapi you can just get the device path and it returns the cached data (if any). 

Is there any documentation about the wsapi (or i do i look over it ?).

> So if retain is enabled you can dynamically subscribe and unsubscribe to the device topic `zigbee2mqtt/device` to trigger the broker to send the retained information. (newest node-red's mqtt node supports dynamic subscriptions)

I am not a fan of retain as it can trigger unexpected automations when i restart things, so i disable it everywhere i can.

Wouldn't it be an idea to make the data shown in the state tab available on some specific mqtt topic/endpoint ?

sjorge, 2022-09-16:
> Is there any documentation about the wsapi (or i do i look over it ?).
I was just seeing what the frontend was doing with firefox' web dev tools to mess around a bit.

> I am not a fan of retain as it can trigger unexpected automations when i restart things, so i disable it everywhere i can.
On the flows where I do not want the retained data (like you said, automations) I just add an extra function node that filters it out. You can do that by checking `msg.retain`.

For where I do want, I don't filter it out obviously :)

> Wouldn't it be an idea to make the data shown in the state tab available on some specific mqtt topic/endpoint ?
I think the retain flag is more or less in line with how mqtt brokers should handle this type of stuff, but that's my personal opinion, if someone does the work and adds another way to  trigger a republish of any cached state for a device they can open a PR and then it's up to @Koenkk if it gets merged or not.

Koenkk, 2022-09-19:
> I am not a fan of retain as it can trigger unexpected automations when i restart things, so i disable it everywhere i can.

Then you should fix your automations instead. You have to make them rely on the state change, not the payload itself.

closed 2022-10-28 by github-actions[bot]