Koenkk/zigbee2mqtt #14023: Xiaomi ZNCLBL01LM: Template variable warning: 'dict object' has no attribute 'motor_state' when rendering '{{ value_json.motor_state }}'

buhito81, 2022-09-16:
### What happened?

Very frequent error 

`Template variable warning: 'dict object' has no attribute 'motor_state' when rendering '{{ value_json.motor_state }}'`

When using Xiaomi ZNCLBL01LM 

`2022-09-16 13:42:04MQTT publish: topic 'zigbee2mqtt/Bedroom Curtain Left', payload '{"action":null,"battery":100,"charging":false,"device_temperature":24,"hooks_state":"unlocked","linkquality":255,"motor_state":null,"position":86,"power_outage_count":2,"power_source":"battery","running":false,"state":"OPEN","target_position":72,"update":{"progress":24.53,"remaining":1393,"state":"updating"},"update_available":false,"voltage":3000}'`

### What did you expect to happen?

No Errors

### How to reproduce it (minimal and precise)

Connect the device to Zigbee2MQTT

### Zigbee2MQTT version

1.27.2 commit: unknown

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

Koenkk, 2022-09-19:
Given that `motor_state` is in the published payload, this seems to be a HA issue (or a very confusing error message)

closed 2022-09-19 by Koenkk

buhito81, 2022-11-21:
> Given that `motor_state` is in the published payload, this seems to be a HA issue (or a very confusing error message)

@Koenkk I thought I had this fixed, but I don't. `motor_state` is not in the published payload, please see below:

```
2022-11-21 10:22:28.624 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on zigbee2mqtt/Bedroom Curtain Right: b'{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}'
2022-11-21 10:22:28.625 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_battery', 'name': 'Bedroom Curtain Right battery', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_battery)>} with default value '100' and Template("{{ value_json.battery }}")
2022-11-21 10:22:28.626 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_device_temperature', 'name': 'Bedroom Curtain Right device temperature', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_device_temperature)>} with default value '18' and Template("{{ value_json.device_temperature }}")
2022-11-21 10:22:28.626 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_action', 'name': 'Bedroom Curtain Right action', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_action)>} with default value 'None' and Template("{{ value_json.action }}")
2022-11-21 10:22:28.626 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_motor_state', 'name': 'Bedroom Curtain Right motor state', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_motor_state)>} with default value '' and Template("{{ value_json.motor_state }}")
2022-11-21 10:22:28.626 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'motor_state' when rendering '{{ value_json.motor_state }}'
2022-11-21 10:22:28.629 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_hooks_state', 'name': 'Bedroom Curtain Right hooks state', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_hooks_state)>} with default value 'locked' and Template("{{ value_json.hooks_state }}")
2022-11-21 10:22:28.630 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z","linkquality":255,"position":72,"power_outage_count":7,"power_source":"battery","running":false,"state":"CLOSE","target_position":72,"update":{"state":"available"},"update_available":true,"voltage":3000}' with variables {'entity_id': 'sensor.bedroom_curtain_right_target_position', 'name': 'Bedroom Curtain Right target position', 'this': <template TemplateStateFromEntityId(sensor.bedroom_curtain_right_target_position)>} with default value '72' and Template("{{ value_json.target_position }}")
2022-11-21 10:22:28.630 DEBUG (MainThread) [homeassistant.components.mqtt.models] Rendering incoming payload '{"action":null,"battery":100,"charging":false,"device_temperature":18,"hooks_state":"locked","last_seen":"2022-11-21T09:22:28.609Z"," [... cut ...]
2022-11-21 10:22:28.631 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'motor_state' when rendering '{% if not value_json.motor_state %} stopped {% else %} {{ value_json.motor_state }} {% endif %}'
```

I'm a bit lost, but because of this error automations from HA either run/stop running inconsistently. I'd love to help on this, but I don't understand how to change the payload to always include a value or where to find the information for this...