Koenkk/zigbee2mqtt #14036: Devices being "far" from the coordinator don't work as expected even if they had lots of repeaters around.

joaquinvacas, 2022-09-16:
### What happened?

They show a "low" linkquality. But this isn't always the case. It can show 21 lqi and 70 in the next report.
They are fine reporting its state, even showing "0 lqi" they are able to report if they turn on or off (using a button switch or pressing the device's button to switch off/on)

They show this kind of errors "Data request failed with error: 'MAC no ack'" instead of switching off/on, but after some time, let's say 15-30s, they start switching fast as much as you tried before the error appeared.

[image]

Somehow, after some time, they "work fine" but after a while they are a bit unreliable.

Also, trying to switch, works for the first time, maybe turning it off again, but the third time, will get stuck and show some errors like this one:
[image]

It has a "good" number of repeaters around, even more (3 more, 1 this week) expected to be added soon.

But no direct communication with the coordinator.

[image]

Tried with -dev version but nothing changes.

Also, using CC2652P (Sonoff ITead Zigbee USB), not sure if +20dBm is working, changed config in Zigbee2mqtt settings to 20 (maybe any firmware config that I should change too?)

### What did you expect to happen?

All devices to work fine, closer or far away to the coordinator, that's the way you expect mesh networks to work.

### How to reproduce it (minimal and precise)

Add a new device, the interview goes fine, stand far away from coordinator but have many repeaters around.

Try to switch on and off couple times in a row.

Wait until you get stuck with latest status of the device without being able to switch it on or off and "MAC no ack" error to appear.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

 20220219

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus by ITead CC2652P

joaquinvacas, 2022-09-16:
[image]
[image]
[image]

Those are some screenshots from logs where you can check how it's showing errors and then sending all previous commands (stuck before the error) at the same time

jerrm, 2022-09-18:
Assuming you have checked the normal interfenece culprits, I would try dropping the coordinator down to the default +9db, or maybe even lower. 

Having it shout so loudly can confuse devices to try linking with the coordinator directly when they shouldn't.  Also, it looks like there are some bulbs in the grayed out part of the map. Make sure they are always powered.

joaquinvacas, 2022-09-18:
> Assuming you have checked the normal interfenece culprits, I would try dropping the coordinator down to the default +9db, or maybe even lower.
>
> Having it shout so loudly can confuse devices to try [...]

WiFi was made from zero to prevent interferences with Zigbee, Bluetooth isn't nearby and also tried disabling it at all.

Everything is always powered, you see greyed bulbs because when you touch a device only shows the links they made with another device

joaquinvacas, 2022-10-02:
Somehow dropping the coordinator to default dBm value "fixes" the issue. (Deleting the  transmit_power line in configuration.yaml)

20dBm breaks almost every connection that's far from coordinator.

Not sure why, maybe firmware problems?

joaquinvacas, 2022-11-02:
Not stale.

joaquinvacas, 2022-12-04:
Not stale.

877dev, 2022-12-04:
> Somehow dropping the coordinator to default dBm value "fixes" the issue. (Deleting the transmit_power line in configuration.yaml)
> 
> 20dBm breaks almost every connection that's far from coordinator.
> 
> Not sure why, maybe firmware problems?

Having a similar issue where devices next to a router and instead connecting to the coordinator. I will try dropping down the coordinator transmit power, mine is a ZZH so I believe the only way is via `configuration.yaml`.

joaquinvacas, 2023-01-04:
Not stale.

joaquinvacas, 2023-02-04:
> > Somehow dropping the coordinator to default dBm value "fixes" the issue. (Deleting the transmit_power line in configuration.yaml)
> > 20dBm breaks almost every connection that's far from coordinator.
> > Not sure why, maybe firmware problems?
> 
> Having a similar issue where devices next to a router and instead connecting to the coordinator. I will try dropping down the coordinator transmit power, mine is a ZZH so I believe the only way is via `configuration.yaml`.
> 
> On a similar topic, wondering what the transmit power limits are, i.e. is it 0-5 (zzh) or can it be negative i.e. -5 to 5?

Not sure if they can be lower

joaquinvacas, 2023-03-07:
Not stale. Maybe related to fw issues.

joaquinvacas, 2023-04-08:
Nope

joaquinvacas, 2023-05-13:
Not stale

closed 2023-06-21 by github-actions[bot]