Koenkk/zigbee2mqtt #14037: Add time sync to X5H-GB-B wall mounted thermostat

kalbfuss, 2022-09-17:
### Is your feature request related to a problem? Please describe

I bought myself a floor heating thermostat model TGR85-ZB-WP from Beok. The thermostat is recognized as [TuYa X5H-GB-B](https://www.zigbee2mqtt.io/devices/X5H-GB-B.html#tuya-x5h-gb-b). It uses a TS0601 zigbee device.
Everything seems to work well so far. Only the clock of the device is off and I have not found a way set to set it correctly.

### Describe the solution you'd like

a) Time on the device is either synchronized/set regularly by z2m (preferred), or
b) time can be set via the z2m front-end / home assistant through mqtt.

It appears the problem has already been solved for other thermostats depending on the TS0601 zigbee device (see [issue #5620](https://github.com/Koenkk/zigbee2mqtt/issues/5620)). Potentially, the same fix can be applied.

### Describe alternatives you've considered

None

### Additional context

None

Koenkk, 2022-09-19:
>  Only the clock of the device is off and I have not found a way set to set it correctly.

Is it off by exactly X number of hours?

kalbfuss, 2022-09-19:
That is possible. The device appears to be off by exactly 6 hours. Funnily the displayed time even exceeds the 24 hours. At the moment, the device is showing 25:22.

Koenkk, 2022-09-20:
Could you check if the issue is fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    x5h_thermostat: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'x5h_thermostat');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);

            switch (dp) {
            case tuya.dataPoints.x5hState: {
                return {system_mode: value ? 'heat' : 'off'};
            }
            case tuya.dataPoints.x5hWorkingStatus: {
                return {running_state: value ? 'heat' : 'idle'};
            }
            case tuya.dataPoints.x5hSound: {
                return {sound: value ? 'ON' : 'OFF'};
            }
            case tuya.dataPoints.x5hFrostProtection: {
                return {frost_protection: value ? 'ON' : 'OFF'};
            }
            case tuya.dataPoints.x5hWorkingDaySetting: {
                return {week: tuya.thermostatWeekFormat[value]};
            }
            case tuya.dataPoints.x5hFactoryReset: {
                if (value) {
                    clearTimeout(globalStore.getValue(msg.endpoint, 'factoryResetTimer'));
                    const timer = setTimeout(() => publish({factory_reset: 'OFF'}), 60 * 1000);
                    globalStore.putValue(msg.endpoint, 'factoryResetTimer', timer);
                    meta.logger.info('The thermostat is resetting now. It will be available in 1 minute.');
                }

                return {factory_reset: value ? 'ON' : 'OFF'};
            }
            case tuya.dataPoints.x5hTempDiff: {
                return {deadzone_temperature: parseFloat((value / 10).toFixed(1))};
            }
            case tuya.dataPoints.x5hProtectionTempLimit: {
                return {heating_temp_limit: value};
            }
            case tuya.dataPoints.x5hBackplaneBrightness: {
                const lookup = {0: 'off', 1: 'low', 2: 'medium', 3: 'high'};

                if (value >= 0 && value <= 3) {
                    globalStore.putValue(msg.endpoint, 'brightnessState', value);
                    return {brightness_state: lookup[value]};
                }

                // Sometimes, for example on thermostat restart, it sends message like:
                // {"dpValues":[{"data":{"data":[90],"type":"Buffer"},"datatype":4,"dp":104}
                // It doesn't represent any brightness value and brightness remains the previous value
                const lastValue = globalStore.getValue(msg.endpoint, 'brightnessState') || 1;
                return {brightness_state: lookup[lastValue]};
            }
            case tuya.dataPoints.x5hWeeklyProcedure: {
[... 208 lines cut ...]
        exposes.binary('factory_reset', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Resets all settings to default. Doesn\'t unpair device.'),
        exposes.numeric('heating_temp_limit', ea.STATE_SET).withUnit('°C').withValueMax(60)
            .withValueMin(5).withValueStep(1).withPreset('default', 35, 'Default value')
            .withDescription('Heating temperature limit'),
        exposes.numeric('deadzone_temperature', ea.STATE_SET).withUnit('°C').withValueMax(9.5)
            .withValueMin(0.5).withValueStep(0.5).withPreset('default', 1, 'Default value')
            .withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat'),
        exposes.numeric('upper_temp', ea.STATE_SET).withUnit('°C').withValueMax(95)
            .withValueMin(35).withValueStep(1).withPreset('default', 60, 'Default value'),
    ],
    onEvent: tuya.onEventSetLocalTime,
};

module.exports = definition;

```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, wait 24 hours and check if issue is fixed

gargarse, 2022-09-24:
I have that thermostat, it also had synchronization problems. I put the thermostat on the correct time and after a while, the time was totally out of sync. Both the time that was exactly 6 hours ahead, and the day that was two days back from the established date.

With the ext_converter file, something seems to have been fixed. The day no longer changes, but the time is exactly 6 hours ahead on the device. It even goes past 24:00, which is very strange.

root@has:~# date
Sat Sep 24 08:41:21 PM CEST 2022
root@has:~# date -u
Sat Sep 24 06:41:27 PM UTC 2022

In the thermostat: Day: 6, Hour: 26:41

https://www.dropbox.com/s/g1mpr1xpnyvnymu/CM220924-204056001.jpg?dl=0

gargarse, 2022-09-24:
What could I do to solve this problem?

kalbfuss, 2022-09-24:
Same here. In configured the external converter, which koenkk posted. For a while the delta seemed to have decreased to +4 hours. At 21:00 CEST (I was by coincident watching in front of the device), the delta change back to +6 hours.
I assume, it is the line "onEvent: tuya.onEventSetLocalTime", which is supposed to do the trick? Would it be possible to specify a delta for the time and optionally the date/day of week?

Koenkk, 2022-09-25:
@kalbfuss how are you running z2m, is the time on your machine you are running z2m on correct?

kalbfuss, 2022-09-25:
I run z2m as a docker container in Ubuntu Linux on a Raspberry Pi Zero W.
This is the command I have used to create the container:
```
$ docker run -d \
    --name zigbee2mqtt \
    --restart unless-stopped \
    --publish 8080:8080 \
    --volume /data/zigbee2mqtt:/app/data \
    --volume /run/udev:/run/udev:ro \
    --volume /etc/ssl/certs/:/etc/ssl/certs:ro \
    --env TZ=Europe/Berlin \
    --device=dev/ttyS0:/dev/ttyS0 \
    --platform linux/arm/v6 \
    --label=com.centurylinklabs.watchtower.monitor-only=true \
    koenkk/zigbee2mqtt
```
Date and time within the container are accurate (verified at the time of testing). The timezone is CEST (same time zone as the Ubuntu Linux system).
```
$ docker exec -it zigbee2mqtt /bin/sh
$ date
Sun Sep 25 12:25:30 CEST 2022
```
The time shown by the thermostat appears to be 6 hours later. If I am not mistaken the displayed time corresponds to Chinese time, which is UTC+8, while CEST corresponds to UTC+2. This could explain the delta, but not why the displayed time sometimes exceeds 24:00.

gargarse, 2022-09-30:
Any solution for this issue?

kalbfuss, 2022-10-02:
Do you mean, the new proposed external converter only works in combination with z2m v1.28.0? Or do you mean to say the new converter has already been integrated into version 1.28.0?  I am asking because v1.28.0 is not yet available as docker container. If it is number one I will have to wait before I can do the next test.

kalbfuss, 2022-10-02:
Ok, I believe I can answer this question myself. I tried the new proposed converter and z2m fails to load due to absence of the globalStore. Will wait for release of the new container and test afterwards. Thank you very much in any case.

gargarse, 2022-10-02:
I update z2m to v1.28.0, and use new candidate caonverter, with a few changes to get z2m started.

This is the converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');

const tzDataPoints = {
    ...tuya.tzDataPoints,
    keys: ['system_mode', 'preset', 'running_state', 'current_heating_setpoint', 'upper_temperature', 'local_temperature', 'deadzone_temperature', 'heating_temperature_limit'],
}

const valueConverter = {
    systemMode: tuya.valueConverterBasic.lookup({'heat': true, 'off': false}),
    runningState: tuya.valueConverterBasic.lookup({'heat': true, 'idle': false}),
    preset: tuya.valueConverterBasic.lookup({'manual': 0, 'program': 1}),
    divideBy10: tuya.valueConverterBasic.divideBy(10),
};

async function onEventSetLocalTime(type, data, device) {
    // FIXME: What actually nextLocalTimeUpdate/forceTimeUpdate do?
    //  I did not find any timers or something else where it was used.
    //  Actually, there are two ways to set time on TuYa MCU devices:
    //  1. Respond to the `commandMcuSyncTime` event
    //  2. Just send `mcuSyncTime` anytime (by 1-hour timer or something else)

    const nextLocalTimeUpdate = globalStore.getValue(device, 'nextLocalTimeUpdate');
    const forceTimeUpdate = nextLocalTimeUpdate == null || nextLocalTimeUpdate < new Date().getTime();

    if ((data.type === 'commandMcuSyncTime' && data.cluster === 'manuSpecificTuya') || forceTimeUpdate) {
        globalStore.putValue(device, 'nextLocalTimeUpdate', new Date().getTime() + 3600 * 1000);

        try {
            const utcTime = Math.round(((new Date()).getTime()) / 1000);
            const localTime = utcTime - ((new Date()).getTimezoneOffset() + 360) * 60;
            const endpoint = device.getEndpoint(1);

            const payload = {
                payloadSize: 8,
                payload: [
                    ...convertDecimalValueTo4ByteHexArray(utcTime),
                    ...convertDecimalValueTo4ByteHexArray(localTime),
                ],
            };
            await endpoint.command('manuSpecificTuya', 'mcuSyncTime', payload, {});
        } catch (error) {
            // endpoint.command can throw an error which needs to
            // be caught or the zigbee-herdsman may crash
            // Debug message is handled in the zigbee-herdsman
        }
    }
}

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_2ekuz3dz']),
    model: 'TS0601_thermostat',
    vendor: 'TuYa',
    description: 'Thermostat',
    fromZigbee: [tuya.fzDataPoints],
    toZigbee: [tzDataPoints],
[... 30 lines cut ...]
            .withDescription('Antifreeze function'),
        exposes.binary('factory_reset', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Resets all settings to default. Doesn\'t unpair device.'),
        exposes.numeric('heating_temperature_limit', ea.STATE_SET).withUnit('°C').withValueMax(60)
            .withValueMin(5).withValueStep(1).withPreset('default', 35, 'Default value')
            .withDescription('Heating temperature limit'),
        exposes.numeric('deadzone_temperature', ea.STATE_SET).withUnit('°C').withValueMax(9.5)
            .withValueMin(0.5).withValueStep(0.5).withPreset('default', 1, 'Default value')
            .withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat'),
        exposes.numeric('upper_temperature', ea.STATE_SET).withUnit('°C').withValueMax(95)
            .withValueMin(35).withValueStep(1).withPreset('default', 60, 'Default value'),
    ],
    onEvent: onEventSetLocalTime,
};

module.exports = definition;
```
The thermostat works correctly, but I think it does not synchronize the time.

In log of z2m I see this:

```
debug 2022-10-02 13:55:42: Received Zigbee message from 'Beok', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":80}' from endpoint 1 with groupID 0
debug 2022-10-02 13:55:42: No converter available for 'TS0601_thermostat' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":80}'
```

Koenkk, 2022-10-04:
I fixed the globalStore issue, please check again with https://gist.github.com/Koenkk/ee621c8913f698c432d05587ba12d9c7

DTidey, 2022-10-04:
I'm running zigbee2mqtt as a home assistant addon. Version 1.28.0-1.

I have added this https://gist.github.com/Koenkk/ee621c8913f698c432d05587ba12d9c7 as TS0601_thermostat.js in the same directory as configuration.yaml and added to configuration.yaml:

external_converters:
  - TS0601_thermostat.js

I now get the following error when starting the addon.

[22:29:00] INFO: Preparing to start...
[22:29:01] INFO: Socat not enabled
[22:29:03] INFO: Starting Zigbee2MQTT...
/app/dist/util/externally-loaded.js:20
    systemMode: valueConverterBasic.lookup({'heat': true, 'off': false}),
                ^
ReferenceError: valueConverterBasic is not defined
    at /app/dist/util/externally-loaded.js:20:17
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

I'm probably doing something obviously wrong - but am happy to test this fix as I have just received a couple of these thermostats.

gargarse, 2022-10-04:
> I fixed the globalStore issue, please check again with https://gist.github.com/Koenkk/ee621c8913f698c432d05587ba12d9c7

globalStore issue fixed. But don't start correctly z2m.

Log error:
```
Using '/app/data' as data directory
/app/dist/util/externally-loaded.js:20
    systemMode: valueConverterBasic.lookup({'heat': true, 'off': false}),
                ^

ReferenceError: valueConverterBasic is not defined
    at /app/dist/util/externally-loaded.js:20:17
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)

```

Solve with:

diff ext_converter.js ext_converter.js~

```
20,22c20,22
<     systemMode: tuya.valueConverterBasic.lookup({'heat': true, 'off': false}),
<     runningState: tuya.valueConverterBasic.lookup({'heat': true, 'idle': false}),
<     preset: tuya.valueConverterBasic.lookup({'manual': 0, 'program': 1}),
---
>     systemMode: valueConverterBasic.lookup({'heat': true, 'off': false}),
>     runningState: valueConverterBasic.lookup({'heat': true, 'idle': false}),
>     preset: valueConverterBasic.lookup({'manual': 0, 'program': 1}),
```

With that fix, yet start correctly z2m, but this converter fail with severals functions:
```
debug 2022-10-04 21:41:35: Received MQTT message on 'zigbee2mqtt/Beok/set' with data '{"brightness_state":"high"}'
error 2022-10-04 21:41:35: No converter available for 'brightness_state' ("high")
--
debug 2022-10-04 21:41:52: Received MQTT message on 'zigbee2mqtt/Beok/set' with data '{"heating_temperature_limit":35}'
error 2022-10-04 21:41:52: No converter available for 'heating_temperature_limit' (35)
--
debug 2022-10-04 21:41:56: Received MQTT message on 'zigbee2mqtt/Beok/set' with data '{"upper_temperature":60}'
error 2022-10-04 21:41:56: No converter available for 'upper_temperature' (60)

```
Time sync, don't work correctly.

```
debug 2022-10-04 21:59:48: Received Zigbee message from 'Beok', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":163}' from endpoint 1 with groupID 0
debug 2022-10-04 21:59:48: No converter available for 'TS0601_thermostat' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":163}'

```
**** const localTime = utcTime - ((new Date()).getTimezoneOffset() + 360) * 60;
```

     Date: 2022-10-04 (Thuesday (2)) 22:01:18
     Thermostat:                (1)  00:01 

```
https://www.dropbox.com/s/9lk9b9tmdt8sgoy/Screenshot_20221004-220121~2.png?dl=0

Any new ideas or changes to try?

gargarse, 2022-10-04:
More data about issue. Debug log for 'commandMcuSyncTime'

```
Zigbee2MQTT:debug 2022-10-04 23:59:18: Received Zigbee message from 'Beok', type 'commandMcuSyncTime', cluster 'manuSpecificTuya', data '{"payloadSize":103}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-10-04 23:59:18: No converter available for 'TS0601_thermostat' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":103}'
2022-10-04T21:59:18.595Z zigbee-herdsman:controller:endpoint DefaultResponse 0xa4c138c2a86d3a20/1 61184(36, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-10-04T21:59:18.595Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xa4c138c2a86d3a20:45754/1 (0,0,1)
2022-10-04T21:59:18.595Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":45754,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":27,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,125,11,36,0]}}
2022-10-04T21:59:18.595Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,186,178,1,1,0,239,27,0,30,5,24,125,11,36,0,135]
2022-10-04T21:59:18.595Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-04T21:59:18.606Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-10-04T21:59:18.606Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-10-04T21:59:18.606Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-10-04T21:59:18.606Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-10-04T21:59:18.606Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-04T21:59:18.609Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,27,221]
2022-10-04T21:59:18.609Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,27,221]
2022-10-04T21:59:18.609Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,27] - 221
2022-10-04T21:59:18.609Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":27}
2022-10-04T21:59:18.609Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-04T21:59:28.563Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,33,68,129,0,0,0,239,186,178,1,1,0,207,0,36,107,175,0,0,13,9,126,2,0,104,24,2,0,4,0,0,0,217,186,178,29,238]
2022-10-04T21:59:28.563Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,33,68,129,0,0,0,239,186,178,1,1,0,207,0,36,107,175,0,0,13,9,126,2,0,104,24,2,0,4,0,0,0,217,186,178,29,238]
2022-10-04T21:59:28.563Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 33 - 2 - 4 - 129 - [0,0,0,239,186,178,1,1,0,207,0,36,107,175,0,0,13,9,126,2,0,104,24,2,0,4,0,0,0,217,186,178,29] - 238
2022-10-04T21:59:28.563Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":45754,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":207,"securityuse":0,"timestamp":11496228,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,126,2,0,104,24,2,0,4,0,0,0,217]}}

```

Koenkk, 2022-10-05:
- Updated https://gist.github.com/Koenkk/ee621c8913f698c432d05587ba12d9c7
- How much hours is it ahead now? If it is e.g. 2 hours ahead in the line `const localTime = utcTime - ((new Date()).getTimezoneOffset() + 360) * 60;` change `360` to `480`

gargarse, 2022-10-05:
The problem must be another. Regardless of the value you put in, the thermostat does not synchronize the time.

I am using this converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

// set UTC and Local Time as total number of seconds from 00: 00: 00 on January 01, 1970
// force to update every device time every hour due to very poor clock
async function onEventBeokSetLocalTime(type, data, device) {
    // FIXME: What actually nextLocalTimeUpdate/forceTimeUpdate do?
    //  I did not find any timers or something else where it was used.
    //  Actually, there are two ways to set time on TuYa MCU devices:
    //  1. Respond to the `commandMcuSyncTime` event
    //  2. Just send `mcuSyncTime` anytime (by 1-hour timer or something else)

    const nextLocalTimeUpdate = globalStore.getValue(device, 'nextLocalTimeUpdate');
    const forceTimeUpdate = nextLocalTimeUpdate == null || nextLocalTimeUpdate < new Date().getTime();

    if ((data.type === 'commandMcuSyncTime' && data.cluster === 'manuSpecificTuya') || forceTimeUpdate) {
        globalStore.putValue(device, 'nextLocalTimeUpdate', new Date().getTime() + 300 * 1000);

        try {
            const utcTime = Math.round(((new Date()).getTime()) / 1000);
            const localTime = utcTime - ((new Date()).getTimezoneOffset()+480) * 60;
            const endpoint = device.getEndpoint(1);

            const payload = {
                payloadSize: 8,
                payload: [
                    ...convertDecimalValueTo4ByteHexArray(utcTime),
                    ...convertDecimalValueTo4ByteHexArray(localTime),
                ],
            };
            await endpoint.command('manuSpecificTuya', 'mcuSyncTime', payload, {});
        } catch (error) {
            // endpoint.command can throw an error which needs to
            // be caught or the zigbee-herdsman may crash
            // Debug message is handled in the zigbee-herdsman
        }
    }
}

const fzLocal = {
    x5h_thermostat: {
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg, publish, options, meta) => {
            const dpValue = tuya.firstDpValue(msg, meta, 'x5h_thermostat');
            const dp = dpValue.dp;
            const value = tuya.getDataValue(dpValue);

            switch (dp) {
            case tuya.dataPoints.x5hState: {
                return {system_mode: value ? 'heat' : 'off'};
            }
            case tuya.dataPoints.x5hWorkingStatus: {
                return {running_state: value ? 'heat' : 'idle'};
            }
            case tuya.dataPoints.x5hSound: {
                return {sound: value ? 'ON' : 'OFF'};
[... 278 lines cut ...]
Zigbee2MQTT:debug 2022-10-05 17:54:36: No converter available for 'X5H-GB-B*' with cluster 'manuSpecificTuya' and type 'commandMcuSyncTime' and data '{"payloadSize":144}'
2022-10-05T15:54:36.706Z zigbee-herdsman:controller:endpoint DefaultResponse 0xa4c138c2a86d3a20/1 61184(36, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2022-10-05T15:54:36.706Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0xa4c138c2a86d3a20:45754/1 (0,0,1)
2022-10-05T15:54:36.706Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":45754,"destendpoint":1,"srcendpoint":1,"clusterid":61184,"transid":20,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,68,11,36,0]}}
2022-10-05T15:54:36.706Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,186,178,1,1,0,239,20,0,30,5,24,68,11,36,0,177]
2022-10-05T15:54:36.706Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-05T15:54:36.718Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2022-10-05T15:54:36.718Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2022-10-05T15:54:36.718Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2022-10-05T15:54:36.718Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2022-10-05T15:54:36.719Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2022-10-05T15:54:36.724Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,20,210]
2022-10-05T15:54:36.724Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,20,210]
2022-10-05T15:54:36.724Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,20] - 210
2022-10-05T15:54:36.725Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":20}
2022-10-05T15:54:36.725Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[... cut ...]
2022-10-05T15:54:44.673Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":45754,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":207,"securityuse":0,"timestamp":1470178,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,69,2,0,145,24,2,0,4,0,0,0,209]}}

```
After message commandMcuSyncTime are processed: ( https://www.dropbox.com/s/vtn8d1zm9pxex5p/Screenshot_20221005-175453~2.png?dl=0 )

Both the day of the week on the thermostat (1), which should be (3), and the time, have not changed.

gargarse, 2022-10-05:
@Koenkk , Is there a way to inject a packet with, for example, Node Red, to send the **commandMcuSyncTime** message and not have to wait for it to be sent, which occurs every hour?

Koenkk, 2022-10-05:
Updated https://gist.github.com/Koenkk/ee621c8913f698c432d05587ba12d9c7 , a time update will now be send at z2m startup. To verify check if you see `SENDING TIME UPDATE` in your log

gargarse, 2022-10-05:
> Updated https://gist.github.com/Koenkk/ee621c8913f698c432d05587ba12d9c7 , a time update will now be send at z2m startup. To verify check if you see `SENDING TIME UPDATE` in your log

At z2m startup, don't see the message "SENDING TIME UPDATE".  Waiting for "commandMcuSyncTime" message.

gargarse, 2022-10-05:
Problem detected, in try-catch, the call to function ...convertDecimalValueTo4ByteHexArray(), not work.
I change ...convertDecimalValueTo4ByteHexArray by tuya.convertDecimalValueTo4ByteHexArray.

And when restart z2m, yet I see "SENDING TIME UPDATE"

Debug log:
```
Zigbee2MQTT:info  2022-10-05 23:58:51: Connected to MQTT server
Zigbee2MQTT:info  2022-10-05 23:58:51: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
 === ENTERING FUNCTION TO TIME UPDATE === 
data.type:  undefined , data.cluster:  undefined
Date.getTime:  1665007131774
nextLocalTimeUpdate:  undefined , forceTimeUpdate:  true 
 === SENDING TIME UPDATE ===
2022-10-05T21:58:51.775Z zigbee-herdsman:controller:endpoint Command 0xa4c138c2a86d3a20/1 manuSpecificTuya.mcuSyncTime({"payloadSize":8,"payload":[[99,61,254,28],[99,61,197,220]]}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
```
My Timesyn function, with debug options are:
```
// set UTC and Local Time as total number of seconds from 00: 00: 00 on January 01, 1970
// force to update every device time every hour due to very poor clock
async function onEventSetLocalTime(type, data, device) {
    // FIXME: What actually nextLocalTimeUpdate/forceTimeUpdate do?
    //  I did not find any timers or something else where it was used.
    //  Actually, there are two ways to set time on TuYa MCU devices:
    //  1. Respond to the `commandMcuSyncTime` event
    //  2. Just send `mcuSyncTime` anytime (by 1-hour timer or something else)

    console.log('\n\nENTERING FUNCTION TO TIME UPDATE\n\n');

    const nextLocalTimeUpdate = globalStore.getValue(device, 'nextLocalTimeUpdate');
    const forceTimeUpdate = nextLocalTimeUpdate == null || nextLocalTimeUpdate < new Date().getTime();
    
    console.log('\ndata.typ: ' +  data.type + ', data.cluster: ' + data.cluster +  ', forceTimeUpdate: ' + forceTimeUpdate + '\n\n')
    
    if ((data.type === 'commandMcuSyncTime' && data.cluster === 'manuSpecificTuya') || forceTimeUpdate) {
        globalStore.putValue(device, 'nextLocalTimeUpdate', new Date().getTime() + 3600 * 1000);

        try {
            const utcTime = Math.round(((new Date()).getTime()) / 1000);
            const localTime = utcTime - ((new Date()).getTimezoneOffset() - 240) * 60;
            const endpoint = device.getEndpoint(1);

            const payload = {
                payloadSize: 8,
                payload: [
                    tuya.convertDecimalValueTo4ByteHexArray(utcTime),
                    tuya.convertDecimalValueTo4ByteHexArray(localTime),
                ],
            };
            console.log('\n\nSENDING TIME UPDATE\n\n');
            await endpoint.command('manuSpecificTuya', 'mcuSyncTime', payload, {});
        } catch (error) {
            // endpoint.command can throw an error which needs to
            // be caught or the zigbee-herdsman may crash
            // Debug message is handled in the zigbee-herdsman
        }
    }
}
``` 
The `mcuSyncTime` message is already sent with its proper payload.

But the thermostat is still out of sync. Ignore the `mcuSyncTime` message.
The same payload is not correct.
That I can continue investigating @Koenkk ?

kkossev, 2022-10-06:
@gargarse the time sync is requested by the thermostat every time you switch its main power supply off and then on.

BTW, no success with the time synchronization for this BEOK clone in another platform too.. :(  The time sync function works fin on AVATTO thermostat, on multiple Tuya T/H sensors with LCD display that shows the date and time - but not on this device.

The weirdest thing is that I can't set the correct time on _TZE200_2ekuz3dz using Tuya own Zigbee gateways as well.  I have 2 different Tuya gateways, both fail to set the correct time.

Koenkk, 2022-10-06:
@gargarse does the timeshift in hours change when you modify the `240` value in `const localTime = utcTime - ((new Date()).getTimezoneOffset() - 240) * 60;` to e.g. `480`?

gargarse, 2022-10-07:
> @gargarse¿Cambia el cambio de hora en horas cuando modifica el `240`valor en, `const localTime = utcTime - ((new Date()).getTimezoneOffset() - 240) * 60;`por ejemplo `480`,?

No. I have tried changing that value several times, but the thermostat ignores the changes. It seems to ignore the message commandMcuSyncTime

Koenkk, 2022-10-07:
Thats strange, maybe the firmware is bugged. 

> The weirdest thing is that I can't set the correct time on _TZE200_2ekuz3dz using Tuya own Zigbee gateways as well. I have 2 different Tuya gateways, both fail to set the correct time. The time is always 8 hours ahead of GMT.

If it doesn't even work with the official gateway there is little chance we can make it works with z2m.

gargarse, 2022-10-07:
Definitely, the thermostat is a "shit".

Not even with a GW Tuya do they work correctly.

With Smart Life App:
https://www.dropbox.com/s/rac3jvj50z2v2sg/Screenshot_20221007-205654~2.png?dl=0

And the time in thermostat:
https://www.dropbox.com/s/1jr7wwu945e43he/IMG_20221007_205528.jpg?dl=0

Thank you all for the help offered.

kalbfuss, 2022-10-08:
> If it doesn't even work with the official gateway there is little chance we can make it works with z2m.

I would not give up so quickly, @Koenkk. This thermostat comes in many flavours and firmwares naturally evolve. My thermostat definitely accepts the time sync, maybe not if forced, but definitely periodically. Otherwise the time would not be exactly 6 hours in advance.
I assume the offset originates from an incorrect (or better default) time zone set in the device. So one either has to configure the time zone correctly, which unfortunately requires a Tuya gateway or one needs to program an offset. Since I do not own a Tuya gateway I am going to play with the offset as advised and report the result back in the next days.

gargarse, 2022-10-08:
> I would not give up so quickly, @Koenkk. This thermostat comes in many flavours and firmwares naturally evolve. My thermostat definitely accepts the time sync, maybe not if forced, but definitely [...]

As indicated here ( https://support.tuya.com/en/help/_detail/K9hutfy2zi0y9 ) and in some other, the synchronization of the time zone is done with the one you have on the mobile, when installing the app and synchronizing the gateway and the devices.
I bought a gateaway to try it and definitely as I said, that didn't work. The thermostat does not synchronize the clock. moreover, it has a drift of the order of +3 minutes every 12 hours.
So even adjusting it manually does not work in conditions with so much time drift.

I have contacted Beok to see if they have any firmware or a solution to the problem.

kalbfuss, 2022-10-16:
I have update my z2m installation to version 1.28.0 and installed the latest custom converter proposed by @Koenkk on 2022-Oct-05. As a result the thermostat no longer syncs (not even passively). I can tell because the time has started to drift away and was more than 6h ahead. I thus disabled the custom converter again, but the thermostat still does not sync. I have waited for more than 24h in either case. This used to work before, albeit I had a time offset of exactly 6h. It seems that something has changed from v1.27.0 to v.1.28.0.
I have now re-enabled the custom converter with the correction proposed by @gargarse. Curious to see whether that makes a difference.

kalbfuss, 2022-10-16:
First observation: I am not seeing the message "SENDING TIME UPDATE" in the log after startup. I neither see the message "ENTERING FUNCTION TO TIME UPDATE"  from the additional log call that @gargarse added. The function onEventSetLocalTime is apparently not called at z2m startup. 

@Koenkk What may cause this behavior? I checked the tuya.fingerprint of the converter. The zigbee model and manufacturer seem to mach. How can I verify that the external converter is used.

I am going to wait now and see whether the time sync is triggered by the device later in the day.

DTidey, 2022-10-16:
@kalbfuss 
I am having similar problems to you. I've got the attached external converter and I am sure it is being picked up.
Have a look at the description at line 313 - change this and you can see if the converter is being used.

The time update function is being called with this converter (messages are appearing in the loc) but the sent message is not being picked up by the end device. Well - the time is not changing, doesn't matter what offset is sent.

Looking at the original source code of the time update function and there is an offset from Jan 1 2000 in one of them. I am suspicious that the format of the time update message sent is not correct. As I said, it doesn't matter what offset I set in the function it has no impact on the end device.

I have also used an external Tuya hub and the Tuya app and that doesn't work either. It keeps setting the time to Bejing time!

[my_X5H-GB-B.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/9794147/my_X5H-GB-B.js.txt)

kalbfuss, 2022-10-16:
Thanks, @DTidey. I changed the description as you suggested and indeed the custom converter is being used for the thermostat. Still, the function onEventSetLocalTime is not called during z2m as suggested by @Koenkk. What might be the reason? Which part of the code is supposed to trigger the functional call during startup?
@DTidey I have not tried your version of the converter yet, but will do so in the next step.

kkossev, 2022-10-16:
I've succeeded to make a  workaround for the _TZE200_2ekuz3dz time synchronization firmware bug:
* localTime must always have a hardcoded offset of 8 hours
* utcTime must have an offset calculated as (8hours - trueOffsetHours)
* the TuyaSetTime command (0x24) must be skipped (not sent to the thermostat) between 00:00 and 07:59 local time, otherwise the thermostat will display a buggy time value.
The workaround was successfully tested in CET and EET time zones (on Hubitat platform)

DTidey, 2022-10-16:
@kkossev good news. Could you share your code?

kkossev, 2022-10-16:
The Groovy code is [here ](https://github.com/kkossev/Hubitat-Tuya-Wall-Thermostat/blob/main/Tuya-Wall-Thermostat.groovy)( lines 203 ... 234)

DTidey, 2022-10-17:
@kkossev - thank you so much for sharing your code. Could I ask a couple of questions about the time offsets.
it appears that utcTime is:  now() minus (8 hours - offsethours) 
BUT
localTIme is:  now() plus 8 hours
One minus and one plus, Is that correct?

Attached is my attempt at implementing this in a custom converter. It isn't working but may help someone else solve this.
[my_X5H-GB-B.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/9802754/my_X5H-GB-B.js.txt)

kkossev, 2022-10-18:
> One minus and one plus, Is that correct?

Yes, it is correct. 
8 hours are added to localTime 
(8-offsetHours) are subtracted from the utcTime 

I looked at your code, seems to be correct.

kkossev, 2022-10-18:
AVATTO M81AH is on the left, and BEOK X5H is on the right,

[image]

BEOK (clone?) build quality is not great, but the big advantage for me is that the annoying beep sound can be turned off and that the LCD backlight can be turned off completely as well. I am planning to use one cheap PIR sensor to activate the LCD display only when there is a movement very close to the thermostat.

gargarse, 2022-10-18:
Hello.
I wrote an email to Beok with some screenshots and a video, showing them how the thermostat syncs (badly), when you start up and with the Tuya gateway and the Smart Life app.
Then I referred you to this issue.

They tell me that they send me a new one, when I receive it, I will see if it works correctly first with the Tuya gateway and the Smart Life app, and then with z2m.

In this video that I recorded and sent to Beok you can see how by putting the thermostat out of time, disconnecting it from the power for a couple of minutes, I get it to synchronize at startup (badly, with a drift of 6 hours) and also with a totally incorrect time (> 24 hours).

The video: https://www.dropbox.com/s/ta80nj8b3qmof97/VID-20221014-WA0001.mp4?dl=0

vladyspavlov, 2022-11-06:
I tested it with the Tuya gateway and can confirm the time issue still persists.

It seems the thermostat time runs OK, but due to a firmware bug, there is an 8-hour offset. So if you try to put 00:00, the thermostat will interpret it as 08:00; if you put 23:59, it will be 31:59. I didn't find any way to put, for example, 04:00, it will be interpreted as 12:00 on the thermostat. The next time after 31:59 will be 00:00, and if you will not send the new time, it will count time correctly, i.e., 00:01, 00:02, etc. You can send the new time only after 08:00 to synchronize it, with an offset of 8 hours (so the real sent time will be 00:00).

So I see two possible ways to resolve this:
1) Leave it with the bugged offset. With synchronization, it will be from 08:00 (00:00) to 31:59 (23:59). It seems like there are no issues except for the displayed time.
2) Try to correct the time to 00:00 and 23:59. There will be no synchronization from 00:00 to 08:00, and in this range there may be some time drift, which we can't control until 08:00. This was already previously done https://github.com/Koenkk/zigbee2mqtt/issues/14037#issuecomment-1280051940.

Or maybe the manufacturer will update the firmware and fix this issue.

kalbfuss, 2022-11-07:
I can confirm that. I bought myself a Tuya gateway to find out whether the timezone of the Beok thermostat can be adjusted. After connection to the gateway the timezone of the device is displayed as Europe/Amsterdam (GMT+1), which is the timezone of my smartphone, but the time displayed is GMT+8. It indeed seems to be a firmware bug. Unfortunately, there was no new firmware available according to the Tuya app. In other words, a customized time setting function, which considers the offset is probably the only way to go.

kalbfuss, 2022-11-07:
In principle, setting the time with an offset to correct for the firmware bug should be straight forward. The respective calculations should be something along the following lines:
```
const utcTime = Math.round(((new Date()).getTime()) / 1000) - 7 * 3600;
const localTime = utcTime + 8 * 3600;
```
A somewhat smarter implementation would consider the offset of the timezone on the z2m instance and do the UTC calculation accordingly.
Unfortunately, the time sync has been rather erratic in my setup, recently. Sometimes the time on the Beok thermostat is synced, sometimes it is not. I am not even sure that the time setting function is called on a regular basis.
I tried to use console.log to drop log messages, but these do not appear in the z2m log. I understand that z2m has a built-in logger, but I am not sure how to use it nor have I found any documentation about it.
Can anyone tell me how to write log message to the z2m log file? That would help me greatly to understand the problem.

vladyspavlov, 2022-11-07:
> The respective calculations should be something along the following lines:
> 
> ```
> const utcTime = Math.round(((new Date()).getTime()) / 1000) - 7 * 3600;
> const localTime = utcTime + 8 * 3600;
> ```

It doesn't respect local time. I tried multiple times, but it only reads the UTC time. But anyway, if we try to fix it via converter, we will need to create a new one especially for this thermostat for time sync, and we can consider the time zone of the user. It's not a problem.

kalbfuss, 2022-11-16:
I am getting the following error message with your coverter:
```
/app/dist/util/externally-loaded.js:38
    systemMode: tuya.valueConverterBasic.lookup({'heat': true, 'off': false}),
                                         ^

TypeError: Cannot read properties of undefined (reading 'lookup')
    at /app/dist/util/externally-loaded.js:38:42
    at Script.runInContext (node:vm:139:12)
    at Script.runInNewContext (node:vm:144:17)
    at Object.runInNewContext (node:vm:298:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
    at Object.getExternalConvertersDefinitions (/app/lib/util/utils.ts:165:25)
    at getExternalConvertersDefinitions.next (<anonymous>)
    at new ExternalConverters (/app/lib/extension/externalConverters.ts:12:20)
    at new Controller (/app/lib/controller.ts:84:58)
```

gargarse, 2022-11-19:
Hello everyone.

With the external converter of sheaur, I have done some regression tests with the thermostat.

Arriving at several consequences:

1.- With any change on the localTime variable, no result is obtained on the synchronization of the thermostat clock.

In the last state of tests, I even set this variable to 0. And the thermostat works and synchronizes correctly.

2.- For any change, it would be necessary to do it on the utcTime variable.

After much trial and error, I have come up with a combination that makes the thermostat sync correctly. And for this I had to force NOT to synchronize during the interval from 00 hours to 08 hours (Localtime).

Since if synchronizing exceeds 23:59 h, the thermostat due to its internal problem would jump to wrong hour numbers (24 .. 31) and after 31:59, that is when it would jump to 08:00.

Here is the log of the synchronization attempts:
```
Zigbee2MQTT:info  2022-11-18 19:53:43: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 19:53:43: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 19:57:30: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 19:57:30: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 19:59:59: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 20:59:42: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 21:59:25: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 22:00:01: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 22:59:44: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 22:59:44: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 22:59:44: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 22:59:45: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 22:59:45: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 23:00:14: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 23:53:21: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-18 23:53:21: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-19 00:00:01: NOT Sync Time message to TGR85-ZB, to solve gap problem
Zigbee2MQTT:info  2022-11-19 00:16:18: NOT Sync Time message to TGR85-ZB, to solve gap problem
Zigbee2MQTT:info  2022-11-19 00:16:19: NOT Sync Time message to TGR85-ZB, to solve gap problem
Zigbee2MQTT:info  2022-11-19 00:59:25: NOT Sync Time message to TGR85-ZB, to solve gap problem
Zigbee2MQTT:info  2022-11-19 01:59:08: NOT Sync Time message to TGR85-ZB, to solve gap problem
Zigbee2MQTT:info  2022-11-19 02:58:51: NOT Sync Time message to TGR85-ZB, to solve gap problem
Zigbee2MQTT:info  2022-11-19 03:58:33: NOT Sync Time message to TGR85-ZB, to solve gap problem
Zigbee2MQTT:info  2022-11-19 04:58:16: NOT Sync Time message to TGR85-ZB, to solve gap problem
Zigbee2MQTT:info  2022-11-19 05:57:58: NOT Sync Time message to TGR85-ZB, to solve gap problem
Zigbee2MQTT:info  2022-11-19 06:57:40: NOT Sync Time message to TGR85-ZB, to solve gap problem
Zigbee2MQTT:info  2022-11-19 07:57:22: NOT Sync Time message to TGR85-ZB, to solve gap problem
Zigbee2MQTT:info  2022-11-19 08:57:04: NOT Sync Time message to TGR85-ZB, to solve gap problem
Zigbee2MQTT:info  2022-11-19 09:56:44: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-19 10:30:53: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-19 10:30:54: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-19 10:59:51: Send Sync Time message to TGR85-ZB
Zigbee2MQTT:info  2022-11-19 11:00:03: Send Sync Time message to TGR85-ZB
```
Attached the external converter used as definitive,

```
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = `require('zigbee-herdsman-converters/lib/reporting');`
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const x5h_set_time_request = {
    cluster: 'manuSpecificTuya',
    type: ['commandMcuSyncTime'],
[... 318 lines cut ...]
            'For example: `06:00/20.5 08:00/15 11:30/15 13:30/15 17:00/22 22:00/15 06:00/20 22:00/15`. ' +
            'The thermostat doesn\'t report the schedule by itself even if you change it manually from device'),
        e.child_lock(), e.week(),
        exposes.enum('brightness_state', ea.STATE_SET, ['off', 'low', 'medium', 'high'])
            .withDescription('Screen brightness'),
        exposes.binary('sound', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Switches beep sound when interacting with thermostat'),
        exposes.binary('frost_protection', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Antifreeze function'),
        exposes.binary('factory_reset', ea.STATE_SET, 'ON', 'OFF')
            .withDescription('Resets all settings to default. Doesn\'t unpair device.'),
        exposes.numeric('heating_temp_limit', ea.STATE_SET).withUnit('Â°C').withValueMax(60)
[... cut ...]
        exposes.numeric('deadzone_temperature', ea.STATE_SET).withUnit('Â°C').withValueMax(9.5)
            .withValueMin(0.5).withValueStep(0.5).withPreset('default', 1, 'Default value')
            .withDescription('The delta between local_temperature and current_heating_setpoint to trigger Heat'),
        exposes.numeric('upper_temp', ea.STATE_SET).withUnit('Â°C').withValueMax(95)
            .withValueMin(35).withValueStep(1).withPreset('default', 60, 'Default value'),
    ],
};

module.exports = definition;

```

gargarse, 2022-11-19:
A couple of weeks ago I contacted with Beok and informed them of this problem with their thermostat.
Well, they have sent me a "reprogrammed" one, which I have tested for a week with the Tuya gateway and the Tuya application. The result has been satisfactory. The thermostat has stayed in sync all week with no problem.

Now I will test it with zigbee2mqtt. I will inform you by this means of it.

Best regards

gargarse, 2022-11-19:
The tests on the new device **have not been very good**.

The "patch" that Beok's have put has been to **completely disable synchronization via Zigbee messages**.

The device is much more stable and has no drift, but it does not send the request message: "commandMcuSyncTime" at any time.

Functional, it may be, but for example you would have to change your time when the time changes from summer to winter in Europe. In my case from CET to CEST and viceverse.

@Koenkk  is there a way from z2m to send you a forced sync message?
Let me explain, send him the synchronization package as if he had sent the request through the command "commandMcuSyncTime"

Best regards

vladyspavlov, 2022-11-19:
> Let me explain, send him the synchronization package as if he had sent the request through the command \"commandMcuSyncTime\

How will the thermostat react if you manually try to send time? Does it work?

gargarse, 2022-11-19:
> > Let me explain, send him the synchronization package as if he had sent the request through the command "commandMcuSyncTime\
> 
> How will the thermostat react if you manually try to send time? Does it work?

I do not know if it works. What I would like is to test if it works or not.

vladyspavlov, 2022-11-19:
> I do not know if it works. What I would like is to test if it works or not.

In dev console set endpoint `1`, cluster `0xef00`, command `0x24` and try to execute the next command:
`{"payloadSize": 8, "payload": [99, 120, 253, 128, 99, 120, 253, 128]}`
If the thermostat processes this command correctly, it should be set to 16:00.

You could also try to test the following commands:
`{"payloadSize": 8, "payload": [99, 120, 28, 128, 99, 120, 28, 128]}` - 00:00 19.11.2022
`{"payloadSize": 8, "payload": [99, 120, 141, 0, 99, 120, 141, 0]}` - 08:00 19.11.2022
`{"payloadSize": 8, "payload": [99, 121, 95, 240, 99, 121, 95, 240]}` - 23:00 19.11.2022

gargarse, 2022-11-19:
> In dev console set endpoint `1`, cluster `0xef00`, command `0x24` and try to execute the next command: `{"payloadSize": 8, "payload": [99, 120, 253, 128, 99, 120, 253, 128]}` If the thermostat [...]

Yeah!!

Works correctly. The thermostat is synchronized using the developer console.

But Beok people, didn't fix the problem, just hide it. Because the thermostat continues with the drift of +8 hours that we had identified.

The only "good" thing about this firmware version is that it doesn't deviate so much over the hours.

sheaur, 2022-11-20:
El external convert de gargarse funciona genial, gracias a todos.

sheaur, 2022-11-24:
Ahora mi dispositivo después de actualizar vía OTA ha cambiado de ts0610 a ts011f y no enlaza. Le ha pasado a alguien, como puedo regresar?

kkossev, 2022-11-25:
@gargarse have BEOK changed the Tuy application version after this "fix" ?

sheaur, 2022-11-25:
No, fué automático a través de z2m

sheaur, 2022-11-25:
En mi external convert puse:
const ota = require('zigbee-herdsman-converters/lib/ota');

const definition = { 
fingerprint: [{
modelID: 'TS0601',
 manufacturerName: '_TZE200_2ekuz3dz'}], 
model: 'TGR85-ZB (Beok Ext. Dev.)', 
vendor: 'TuYa', 
whiteLabel: [{vendor: 'Beok', model: 'TGR85-ZB'}], 
ota: ota.zigbeeOTA,
description: 'Wall-mount thermostat (EXTERNAL)', fromZigbee: [fz.ignore_basic_report, fzLocal.x5h_thermostat, x5h_set_time_request], toZigbee: [tzLocal.x5h_thermostat], configure: tuya.configureMagicPacket,

kain, 2023-01-01:
Following

codo70, 2023-01-15:
still an issue.

codo70, 2023-01-15:
can anyone provide me with a firmware. mine was recogniced as a powerplug after uto update

[7 more replies]

closed 2023-02-24 by github-actions[bot]