Koenkk/zigbee2mqtt #14049: Unable to pair devices with CC2652RB

martinushm, 2022-09-17:
### What happened?

Spent about a weeks worth of time now trying to get Zigbee working so some help would be much appreciated. As my old setup stopped working out of nowhere, I started anew with a reinstall of HA on my RPI and with a new Zigbee coordinator (CC2652RB). 

First I experienced '502 bad gateway error' for Z2M, which I fixed by changing 'localhost' in the 'server: mqtt://localhost:1883' to the IP-address (don't know why). Now Z2M is working without errors in the log, but I am unable to pair any device. I tried touchlink since I have Philips hue bulbs, but this caused the CC2652RB to crash. I fixed this by updating the firmware to the latest dev version 20220726. Now touchlink does not crash the coordinator, but the inability to pair devices remains.

Everything is up-to-date, there are no logs showing issues. I've read some similar threads about this and tried all the provided solutions - but no luck. Thanks in advance!!

### What did you expect to happen?

When pairing is initiated for the interview process to begin and the device to join the network as visible in the Z2M frontend.

### How to reproduce it (minimal and precise)

1. Enable joining
2. Put device in pairing mode
3. See nothing happen

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220726

### Adapter

Slaesh's CC2652RB stick (with USB-cable)

### Debug log

[Archief.zip](https://github.com/Koenkk/zigbee2mqtt/files/9592328/Archief.zip)

martinushm, 2022-09-19:
Yes, but did a full reinstall of HA afterwards.

martinushm, 2022-10-01:
I might have misunderstood what you mean with 'repairing all devices'. Do you mean 'to re-pair', as in 'pair again'; or 'to repair', as in 'fix something that's broken'?

I've followed the procedure here: https://www.zigbee2mqtt.io/guide/faq/#how-do-i-migrate-from-one-adapter-to-another, still no change however.

Koenkk, 2022-10-02:
I mean pair again. Does the device pair when you hold it very close to the coordinator?

martinushm, 2022-10-02:
It does not. I tried it with multiple different smart devices, including an Aqara temperature/humidity sensor that I can hold very close to the coordinator.

Koenkk, 2022-10-03:
Then I suggest to power down all Zigbee routers and try to pair again. If that doesn't help, reflash your coordinator, remove your `coordinator_backup.json` and do a full re-pair (all devices)

martinushm, 2022-11-03:
Currently very busy, will attempt to fix this later.

closed 2022-12-13 by github-actions[bot]

martinushm, 2023-02-26:
Been giving this another crack after updating to the new versions of HA and Zigbee2MQTT. I have now been able to pair one Aqara sensor, but for some reason not others of the same sensor model.

I am currently trying to pair Philips Hue bulbs, which I am able to connect with via TouchLink and then factory reset. Weirdly enough, after the factory resetting and permitting join they do not show up. Also tried unplugging and plugging again while searching, but to no avail.

Any suggestions would be much appreciated.

martinushm, 2023-02-26:
Solved by switching the zigbee channel to 25. Issue probably caused due to interference on channel 11 with old devices and my wifi router.