Koenkk/zigbee2mqtt #14056: Namron Touch Thermostat - Energy consumption

newlund, 2022-09-18:
### What happened?

Trying to add the energy consumption from my Namron floor thermostat into home assistant energy panel.
https://www.zigbee2mqtt.io/devices/4512737_4512738.html

It is not listed when I try to add the energy source.
https://www.home-assistant.io/more-info/statistics/
https://developers.home-assistant.io/docs/core/entity/sensor#long-term-statistics

### What did you expect to happen?

Expect to be able to add the energy source to the energy panel.

### How to reproduce it (minimal and precise)

Go to https://hass_host/config/energy
Add individual device
Try to find the entity in the popup list

### Zigbee2MQTT version

1.27.2-1

### Adapter firmware version

0x26720700

### Adapter

Conbee2

owangen, 2022-09-19:
@newlund I have 3 of same devices and they report energy consumption for me. Could you check z2m frontend and check if energy is reported when you turn thermostat in heating mode?

newlund, 2022-09-19:
Strange, I have it in heating mode

[image]

newlund, 2022-09-19:
Sorry, now I found it when I searched for the entity name.
It did not show up in the list.
My bad... :(

closed 2022-09-19 by newlund