Koenkk/zigbee2mqtt #14065: [New device support]:

netware0, 2022-09-18:
### Link

https://merxu.com/en/offer/philips-hue-svithue-adore-led-20w-3418231-p6-b700737e-3fc3-49c6-b068-1cf34a45f6f6/

### Database entry

{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0024c0b220","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null,"defaultSendRequestWhen":"immediate"} {"id":2,"type":"Router","ieeeAddr":"0x00178801085746b3","nwkAddr":55760,"manufId":4107,"manufName":"Philips","powerSource":"Mains (single phase)","modelId":"3418231P6","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":268,"inClusterList":[0,3,4,5,6,8,4096,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20190529","swBuildId":"1.56.7_r29434","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1663528818549,"defaultSendRequestWhen":"immediate"}

### Comments

i tried to follow the supporting new device page but got stuck at...
model id:  3418231P6

### Supported color modes

Color temperature

### Color temperature range

{"colorTempPhysicalMin":153,"colorTempPhysicalMax":454}

danieledwardgeorgehitchcock, 2022-09-19:
I'll have a crack at this... leave it with me

danieledwardgeorgehitchcock, 2022-09-19:
I've created a pull request - see above. Please wait for it to be merged and switch to the Dev branch to test

closed 2022-09-19 by Koenkk