Koenkk/zigbee2mqtt #14067: SNZB-02 fail paired

clem2107, 2022-09-19:
### What happened?

Hello,

I try to connect my 2 SNZB-02, my devices are well discovered but do not connect.
Here are the logs:

`Zigbee: allowing new devices to join.
Device '0x00124b0022671da2' joined
Starting interview of '0x00124b0022671da2'
Failed to interview '0x00124b0022671da2', device has not successfully been paired`

I am under Home-Assistant, with a Zb Dongle-E.

I have other devices that work fine.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ZB Dongle-E

Koenkk, 2022-09-19:
Today a pairing improvement has been merged for EZSP adapters; can you try it? (https://github.com/Koenkk/zigbee-herdsman/pull/580). 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

tomm1ed, 2022-09-20:
To wit, my SNZB-02 connected without issues to my setup
Zigbee2MQTT version 1.27.2
Adapter firmware version 6.7.8.0 build 373
Adapter Silvercrest Zigbee Gateway Model TYGWZ-01 manufactured by Lidl

clem2107, 2022-09-21:
Hello,

I went to dev branch, my sensor is configured well but no temperature return with the following error:

`error 2022-09-20 20:43:44: Configure failed: Error: Bind 0x00124b0022671da2/1 msTemperatureMeasurement from '0xe0798dfffe83a1da/1' failed (Route error)`

clem2107, 2022-10-04:
up ?

closed 2022-11-12 by github-actions[bot]