Koenkk/zigbee2mqtt #14070: Zigbee2MQTT Web Ui not working

workpioupiou, 2022-09-19:
### What happened?

I can't connect to the Ui Web interface of my container zigbee2mqtt (other dockers works). 

There is no error on the logs and when I tried to access it from my-ip:5001 (my forxarding port)  it doesn't work.

docker ps show me that the port is map and I really don't want to change the network to host.

### What did you expect to happen?

Access to the web site. 

### How to reproduce it (minimal and precise)

On a raspberry pi 4, install docker and docker compose. Add a mosquitto and zigbee2mqtt docker-compose. 
Configure the docker compose of the zigbee2mqtt  with a port forwarding. 
Try to access the Web UI

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

0000000

### Adapter

CC2531

### Debug log

info  2022-09-19 09:38:01: Logging to console and directory: '/app/data/log/2022-09-19.09-38-01' filename: log.txt
info  2022-09-19 09:38:01: Starting Zigbee2MQTT version 1.27.2 (commit #7dc48fb)
info  2022-09-19 09:38:01: Starting zigbee-herdsman (0.14.53)
info  2022-09-19 09:38:02: zigbee-herdsman started (resumed)
info  2022-09-19 09:38:02: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}'
info  2022-09-19 09:38:02: Currently 6 devices are joined:
info  2022-09-19 09:38:02: 0x0017880109179034 (0x0017880109179034): 9290012607 - Philips Hue motion sensor (EndDevice)
info  2022-09-19 09:38:02: 0x001788010807fb9a (0x001788010807fb9a): 324131092621 - Philips Hue dimmer switch (EndDevice)
info  2022-09-19 09:38:02: 0x0017880102c630f4 (0x0017880102c630f4): 8718696449691 - Philips Hue White A60 Single bulb E27/B22 (Router)
info  2022-09-19 09:38:02: 0x0017880102b5bf7e (0x0017880102b5bf7e): 8718696449691 - Philips Hue White A60 Single bulb E27/B22 (Router)
info  2022-09-19 09:38:02: 0x0c4314fffe60ad11 (0x0c4314fffe60ad11): TS011F_plug_1 - TuYa Smart plug (with power monitoring) (Router)
info  2022-09-19 09:38:02: 0x0c4314fffe5d969e (0x0c4314fffe5d969e): TS011F_plug_1 - TuYa Smart plug (with power monitoring) (Router)
warn  2022-09-19 09:38:02: `permit_join` set to  `true` in configuration.yaml.
warn  2022-09-19 09:38:02: Allowing new devices to join.
warn  2022-09-19 09:38:02: Set `permit_join` to `false` once you joined all devices.
info  2022-09-19 09:38:02: Zigbee: allowing new devices to join.
info  2022-09-19 09:38:03: Connecting to MQTT server at mqtt://mosquitto:1883
info  2022-09-19 09:38:03: Connected to MQTT server
info  2022-09-19 09:38:03: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2022-09-19 09:38:03: Configuring '0x0c4314fffe60ad11'
error 2022-09-19 09:38:13: Failed to configure '0x0c4314fffe60ad11', attempt 1 (Error: Read 0x0c4314fffe60ad11/1 genBasic(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:415:23)
    at Queue.executeNext (/app/node_modules/zigbee-herdsman/src/utils/queue.ts:32:32))
info  2022-09-19 09:38:13: Configuring '0x0c4314fffe5d969e'

iMartien, 2022-09-19:
I have the same issue, not using docker, just plain linux installation on RPI3. Web UI stopped working after updating to the latest version. The rest just works fine.

tskulbru, 2022-09-23:
Ditto, MQTT events still being sent. WebUI is blank though.

`
zigbee2mqtt     | Zigbee2MQTT:info  2022-09-23 13:31:27: Starting Zigbee2MQTT version 1.27.2 (commit #7dc48fb)
zigbee2mqtt     | Zigbee2MQTT:info  2022-09-23 13:31:27: Starting zigbee-herdsman (0.14.53)
`

thegarywilder, 2022-10-01:
Same here. Events still firing. Of the UI pages, only Map is functional and will Load all devices. Initial load of the UI framework is much slower than usual. Restart of add-n or HAAS made no difference. Different browsers, clear cache also no difference.
As an aside, I have a Sengled light that drops from the network when I lose power or update HAAS OS and without the UI I can't fix it (Permit Join, put bulb in pairing mode, click the Reconfigure button in the interface).

Update: Updated the add-on configuration to enable zigbee_herdsman_debug, restart the add-on, disable herdsman, restart the add-on, go to Z2M UI and load map, Devices page now populates, as well as all the rest. Load speed is back to normal, too.

anthosz, 2022-10-08:
Hello,

I have also the same issue, Zigbee2mqtt start well (cf screenshots) and it seems to receive datas (db is updated) but 8080 is not open (even in the container itself)
[image]
[image]

I also tried a rollback to previous version (27.2) and same issue. So HA cannot receive any update.

FrankGiesecke, 2022-11-14:
Did any one know if the problem still exists with the latest version [1.28.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.2)?

workpioupiou, 2022-11-22:
> Did any one know if the problem still exists with the latest version [1.28.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.28.2)?

Yes, for me it still exist. No UI displayed on " Starting Zigbee2MQTT version 1.28.2"

workpioupiou, 2022-11-22:
It's working, my bad !!! I don't know why but "frontend : true" was missing in my configuration.yaml. Does this appear in last version ?

closed 2022-11-22 by workpioupiou

cheak1974, 2023-04-01:
Well, yes. Basically you are showing how to fire up a node-red docker container and install the widget package freshly which is not really a good idea. Afterwards you show a way to persist the /data folder. But nothing of these pieces of information are reagrding flexdash issues.

I use docker compose to start up the container. The docker-compose.yml looks like this:

```
  nodered:
    privileged: true
    image: nodered/node-red:latest
    container_name: nodered
    restart: unless-stopped
    environment:
      - TZ=Europe/Berlin
    volumes:
    - "/opt/docker_nodered:/data"
 ```
The container is running for month without any issues. The installation of flexdash was also not a problem, but still it is not accessible from the browser. So this issue is not completed at all.