Koenkk/zigbee2mqtt #14071: Offline device is shown as on.

87insane, 2022-09-19:
### What happened?

When a zigbee LED is on and then goes offline. Will continue to show that it is on.

### What did you expect to happen?

An offline LED should be displayed as switched off.

### How to reproduce it (minimal and precise)

Switch on the LED, remove the current from the LED.

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

Koenkk, 2022-09-19:
Please check the availability feature: https://www.zigbee2mqtt.io/guide/configuration/device-availability.html

jerrm, 2022-09-21:
> shows offline and was previously powered on still stays powered on in offline mode and that is incorrect.

It is the most correct way of displaying the known data.  The best z2m can do is give last known state +  offline status.  

"Off" would be an assumption.  It may be a 99.9% correct assumption for your particular use case, but still an assumption.  What would be the correct assumption for other types of non-binary sensors?

87insane, 2022-09-28:
Hello, I would expect that a device can only be offline at once if the power is gone. The network should not show any changes on fixed powered devices.

It gets more complicated with sensors that don't have a continuous current. But a temperature sensor, for example, that no longer sends me a value for a period of x is either empty or offline. In both cases I would have the last value with a timestamp.

If that's the way it's supposed to be, that's okay with me.

closed 2022-11-07 by github-actions[bot]