Koenkk/zigbee2mqtt #14075: Crash 30s after checking new OTA update for IKEA Symfonisk Sound Remote

marazmarci, 2022-09-19:
### What happened?

I tried to check for new OTA updates for my IKEA Symfonisk Sound Remote, and Z2M crashed after 30s. Sorry, I didn't have time to test this on the dev branch.

#13809 seems like a similar issue, but I think it's different from this one.

### What did you expect to happen?

The OTA update check succeeds and shows that an update is available (I know for sure that there's a newer firmware version available).

### How to reproduce it (minimal and precise)

0. Have an IKEA Symfonisk Sound Remote
1. Pair it with Z2M
2. Go to the OTA page on the frontend
3. Click on the `Check for new updates` for the Symfonisk device
4. Wait for 30s
5. Z2M will crash and restart

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

Sonoff ZigBee 3.0 USB Dongle Plus

### Debug log

Zigbee2MQTT:info  2022-09-19 13:36:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for '0x943469ffXXXXXXXX'","meta":{"device":"0x943469ffXXXXXXXX","status":"checking_if_available"},"type":"ota_update"}'
Zigbee2MQTT:debug 2022-09-19 13:36:14: Check if update available for '0x943469ffXXXXXXXX' (SYMFONISK Sound Controller)
Zigbee2MQTT:debug 2022-09-19 13:36:14: Using endpoint '1'
Error: Timeout - 33785 - 1 - null - 25 - 1 after 30000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
Using '/app/data' as data directory
Zigbee2MQTT:debug 2022-09-19 13:36:49: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2022-09-19 13:36:49: Logging to console and directory: '/app/data/log/2022-09-19.13-36-49' filename: log.txt
Zigbee2MQTT:debug 2022-09-19 13:36:49: Removing old log directory '/app/data/log/2022-09-19.12-44-26'
Zigbee2MQTT:info  2022-09-19 13:36:49: Starting Zigbee2MQTT version 1.27.2 (commit #7dc48fb)
Zigbee2MQTT:info  2022-09-19 13:36:49: Starting zigbee-herdsman (0.14.53)

marazmarci, 2022-09-19:
And one more interesting observation:

I finished updating my IKEA Trådfri Control Outlet an hour ago, and now (after trying to check updates for the Symfonisk) it shows up like this on the OTA page:

[image]

It **finished** updating, and it's already showing that the latest firmware version is installed.

marazmarci, 2022-09-19:
I think this closed issue may be the same: #12794

Koenkk, 2022-09-19:
Can you provide the debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

marazmarci, 2022-09-19:
My `log_level` was already set to `debug`.

The part of the debug log I included in the issue description is the biggest part of the log that I deemed meaningful.

But nevertheless, here are my full logs: https://drive.google.com/drive/folders/1qikf_vVBIiKUK2YksjKxpG12_9t9z8wZ

The first one is before the crash, the second one is after.

marazmarci, 2022-09-19:
And here's (a part of) my `state.json`:

```json
{
    "0x84ba20fffe4XXXXX": { /* IKEA Styrbar Light Remote */
        "battery": 100,
        "linkquality": 80,
        "update": {
            "state": "idle"
        },
        "update_available": false
    },
    "0x84b4dbfffefXXXXX": { /* IKEA Trådfri Plug */
        "update": {
            "state": "updating"
        },
        "update_available": false,
        "linkquality": 112,
        "state": "OFF"
    },
    "0x943469fffe60XXXX": { /* IKEA Trådfri ON/OFF Switch */
        "battery": 100,
        "linkquality": 54,
        "update": {
            "state": "available"
        },
        "update_available": true
    },
    "0x943469fffe63XXXX": { /* IKEA Symfonisk Sound Controller */
        "battery": 100,
        "linkquality": 131,
        "rate": 195,
        "brightness": 37
    },
    "0x0c4314fffe7XXXXX": { /* IKEA Trådfri Shortcut Button */
        "battery": 100,
        "update": {
            "state": "available"
        },
        "update_available": true,
        "linkquality": 109
    }
}
```

marazmarci, 2022-09-19:
On my DNS server, I saw a successfully answered DNS request for `fw.ota.homesmart.ikea.net`.

Is it possible that there's some very strict rate limit that I exhausted successfully? Because I pressed the `Check for new updates` button many-many times in the last 12 hours 😄

marazmarci, 2022-09-19:
And at first, just after setting up the devices, the update check was working OK, and I was even able to start updating 3x devices, but it eventually failed. If I remember correctly what I was doing exactly, I paired at least one new device while the OTA was already running for the 3x other devices.

marazmarci, 2022-09-19:
Wow! According to your https://github.com/Koenkk/zigbee2mqtt/discussions/2921#discussioncomment-194737 from 2020, I woke the device up (by pressing one of its buttons) just before clicking on the `Check for new updates` button, and I didn't got the timeout error! The frontend now shows the `Update device firmware` button.

marazmarci, 2022-09-19:
And in the meantime, I deleted this part from `state.json` for the Trådfri Plug:
```
"update": {
    "state": "updating"
},
```
So it now doesn't constantly show the _updating in progress_ status.

marazmarci, 2022-09-21:
#11138 may be the same issue

marazmarci, 2022-09-21:
this also seems like the same issue: https://github.com/Koenkk/zigbee2mqtt/discussions/2921#discussioncomment-3423605

Koenkk, 2022-09-22:
What is the state now?

I checked your log but I don't see any errors in it:

```
info  2022-09-19 21:47:40: Checking if update available for 'IKEA Symfonisk Sound Controller'
info  2022-09-19 21:47:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'IKEA Symfonisk Sound Controller'","meta":{"device":"IKEA Symfonisk Sound Controller","status":"checking_if_available"},"type":"ota_update"}'
debug 2022-09-19 21:47:40: Check if update available for '0x943469fffe636fc4' (SYMFONISK Sound Controller)
debug 2022-09-19 21:47:40: Using endpoint '1'
```

marazmarci, 2022-09-22:
I still consistently get the crash. Even if I wake the device by pressing a button before clicking on `Check for new updates` (and continue pressing it every second).

Koenkk, 2022-09-22:
Can you provide the debug log? I expect that you get an error in the debug log.

marazmarci, 2022-09-22:
ohh, I see now, the error is not in the debug log file I've sent you... Interesting... I definitely saw the crash on the stdout. Probably the app didn't have time to finish writing the log file when it crashed?

I'll try it again tomorrow, just in case I messed up something.

marazmarci, 2022-09-24:
I tried it again, but there's no error/crash log in the log file.
I watched the Docker container's stdout, and I saw the same that I included in the issue description:
```
Zigbee2MQTT:debug 2022-09-19 13:36:14: Check if update available for '0x943469ffXXXXXXXX' (SYMFONISK Sound Controller)
Zigbee2MQTT:debug 2022-09-19 13:36:14: Using endpoint '1'
Error: Timeout - 33785 - 1 - null - 25 - 1 after 30000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)
```
But the log file ends like this:
```
Zigbee2MQTT:debug 2022-09-19 13:36:14: Check if update available for '0x943469ffXXXXXXXX' (SYMFONISK Sound Controller)
Zigbee2MQTT:debug 2022-09-19 13:36:14: Using endpoint '1'
```

So probably the app terminates before flushing the log...
I made sure that my `log_level` is set to `debug`.

marazmarci, 2022-09-24:
I tried again the method when I press the button every second until I see OTA progress, and it started magically.

However, I could consistently reproduce the crash when I clicked the `Check for new updates` button for the specific device, and I didn't wake the device by pressing a button. I think just checking for new updates shouldn't require the device to be reachable and needing to be "woken up".

And when the OTA starts, the user should be prompted beforehand to wake the device.

Koenkk, 2022-09-25:
So the crash is definitely a z2m issue. Can you provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

marazmarci, 2022-09-25:
In the meantime, by retrying the continuous button pressing approach several times, I was able to produce this log:

```
Zigbee2MQTT:error 2022-09-25 10:09:20: Failed to check if update available for 'IKEA Symfonisk Sound Controller' (Device didn't respond to OTA request)
Zigbee2MQTT:debug 2022-09-25 10:09:20: Error: Device didn't respond to OTA request
    at requestOTA (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:196:15)
    at Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:251:25)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:194:39)
```

I was able to produce this error log two times. The first time it also printed the usual timeout error message, but I couldn't catch it, because I wasn't fast enough, and it slipped out from the Portainer log viewer's buffer. The second time, it only printed the error message above. Surprisingly, it was also printed to the `log.txt` file for the first and the second occurrence (but still not the timeout error).

marazmarci, 2022-09-25:
> Can you provide the herdsman debug log of this?

Yes, I uploaded it to the [Google Drive folder](https://drive.google.com/drive/folders/1qikf_vVBIiKUK2YksjKxpG12_9t9z8wZ) I linked earlier.

Koenkk, 2022-09-28:
I don't see the herdsman debug logging in these logs. Note that this are not logged to the log files but to STDOUT only.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Note that the `Zigbee2MQTT:debug 2022-09-25 10:09:20: Error: Device didn't respond to OTA request` error is probably on the device side. Make sure to wakeup the device right before doing the update, if that doesn't help try again after removing the battery for 30 seconds.

marazmarci, 2022-09-28:
> I don't see the herdsman debug logging in these logs. Note that this are not logged to the log files but to STDOUT only.

Check the `z2m_herdsman_log.txt` 😉 I followed the instructions on how to enable the herdsman debug logging.

Koenkk, 2022-09-30:
Found the issue and fixed it, will be included in the next release tomorrow.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

marazmarci, 2022-09-30:
Wow, thanks! I'll check it out tomorrow!

satter, 2022-10-01:
I've updated to 1.28.0 today and no more crashes on OTA request timeout:
```
Oct 01 10:23:51 homie1 npm[848]: Zigbee2MQTT:debug 2022-10-01 10:23:51: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"aqara_tvoc1","transaction":"5wxkv-9"}'
Oct 01 10:23:51 homie1 npm[848]: Zigbee2MQTT:info  2022-10-01 10:23:51: Checking if update available for 'aqara_tvoc1'
Oct 01 10:23:51 homie1 npm[848]: Zigbee2MQTT:debug 2022-10-01 10:23:51: Check if update available for '0x54ef441000106dca' (lumi.airmonitor.acn01)
Oct 01 10:23:51 homie1 npm[848]: Zigbee2MQTT:debug 2022-10-01 10:23:51: Using endpoint '1'
Oct 01 10:24:51 homie1 npm[848]: Zigbee2MQTT:info  2022-10-01 10:24:51: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"aqara_tvoc1"},"error":"Failed to check if update available for 'aqara_tvoc1' (Device didn't respond to OTA request)","status":"error","transaction":"5wxkv-9"}'
Oct 01 10:24:51 homie1 npm[848]: Zigbee2MQTT:error 2022-10-01 10:24:51: Failed to check if update available for 'aqara_tvoc1' (Device didn't respond to OTA request)
Oct 01 10:24:51 homie1 npm[848]: Zigbee2MQTT:debug 2022-10-01 10:24:51: Error: Device didn't respond to OTA request
Oct 01 10:24:51 homie1 npm[848]:     at requestOTA (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:196:15)
Oct 01 10:24:51 homie1 npm[848]:     at Object.isUpdateAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:251:25)
Oct 01 10:24:51 homie1 npm[848]:     at OTAUpdate.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/otaUpdate.ts:194:39)
Oct 01 10:25:04 homie1 npm[848]: Zigbee2MQTT:debug 2022-10-01 10:25:04: Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"ikea_remote1","transaction":"5wxkv-10"}'
Oct 01 10:25:04 homie1 npm[848]: Zigbee2MQTT:info  2022-10-01 10:25:04: Checking if update available for 'ikea_remote1'
Oct 01 10:25:04 homie1 npm[848]: Zigbee2MQTT:debug 2022-10-01 10:25:04: Check if update available for '0x000d6ffffe409baa' (TRADFRI remote control)
Oct 01 10:25:04 homie1 npm[848]: Zigbee2MQTT:debug 2022-10-01 10:25:04: Using endpoint '1'
Oct 01 10:26:04 homie1 npm[848]: Zigbee2MQTT:info  2022-10-01 10:26:04: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"ikea_remote1"},"error":"Failed to check if update available for 'ikea_remote1' (Device didn't respond to OTA request)","status":"error","transaction":"5wxkv-10"}'
Oct 01 10:26:04 homie1 npm[848]: Zigbee2MQTT:error 2022-10-01 10:26:04: Failed to check if update available for 'ikea_remote1' (Device didn't respond to OTA request)
Oct 01 10:26:04 homie1 npm[848]: Zigbee2MQTT:debug 2022-10-01 10:26:04: Error: Device didn't respond to OTA request
Oct 01 10:26:04 homie1 npm[848]:     at requestOTA (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:196:15)
Oct 01 10:26:04 homie1 npm[848]:     at Object.isUpdateAvailable (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/ota/common.js:251:25)
Oct 01 10:26:04 homie1 npm[848]:     at OTAUpdate.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/otaUpdate.ts:194:39)
```

Thanks!

marazmarci, 2022-10-01:
I tried the `:latest-dev` Docker tag, and it works perfectly so far! The OTA progress started successfully, and my Symfonisk is updating ATM.

marazmarci, 2022-10-01:
Oh, no... it crashed. I'll extract the logs whenever I get home.

marazmarci, 2022-10-01:
Regarding the crash, I can only see this at the end of the log:
```
debug 2022-10-01 15:48:16: Saving state to file /app/data/state.json
info  2022-10-01 15:48:16: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
info  2022-10-01 15:48:16: Disconnecting from MQTT server
info  2022-10-01 15:48:16: Stopping zigbee-herdsman...
info  2022-10-01 15:48:25: Stopped zigbee-herdsman
info  2022-10-01 15:48:25: Stopped Zigbee2MQTT
```
And nothing else that seems unordinary.
Unfortunately, I couldn't get the logs for `zigbee-herdsman`, because that is only on `stdout`.

However, I tried to update the device again and Z2M **didn't crash**, but the OTA failed.
```
Zigbee2MQTT:error 2022-10-01 21:01:38: Update of 'IKEA Symfonisk Sound Controller' failed (Device didn't respond to OTA request)
Zigbee2MQTT:debug 2022-10-01 21:01:38: Error: Device didn't respond to OTA request
    at requestOTA (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:196:15)
    at Object.updateToLatest (/app/node_modules/zigbee-herdsman-converters/lib/ota/common.js:283:21)
    at OTAUpdate.onMQTTMessage (/app/lib/extension/otaUpdate.ts:257:21)
```
It seems like a different problem.

closed 2022-12-04 by marazmarci