Koenkk/zigbee2mqtt #14081: Tuya TV02 heating_stop resets temperature preset

voed, 2022-09-19:
### What happened?

After setting heating_stop to on and off again, current_heating_setpoint resets to 20 °C. Not sure if it can be fixed but i tried with tuya hub and it is resetting too, but to 17 °C instead.

### How to reproduce it (minimal and precise)

1. Set current_heating_setpoint  to 19°C
2. Turn on heating_stop
3. Turn off heating stop
4. Now current_heating_setpoint is 20°C

### Zigbee2MQTT version

1.27.2

### Adapter firmware version

20220219

### Adapter

CC2652R

voed, 2022-09-19:
Guess i should open it in converters repo.

voed, 2022-09-19:
Um, or not. Anyways, here's what i found: when heat_stop is enabled, i can click on preset selector(manual, auto etc..), heat stop will be disabled without resetting the temperature. But if i click on heat_stop switch instead, it resets temperature to default state of previous preset(17 for auto or 20 for manual). So i think we can just remove [this line](https://github.com/Koenkk/zigbee-herdsman-converters/blob/51c60c455533343b13869652389d856ec9f51ba8/converters/toZigbee.js#L3487) to fix the issue. Also maybe try to read current preset instead of just setting 'manual' here?
@vladi1234 take a look please

voed, 2022-09-19:
Also changing `current_heating_setpoint` when `heating_stop` is enabled just resetting `current_heating_setpoint` to 20 so i guess [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/51c60c455533343b13869652389d856ec9f51ba8/converters/toZigbee.js#L3513) we should change mode first and then change heating setpoint.

vladi1234, 2022-09-20:
Hello @voed, 
it is due to the internal TVR function as soon as the value or parameter has changed, then TVR always changes to Auto_Mode.

It is only possible in this way. Otherwise it didn't work.

closed 2022-10-31 by github-actions[bot]

hans9t9, 2023-02-02:
> Hello @voed, it is due to the internal TVR function as soon as the value or parameter has changed, then TVR always changes to Auto_Mode.
> 
> I have examined this effect again today and found a solution that can handle these effects.

Please can you let me know what your solution is? I experience the same behaviour of the TVR.

voed, 2023-02-02:
@hans9t9 instead of turning on just set preset mode to manual. It will turn on without resetting the temperature.

hans9t9, 2023-02-02:
Thanks @voed, I tried it out right away and this works!