Koenkk/zigbee2mqtt #14085: [New device support]: Aeotec Range Extender Zi - Model ZGA001-C15 (EU)

jozk, 2022-09-19:
### Link

https://aeotec.freshdesk.com/support/solutions/articles/6000248297-aeotec-range-extender-zi-technical-specifications

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0xe0798dfffee09c8a","nwkAddr":16779,"manufId":4880,"manufName":"AL001","powerSource":"Mains (single phase)","modelId":"WG001-Z01","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,3],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"WG001-Z01","manufacturerName":"AL001","powerSource":1,"zclVersion":8,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","swBuildId":""}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":2080,"inClusterList":[0,3,4096],"outClusterList":[3,6,8,4096],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":"","swBuildId":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","swBuildId":"","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1663604024661,"defaultSendRequestWhen":"immediate"}

### Comments

Log Message: Received message from unsupported device with Zigbee model 'WG001-Z01' and manufacturer name 'AL001'
This is a repeater I tried to add.  I also have installed the latest version of zigbee2mqtt 1.27.2-1
https://store.aeotec.com/products/range-extender-zi-zga001

Device type: Router
Zigbee Model WG001-Z01
Zigbee Manufacturer: AL001
Support status: Not supported
IEEE Address: 0xe0798dfffee09c8a
Network address: 0x418B

Database:
{"id":15,"type":"Router","ieeeAddr":"0xe0798dfffee09c8a","nwkAddr":16779,"manufId":4880,"manufName":"AL001","powerSource":"Mains (single phase)","modelId":"WG001-Z01","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,3],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"WG001-Z01","manufacturerName":"AL001","powerSource":1,"zclVersion":8,"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","swBuildId":""}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":2080,"inClusterList":[0,3,4096],"outClusterList":[3,6,8,4096],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":"","swBuildId":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","swBuildId":"","zclVersion":8,"interviewCompleted":true,"meta":{},"lastSeen":1663604024661,"defaultSendRequestWhen":"immediate"}

jeremynieuwpoort, 2022-09-25:
Is it supported right now? Tried to install a converter but did not work well for me.

expresscisco, 2022-10-01:
Hello. I just posted right now about this device. I’ll be very interested to have it working as well.

Koenkk, 2022-10-01:
`WG001-Z01` will be supported in the next release which will come in 1 hour from now.

expresscisco, 2022-10-01:
Wow! I feel like you just did it for me :-) Thank you so much !!!!

expresscisco, 2022-10-01:
It works !! Well, it correctly shows up, but I can't get my lumi leak sensors to go through it, they still prefer the coordinator. is that a different issue?

Koenkk, 2022-10-02:
Great, xiaomi devices are known to be stubborn. Z2m doesn't have any influence in what device pick as their parent. It may help to re-pair it very close to the desired coordinator.

closed 2022-10-02 by Koenkk

jozk, 2022-10-11:
Just to say thank you for the fix, much appreciated.Joe Klimkowski

expresscisco, 2022-10-11:
Hello,

I confirm the device seems to work, and only one of my six aqara leak detectors has catched it as a repeater (aka the lucky one :-) )

l7s, 2024-01-31:
Hi, unfortunately it looks like `zigbee2mqtt` version as below:

```
commit 56589dccf156915dcc78e9c7ffd669d7e5597db0 (grafted, HEAD -> master, origin/master, origin/HEAD)
Author: koenkk <koenkanters94@gmail.com>
Date:   Sat Dec 2 13:10:03 2023 +0100

    chore: automate HA addon updates
```

still doesn't support `Aeotec ZGA001-C15 Range Extender Zi` 

I just paired on of these to my `zigbee2mqtt`

[image]

and it ended in error as below:

[image]

Additionally I note search for `ZGA001` at https://www.zigbee2mqtt.io/supported-devices/ yields no results.

Is there anything I can do to help making  `ZGA001` compatible?

Regards,
Chris

FROCHE38, 2024-06-12:
Little up because same issue on my side :-(

thomaskonrad, 2024-06-21:
Same here — I see the exact same thing as @l7s in his comment https://github.com/Koenkk/zigbee2mqtt/issues/14085#issuecomment-1918913848

baststar, 2024-10-26:
just got mine from amazon.de and it is a ZGA001-C15 and cannot pair it

baststar, 2024-10-26:
i just got my zga001-c15 from amazon.de and it seemed to not work at first pairing-attempts. resettet the repeater (press 10 seconds the button), tried again... activated "permit join" again and put the extender into the wall and pushed the button.. than it works (doesnt know exactly the order - but it works without any issue. 
it says its supported, doesnt have this issue https://github.com/Koenkk/zigbee2mqtt/issues/14085#issuecomment-1918913848

epoc83, 2024-10-31:
> Hi, unfortunately it looks like `zigbee2mqtt` version as below:
> 
> ```
> commit 56589dccf156915dcc78e9c7ffd669d7e5597db0 (grafted, HEAD -> master, origin/master, origin/HEAD)
> Author: koenkk <koenkanters94@gmail.com>
> Date:   Sat Dec 2 13:10:03 2023 +0100
> 
>     chore: automate HA addon updates
> ```
> 
> still doesn't support `Aeotec ZGA001-C15 Range Extender Zi`
> 
> I just paired on of these to my `zigbee2mqtt`
> 
> ![Screenshot_3](https://private-user-images.githubusercontent.com/5430068/301141876-a2126a61-f743-488b-ac1c-c1a3b3e20842.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAzNjgwNTgsIm5iZiI6MTczMDM2Nzc1OCwicGF0aCI6Ii81NDMwMDY4LzMwMTE0MTg3Ni1hMjEyNmE2MS1mNzQzLTQ4OGItYWMxYy1jMWEzYjNlMjA4NDIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMzFUMDk0MjM4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTc0NjA2Y2NlNWYxOWY2Mzc5MzNjMzY4ZGRkZWI3ZWMxZWEwMTk0MGVlOThiMDNmZTRlOWYyMGZkMDdlZmIyYSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.aVndFCKcwwYpjgqDpmH3Pm96iDNVTY7MQlwjoo1baIE)
> 
> and it ended in error as below:
> 
> ![Screenshot_2](https://private-user-images.githubusercontent.com/5430068/301142064-fe479680-57ae-4766-9b50-2f3092d33764.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAzNjgwNTgsIm5iZiI6MTczMDM2Nzc1OCwicGF0aCI6Ii81NDMwMDY4LzMwMTE0MjA2NC1mZTQ3OTY4MC01N2FlLTQ3NjYtOWI1MC0yZjMwOTJkMzM3NjQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAzMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMzFUMDk0MjM4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZmNjMTIzYThlYTc4MDQyOTRiMjg3MTk5ZWMzMDE1NWQ3NzYwNzQ3MmJhZGM0MDI5OWEwMTIxNWUyMWYwZTRiMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.y9MIicDZq4JR83fKZjKPyWpK4ios9EHPg0EM0Fg6iys)
> 
> Additionally I note search for `ZGA001` at https://www.zigbee2mqtt.io/supported-devices/ yields no results.
> 
> Is there anything I can do to help making `ZGA001` compatible?
> 
> Regards, Chris

I’m having the same problem. I’m not able to use it. After several tries I managed to sync the device, but silently drops connection and makes the whole signee network unstable.

FROCHE38, 2024-11-02:
I finally found the solution and share it with you:

1) Add the file named ZGA001.js to your \\<ip>\config\zigbee2mqtt folder

```
const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['ZGA001'],
    model: 'ZGA001',
    vendor: 'AEOTEC',
    description: 'Aeotec Range Extender Zi-C15 (EU)',
    extend: [],
    fromZigbee: [fz.linkquality_from_basic],
    toZigbee: [],
    exposes: [],
};

module.exports = definition;
```

2) Then on Zigbee2MQTT parameters, declare this file on Settings > External Converters
[image]

3) Delete all previous failed paring with ZGA001

4) Restart Zigbee2MQTT addons

5) Pair again

It works (even if the name is not correct)
[image]

epoc83, 2024-11-10:
> I finally found the solution and share it with you:
> 
> 1. Add the file named ZGA001.js to your \\-your ip-\config\zigbee2mqtt folder (this is for HomeAssistant)
> 
> ```
> const {} = require('zigbee-herdsman-converters/lib/modernExtend');
> // Add the lines below
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const utils = require('zigbee-herdsman-converters/lib/utils');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const definition = {
>     zigbeeModel: ['ZGA001'],
>     model: 'ZGA001',
>     vendor: 'AEOTEC',
>     description: 'Aeotec Range Extender Zi-C15 (EU)',
>     extend: [],
>     fromZigbee: [fz.linkquality_from_basic],
>     toZigbee: [],
>     exposes: [],
> };
> 
> module.exports = definition;
> ```
> 
>  2. Then on Zigbee2MQTT parameters, declare this file on Settings > External Converters
>    ![image](https://private-user-images.githubusercontent.com/37023141/382460417-5890ddd3-37aa-4993-852d-e5fbc36219c7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEyNzQ3NTIsIm5iZiI6MTczMTI3NDQ1MiwicGF0aCI6Ii8zNzAyMzE0MS8zODI0NjA0MTctNTg5MGRkZDMtMzdhYS00OTkzLTg1MmQtZTVmYmMzNjIxOWM3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEwVDIxMzQxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThjM2FlYjQ2M2I1MGVlMDA0OTUwOWViZmJiYmMxZDU5ZGNkYzNjNDJlNzVmYWU2MTZlMGQ0OWQxMmQwNjNiODUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.8iePCQjfR4UUPs4RooE_1Ae3SYpk3kWOdZU8j0eaZqc)
> 3. Delete all previous failed paring with ZGA001
> 4. Restart Zigbee2MQTT addons
> 5. Pair again
> 
> It works (even if the name is not correct) ![image](https://private-user-images.githubusercontent.com/37023141/382460510-83940ca7-e1f9-463c-9883-dcf55084e55f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEyNzQ3NTIsIm5iZiI6MTczMTI3NDQ1MiwicGF0aCI6Ii8zNzAyMzE0MS8zODI0NjA1MTAtODM5NDBjYTctZTFmOS00NjNjLTk4ODMtZGNmNTUwODRlNTVmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEwVDIxMzQxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM0ZmE4ZDJjMjZlZTIwN2QwNzdmNGZiMWMwZDQwYmMyNjcwYzVhZWViMTJmZGFhZjE4YzQ2NWI0MGRiZjczYWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.bAeyPxsh5NVhRMGorljme4h1Srd9yxDFcTLiSdkSlsI)

Thanks for the information, but it didn’t worked for me. After adding the file and try to get recognized I get the same error and now the device cannot get recognized anymore. It just says that it isn’t supported and it doesn’t even show the brand. I tried to erase the file, but nothing…
[image]
[image]

Pofilo, 2025-03-26:
I had it, then last week it stopped sending anything to my coordinator.
I tried to repair it without any success so far ...

The solution provided in https://github.com/Koenkk/zigbee2mqtt/issues/14085#issuecomment-2452928093
is not working either.