Koenkk/zigbee2mqtt #14091: [New device support]:

mihaimdinca, 2022-09-20:
### Link

https://nous.technology/product/lz3.html

### Database entry

{"id":38,"type":"Router","ieeeAddr":"0xa4c1386f35161d4a","nwkAddr":268,"manufId":4417,"manufName":"_TZ3000_h8ngtlxy","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"57344":{"attributes":{"53251":"AAAA"}},"genBasic":{"attributes":{"modelId":"TS0001","manufacturerName":"_TZ3000_h8ngtlxy","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"32768":0,"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c94507","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1663644340044,"defaultSendRequestWhen":"immediate"}

### Comments

The valve works but gets recognized as Tuya Switch TS0001. I get two entities on/ off and link quality. I was hoping to get powerOn behavior as well.

closed 2022-10-29 by github-actions[bot]

mihaimdinca, 2022-12-02:
The issue is mostly solved. Z2M does not recognize the device as a valve but poweOn behavior option is available now so practically everything works.

freedomlives, 2024-06-25:
Well, you had luck. I just bought one, and Z2M doesn't recognize it at all!

mihaimdinca, 2024-06-25:
Actually now the devices (I have multiple ones) are fully supported. They show up as Tuya valve

freedomlives, 2024-06-25:
I'm upgrading to 1.38.0, maybe 1.32 from last fall was too old.

freedomlives, 2024-06-25:
That fixed everything! Now if only it worked with a PPR plastic quarter turn valve. Installing tomorrow a brass one. :-/