Koenkk/zigbee2mqtt #14104: Exception in HA when receiving discovery message

Antonasg, 2022-09-20:
### What happened?

When HA receives an availability message on mqtt it throws an exception:
```
Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/Office-balcony-door/availability'}], 'availability_mode': 'all', 'device': {'configuration_url': '10.0.1.5:8181/#/device/0x00158d000862880c/info', 'identifiers': ['zigbee2mqtt_0x00158d000862880c'], 'manufacturer': 'Xiaomi', 'model': 'Aqara door & window contact sensor (MCCGQ11LM)', 'name': 'Office-balcony-door', 'sw_version': '3000-0001'}, 'device_class': 'voltage', 'enabled_by_default': False, 'entity_category': 'diagnostic', 'json_attributes_topic': 'zigbee2mqtt/Office-balcony-door', 'name': 'Office-balcony-door voltage', 'state_class': 'measurement', 'state_topic': 'zigbee2mqtt/Office-balcony-door', 'unique_id': '0x00158d000862880c_voltage_zigbee2mqtt', 'unit_of_measurement': 'mV', 'value_template': '{{ value_json.voltage }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 350, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 232, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 355, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.10/site-packages/voluptuous/validators.py", line 351, in _exec v = func(v) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 818, in validate_callable return schema(data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 595, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.10/site-packages/voluptuous/schema_builder.py", line 433, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: invalid url for dictionary value @ data['device']['configuration_url']
```
zwave2mqtt works fine with HA and is able to discovery and add new devices.
I have done a clean install of HA but the issue persists.
I am running HA 2022.9.5

### What did you expect to happen?

Devices in zigbee2mqtt to be added to Home Assitant.

### How to reproduce it (minimal and precise)

Install mosquito, zigbee2mqtt and HA.
Enable the mqtt integration.

### Zigbee2MQTT version

2.27.2

### Adapter firmware version

0x26660700

### Adapter

ConBee 2

### Debug log

 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/9609577/log.txt)

Antonasg, 2022-09-20:
@IamMattM seems like you had similar problem related to this [issue](https://github.com/home-assistant/core/issues/24306). Did you find a solution?

IamMattM, 2022-09-20:
> @IamMattM seems like you had similar problem related to this [issue](https://github.com/home-assistant/core/issues/24306). Did you find a solution?

It has been a while...I did solve the issue since that partuclar zigbee device is correcly in use in HA.

Can't be sure 100% but was it something to do with enabling zigbee2mqtt/homeassistant "legacy" entity attributes and trigger ?

Something to do with the fact that sensor report itself as just a "sensor" ?

Try it ....

[image]

 

[image]

Antonasg, 2022-09-20:
Thanks for the reply, but unfortunately I have already tried toggling those on and off.

IamMattM, 2022-09-20:
> Thanks for the reply, but unfortunately I have already tried toggling those on and off.

Hmmm.I'll try to remember what I did then...DEfinitely nothing to do with .db deletion as it had been suggested...

IamMattM, 2022-09-20:
One other thing that I  did (amongst many other things...) was to update the conbee II if that is what your are using. I was under Windows:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-windows

Antonasg, 2022-09-20:
Found the solution!

In Frontend settings, there's an option to put in a URL, used for HA device configuration page.
I had put in `[serverip]:8181`. Removing it solved the issue.

[image]

IamMattM, 2022-09-20:
That's it...you're right . I think I also had filled this URL with the z2m frint end address.(since it says it is for homeassistant why not ;-))

Well done .
[image]

Antonasg, 2022-09-21:
Yeah, don't know why I didn't check this earlier based on the HA exception logs.

I had the URL with port at the end - [ipaddr]:8181 - that might have been the issue.

In any case perhaps a bug should be reported.

closed 2022-10-30 by github-actions[bot]