Koenkk/zigbee2mqtt #14107: LELLKI WP30-EU not changing states in Home Assistant when pressing buttons on power strip

TeHashX, 2022-09-20:
### What happened?

I press physical button on power strip

### What did you expect to happen?

To change correct state in Home Assistant

### How to reproduce it (minimal and precise)

Don't happed even if I press refresh states under Exposes tab

### Zigbee2MQTT version

1.27.2-dev commit: 9b4a03c

### Adapter firmware version

6.10.3.0 build 297

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Debug log

debug 2022-09-20 20:32:47: Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
info  2022-09-20 20:32:47: MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-20T20:32:47+03:00","linkquality":216,"power":0,"power_outage_memory":"restore","state":"OFF","state_l1":"ON","state_l2":"ON","state_l3":"ON","voltage":233}'
debug 2022-09-20 20:32:50: Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 3 with groupID 0
info  2022-09-20 20:32:50: MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-20T20:32:50+03:00","linkquality":144,"power":0,"power_outage_memory":"restore","state":"OFF","state_l1":"ON","state_l2":"ON","state_l3":"ON","voltage":233}'
debug 2022-09-20 20:32:53: Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-09-20 20:32:53: MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-20T20:32:53+03:00","linkquality":200,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"ON","state_l2":"ON","state_l3":"ON","voltage":233}'
debug 2022-09-20 20:32:55: Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
info  2022-09-20 20:32:55: MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-20T20:32:55+03:00","linkquality":208,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"ON","state_l2":"ON","state_l3":"ON","voltage":233}'
debug 2022-09-20 20:32:56: Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 3 with groupID 0
info  2022-09-20 20:32:56: MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-20T20:32:56+03:00","linkquality":204,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"ON","state_l2":"ON","state_l3":"ON","voltage":233}'
debug 2022-09-20 20:33:18: Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genBasic', data '{"65506":54,"65508":0,"appVersion":77}' from endpoint 1 with groupID 0
info  2022-09-20 20:33:18: MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-20T20:33:18+03:00","linkquality":208,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"ON","state_l2":"ON","state_l3":"ON","voltage":233}'

TeHashX, 2022-09-21:
Thank you, unfortunately still the same

`Info 2022-09-21 18:48:34MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:48:33+03:00","linkquality":164,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":240}'
Debug 2022-09-21 18:48:34Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2022-09-21 18:48:34MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:48:33+03:00","linkquality":164,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":240}'
Debug 2022-09-21 18:48:34Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2022-09-21 18:48:34MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:48:33+03:00","linkquality":164,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":240}'
Debug 2022-09-21 18:48:34Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 3 with groupID 0
Info 2022-09-21 18:48:34MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:48:33+03:00","linkquality":164,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":240}'
Debug 2022-09-21 18:48:34Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2022-09-21 18:48:34MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:48:33+03:00","linkquality":164,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":240}'
Debug 2022-09-21 18:48:34Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 3 with groupID 0
Info 2022-09-21 18:48:34MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:48:33+03:00","linkquality":164,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":240}'
Debug 2022-09-21 18:48:37Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2022-09-21 18:48:37MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:48:36+03:00","linkquality":164,"power":0,"power_outage_memory":"restore","state":"OFF","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":240}'
Debug 2022-09-21 18:48:37Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info 2022-09-21 18:48:37MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:48:36+03:00","linkquality":164,"power":0,"power_outage_memory":"restore","state":"OFF","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":240}'
Debug 2022-09-21 18:48:39Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
Info 2022-09-21 18:48:39MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:48:38+03:00","linkquality":196,"power":0,"power_outage_memory":"restore","state":"OFF","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":240}'
Debug 2022-09-21 18:48:39Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
Info 2022-09-21 18:48:39MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:48:38+03:00","linkquality":196,"power":0,"power_outage_memory":"restore","state":"OFF","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":240}'
Debug 2022-09-21 18:48:40Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 3 with groupID 0
Info 2022-09-21 18:48:40MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:48:39+03:00","linkquality":200,"power":0,"power_outage_memory":"restore","state":"OFF","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":240}'
Debug 2022-09-21 18:48:40Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 3 with groupID 0
Info 2022-09-21 18:48:40MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:48:39+03:00","linkquality":200,"power":0,"power_outage_memory":"restore","state":"OFF","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":240}'
Debug 2022-09-21 18:48:44Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 3 with groupID 0
[... cut ...]
Debug 2022-09-21 18:49:10Publishing get 'get' 'state' to 'Lellki Edge'
Debug 2022-09-21 18:49:10Received Zigbee message from 'Lellki Edge', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 3 with groupID 0
Info 2022-09-21 18:49:10MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T18:49:09+03:00","linkquality":192,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"ON","state_l2":"ON","state_l3":"ON","voltage":239}'`

Koenkk, 2022-09-21:
Can you provide the debug log of just one toggle on the device?

TeHashX, 2022-09-21:
Sure, button 1

`Debug 2022-09-21 22:32:24Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2022-09-21 22:32:24MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T22:32:24+03:00","linkquality":188,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"ON","state_l2":"ON","state_l3":"ON","voltage":233}'
Debug 2022-09-21 22:32:24Received Zigbee message from 'Lellki Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
Info 2022-09-21 22:32:24MQTT publish: topic 'zigbee2mqttedge/Lellki Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-21T22:32:24+03:00","linkquality":188,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"ON","state_l2":"ON","state_l3":"ON","voltage":233}'`

Koenkk, 2022-09-22:
Did you turn on channel 1 of the switch with this action? If yes, then it seems that `state_l1` updates correctly (it shows `ON`)

TeHashX, 2022-09-22:
Yes, I turned on channel 1 but state didn't change in home assistant.
If I click switches in HA, will turn them on/off but if I press buttons on power strip, no change in states. it's kind of unidirectional :(

TeHashX, 2022-09-22:
Now I turned off all switches from HA and pressed the channel 2 on strip, maybe this log helps debugging:

`Debug 2022-09-22 21:04:47Received Zigbee message from 'Lellki Z2M Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2022-09-22 21:04:47MQTT publish: topic 'zigbee2mqttedge/Lellki Z2M Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-22T21:04:47+03:00","linkquality":200,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":232}'
Debug 2022-09-22 21:04:47Received Zigbee message from 'Lellki Z2M Edge', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2022-09-22 21:04:47MQTT publish: topic 'zigbee2mqttedge/Lellki Z2M Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-22T21:04:47+03:00","linkquality":200,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":232}'
Debug 2022-09-22 21:04:49Received Zigbee message from 'Lellki Z2M Edge', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":230}' from endpoint 1 with groupID 0
Info 2022-09-22 21:04:49MQTT publish: topic 'zigbee2mqttedge/Lellki Z2M Edge', payload '{"current":0,"energy":0,"last_seen":"2022-09-22T21:04:48+03:00","linkquality":156,"power":0,"power_outage_memory":"restore","state":"ON","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF","voltage":233}'`

Koenkk, 2022-09-22:
>  pressed the channel 2 on strip

So this turned channel 2 (`l2`) from OFF to ON right?

TeHashX, 2022-09-22:
Yes

Koenkk, 2022-09-22:
I found the issue, please check with the external converter I updated in https://github.com/Koenkk/zigbee2mqtt/issues/14107#issuecomment-1253868358

TeHashX, 2022-09-22:
Yes, thank you very very much for your hard work, I was about to return the strip but it's working as is should now!

TeHashX, 2022-09-22:
Unfortunately, I lost power consumptions with this external converter enabled, please advice.

TeHashX, 2022-09-23:
With converter
`{
    "current": 0,
    "current_l1": 0.04,
    "energy": 0.05,
    "last_seen": "2022-09-23T19:37:42+03:00",
    "linkquality": 216,
    "power": 0,
    "power_l1": 6,
    "power_outage_memory": "restore",
    "power_outage_memory_l1": "restore",
    "state": "ON",
    "state_l1": "ON",
    "state_l2": "ON",
    "state_l3": "ON",
    "voltage": 237,
    "voltage_l1": 235
}`

Without
`{
    "current": 0.04,
    "current_l1": 0.04,
    "energy": 0.05,
    "last_seen": "2022-09-23T19:41:14+03:00",
    "linkquality": 220,
    "power": 6,
    "power_l1": 6,
    "power_outage_memory": "restore",
    "power_outage_memory_l1": "restore",
    "state": "ON",
    "state_l1": "ON",
    "state_l2": "ON",
    "state_l3": "ON",
    "voltage": 236,
    "voltage_l1": 235
}`

Koenkk, 2022-09-23:
I updated https://github.com/Koenkk/zigbee2mqtt/issues/14107#issuecomment-1253868358 , can you check again?

TeHashX, 2022-09-23:
Yes, seams to work now, I will make further tests before closing issue, thanks again!

Koenkk, 2022-09-25:
Great, integrated the changes.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

TeHashX, 2022-09-25:
All ok, thank you!

closed 2022-09-25 by TeHashX