Koenkk/zigbee2mqtt #14108: [New device support]: Salus TRV10RFM

dennisgranasen, 2022-09-20:
### Link

https://salus-controls.com/uk/product/trv10rfm/

### Database entry

{"id":24,"type":"EndDevice","ieeeAddr":"0x001e5e09020de1c4","nwkAddr":61005,"manufId":4216,"manufName":"Computime","powerSource":"Battery","modelId":"SAL6EM1","epList":[6],"endpoints":{"6":{"profId":260,"epId":6,"devId":2,"inClusterList":[0,3,64512],"outClusterList":[25,64512],"clusters":{"genBasic":{"attributes":{"modelId":"SAL6EM1","manufacturerName":"Computime","powerSource":3,"zclVersion":1,"appVersion":0,"stackVersion":5,"hwVersion":255,"dateCode":"20210925"}},"genOnOff":{"attributes":{}},"manuSpecificUbisysDeviceSetup":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":5,"hwVersion":255,"dateCode":"20210925","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1663701311023,"defaultSendRequestWhen":"immediate"}

### Comments

I seem to get the device to pair (as far as I can tell), but the only clusters exposed are genIdentity, genOta and **manuSpecificUbisysDeviceSetup**. It sends some data sequences that I cannot parse, most of them short like below. Occasionally a comparatively long sequence with mostly values around 48-52. 

```
Debug 2022-09-20 21:14:05Received Zigbee message from '0x001e5e09020de1c4', type 'raw', cluster 'manuSpecificUbisysDeviceSetup', data '{"data":[13,120,16,29,26,1,0,245,110],"type":"Buffer"}' from endpoint 6 with groupID null
Info 2022-09-20 21:14:05MQTT publish: topic 'zigbee2mqtt/0x001e5e09020de1c4', payload '{"last_seen":"2022-09-20T21:14:06+02:00","linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'
Debug 2022-09-20 21:14:10Received Zigbee message from '0x001e5e09020de1c4', type 'raw', cluster 'manuSpecificUbisysDeviceSetup', data '{"data":[13,120,16,30,169],"type":"Buffer"}' from endpoint 6 with groupID null
Info 2022-09-20 21:14:10MQTT publish: topic 'zigbee2mqtt/0x001e5e09020de1c4', payload '{"last_seen":"2022-09-20T21:14:11+02:00","linkquality":255,"state":"ON","update":{"state":"idle"},"update_available":false}'
```

Any suggestions on where to go from here would be greatly appreciated.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SAL6EM1'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'TRV10RFM', // Vendor model number, look on the device for a model number
    vendor: 'Salus', // Vendor of the device (only used for documentation and startup logging)
    description: 'Salus TRV', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.battery, fz.on_off], // We will add this later
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.switch()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    //manuSpecificUbisysDeviceSetup
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(9);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'seMetering']);
        await reporting.onOff(endpoint);
        await reporting.instantaneousDemand(endpoint, {min: 5, change: 10});
        await reporting.currentSummDelivered(endpoint, {min: 5, change: [0, 10]});
        await endpoint.read('seMetering', ['multiplier', 'divisor']);
    },
    ota: ota.salus,
};

module.exports = definition;
```

dennisgranasen, 2022-09-20:
Here's a little sequence of bytes that the device spits out via the manuSpecificUbisysDeviceSetup cluster

<html>
<body>
<table border="0" cellpadding="0" cellspacing="0" width="262" style="border-collapse: collapse; width: 197pt;"><colgroup><col width="23" style="width: 17pt;"><col width="21" span="2" style="width: 16pt;"><col width="23" span="2" style="width: 17pt;"><col width="21" span="2" style="width: 16pt;"><col width="23" span="2" style="width: 17pt;"><col width="21" span="3" style="width: 16pt;"></colgroup><tbody><tr height="20" style="height: 15pt;"><td height="20" width="23" style="font-family: Calibri, sans-serif; margin: 0px; height: 15pt; width: 17pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">0D</td><td width="21" style="font-family: Calibri, sans-serif; margin: 0px; width: 16pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">78</td><td width="21" style="font-family: Calibri, sans-serif; margin: 0px; width: 16pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">10</td><td width="23" style="font-family: Calibri, sans-serif; margin: 0px; width: 17pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">59</td><td width="23" style="font-family: Calibri, sans-serif; margin: 0px; width: 17pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">10</td><td width="21" style="font-family: Calibri, sans-serif; margin: 0px; width: 16pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">FF</td><td width="21" style="font-family: Calibri, sans-serif; margin: 0px; width: 16pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">87</td><td width="23" style="font-family: Calibri, sans-serif; margin: 0px; width: 17pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">00</td><td width="23" style="font-family: Calibri, sans-serif; margin: 0px; width: 17pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">40</td><td width="21" style="font-family: Calibri, sans-serif; margin: 0px; width: 16pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">00</td><td width="21" style="font-family: Calibri, sans-serif; margin: 0px; width: 16pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">91</td><td width="21" style="font-family: Calibri, sans-serif; margin: 0px; width: 16pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">00</td></tr><tr height="20" style="height: 15pt;"><td height="20" style="font-family: Calibri, sans-serif; margin: 0px; height: 15pt; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">0D</td><td style="font-family: Calibri, sans-serif; margin: 0px; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">78</td><td style="font-family: Calibri, sans-serif; margin: 0px; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">10</td><td style="font-family: Calibri, sans-serif; margin: 0px; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">5A</td><td style="font-family: Calibri, sans-serif; margin: 0px; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">1A</td><td style="font-family: Calibri, sans-serif; margin: 0px; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">01</td><td style="font-family: Calibri, sans-serif; margin: 0px; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">00</td><td style="font-family: Calibri, sans-serif; margin: 0px; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">76</td><td style="font-family: Calibri, sans-serif; margin: 0px; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">BB</td><td style="font-family: Calibri, sans-serif; margin: 0px; padding-top: 1px; pad [... cut ...] er-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial; padding-top: 1px; padding-right: 1px; padding-left: 1px; color: black; font-size: 11pt; vertical-align: bottom; border: none; white-space: nowrap;">A9</td><br class="Apple-interchange-newline">
</body>
</html>

closed 2022-10-29 by github-actions[bot]

wizu, 2023-11-27:
@dennisgranasen  Did you manage to get it working?

dennisgranasen, 2023-11-27:
Sadly no, I gave up and sold my kit. I now use Aqara TRV's instead.