Koenkk/zigbee2mqtt #14115: Moes BRT-100-TRV does not update current_heating_temperature on mode switch

Udeson, 2022-09-20:
### What happened?

When switching from manual to programming mode (sending payload "preset": "programming") current_heating_temperature do not change/update to the scheduled temperature in the zigbee2mqtt frontend

current_heating_temperature updates (on zigbee2mqtt frontend) in the moment touching the TVR surface.

Finding: When changing the mode on the TVR itself no single payload is sent but the whole command set.

### What did you expect to happen?

When switched to programming mode via frontend (not changing the mode on the TVR itself!) current_heating_temperature should update to scheduled temperature.

### How to reproduce it (minimal and precise)

Change TRV mode via SmartHome (FHEM) front end via sending payload "preset": "programming".

### Zigbee2MQTT version

1.27.0 commit: a9b8808

### Adapter firmware version

20211217

### Adapter

Electrorama zig-a-zig-ah!

### Debug log

[logfile.txt](https://github.com/Koenkk/zigbee2mqtt/files/9611569/logfile.txt)

Devicename: zigbee2mqtt/SCH_ZIG_THERMOSTAT

rotilho, 2022-09-21:
I had BRT-100-TRV and returned after I realized the calibration didn't trig any update. It wasn't Z2M problem but the TRV itself

Udeson, 2022-09-21:
> I had BRT-100-TRV and returned after I realized the calibration didn't trig any update. It wasn't Z2M problem but the TRV itself

Sounds like bad news. On the other side wondering that there are many people here using this TRV model with no problems(?). But not updating the current_heating_setpoint to the programmed/schedule temperature is a deal breaker to me.

rotilho, 2022-09-22:
I don't think so. It's a firmware defect and last time I checked it's not patched

Udeson, 2022-09-23:
> I don't think so. It's a firmware defect and last time I checked it's not patched

So you are thinking the mentioned behaviour will be fixed via firmware ota update?

willhaggan, 2022-09-24:
well it should work its based on tuya device which work from experienece of other devices

closed 2022-11-03 by github-actions[bot]