Koenkk/zigbee2mqtt #14122: Aqara P1 Support problem

robkarpati, 2022-09-21:
### What happened?

I am added an Aqara P1 (ms-s02) to my Zigbee2mqtt in Home Assistant. 
Pairing went well but it says it is Not Supported. (Edge version is the same)
All addons and HA is fresh. 

### What did you expect to happen?

I would like to make it work.

### Zigbee2MQTT version

1.22.0

### Adapter firmware version

20210120

### Adapter

Silicon_Labs_CP2102N

jerrm, 2022-09-21:
The RTCGQ14LM was added in 1.25.1. Your older than that, but edge should have been current.  What version shows for edge?

Also, the 20210120 coordinator firmware is VERY old at this point.  Best to upgrade.  I don't know about this device, but there are some devices that won't work with older firmware.

robkarpati, 2022-09-22:
@jerrm Thank you! Edge version has no version number but the newest. Yes, I will upgrade my device soon. 
If I upgrade the wf all device stay paired, do you know it?

jerrm, 2022-09-22:
Yes, z2m should restore the stick after firmware flash.

robkarpati, 2022-09-22:
Thanks a lot!

jerrm ***@***.***> (időpont: 2022. szept. 22., Cs, 18:56) ezt
írta:

robkarpati, 2022-09-27:
Hi, I am back :(
I updated the stick, now it runs with 20220219 but P1 still showed as unsupported. On normal and edge version also.
Any more tricks or to do?
Thank you!

closed 2022-11-05 by github-actions[bot]

Jens-Wymeersch, 2022-11-09:
@robkarpati were you able to get a solution ? I'm not getting all the readings for the P1