Koenkk/zigbee2mqtt #14123: TuYa TS130F - Wrong state since 7dc48fb6

Gorouk, 2022-09-21:
### What happened?

Hello,

Since commit 7dc48fb6, I noticed that my TuYa TS130F are returning the opposite state of what they are really.
Everything is still working as intended, when I close, it closes, when I open, it opens, but the states are the opposite of what it should be.
eg. the cover states it is open but it is fully closed, I still can open it and it opens. When fully opens, its state goes to closed.
in the other hand, when the cover states it is close (but it is fully open), I still can close it and it closes. When fully closed, its state goes to open.
I tried to play with the settings (Motor_reversal, and invert_cover), but no success.

Using Z2M with Home Assistant (but not through the integration), i've noticed the change occured after commit 7dc48fb6.

Best regards

### Zigbee2MQTT version

1.27.2 commit: 7dc48fb6

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20220219

### Adapter

Sonoff Zigbee 3.0 Dongle Plus

colseven, 2022-09-21:
Same BUG for me.

Koenkk, 2022-09-21:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/13813

closed 2022-09-21 by Koenkk