Koenkk/zigbee2mqtt #14124: [Feature request]: Support for Basic Cluster Attribute `0x0012: DeviceEnabled`

Atokulus, 2022-09-21:
### Is your feature request related to a problem? Please describe

Dear all

I am currently doing quite a lot of setups with the smart in-wall switch **zigfred uno** and **zigfred plus** (which I also maintain regarding its firmware a co-founder of the company behind zigfred).

These smart-light switches combine multiple in-wall actuators for lights and covers in one device in a configurable way.

Currently, if we have only a subset of actuators enabled, Zigbee2MQTT is still exposing all their endpoint's applications.
This means, that we have to manually disable certain sub-devices (entities) within Home Assistant which can get quite cumbersome.

It would be very beneficial for UX, if only enabled endpoints show up for control.

### Describe the solution you'd like

Regarding Zigbee Cluster Library, chapter 3.2.2.2.11, zigfred is supporting the optional `DeviceEnabled` boolean attribute on each of its endpoints.

What I would suggest is to let the `zigbee-herdsman-converter` check this attribute on configuration/interview in Zigbee2MQTT to hide disabled endpoints automatically. To maintain compatibility, one has to enable this check explicitly within the converter file.

To keep change on the Zigbee2MQTT low, binding and/or attribute reporting can still be configured for disabled endpoints. Regarding the savings on data traffic, it might make sense to only have them enabled iff the endpoint is actually enabled.

### Describe alternatives you've considered

There are no alternatives I considered so far.

### Additional context

[image]

Koenkk, 2022-09-21:
> Zigbee2MQTT to hide disabled endpoints automatically.

I assume you are talking about the exposes (which is used for the frontend and HA discovery). It is possible to have a dynamic exposes, example: https://github.com/Koenkk/zigbee-herdsman-converters/blob/4fa6e4d230f4ce73063eb17de3998298d5c2fc33/devices/lixee.js#L565

Atokulus, 2022-09-21:
Thanks for the example @Koenkk

I'll try to implement some dynamic expose function, which reads out the mentioned Zigbee attribute :)

Best regards
Markus

Danit2, 2022-09-22:
Hallo @Atokulus 

Ich bin ein Fan von euren Zigfred Schaltern.
Erstmal danke viel mal für dieses Produkt. 
Ich hätte da aber noch einige verbesserungen/änderungen Vorschläge.

Ausserdem bin ich auch ein Fan von Zigbee2MQTT und mache mein ganzes Haus mit Zigbee Smart. 

Ich hatte mich schon einmal per Mail im Januar bei euch gemeldet. Leider habe ich da aber nie wirklich eine Antwort erhalten. 

Hier noch einmal meine Punkte zur Software:

Grundsätzlich finde ich die Software gut und solide. Einige Features fehlen mir aber noch.

Die LED kann nicht auf Blau eingestellt werden. Auf voller Helligkeit und Blau ist die LED immer weiss. Wenn man die LED dimmt wird sie dann irgendwann Blau. Ein schönes Blau kann aber nicht eingestellt werden. 
Auch über Zigbee2MQTT kann die Farbe nicht geändert werden. Da bekomme ich eine Fehlermeldung. 
`Publish 'set' 'color' to '0xf4ce360a6f4e586e' failed: 'Error: Command 0xf4ce360a6f4e586e/242 lightingColorCtrl.moveToColor({"transtime":0,"colorx":9773,"colory":4024}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 44878 - 242 - 136 - 768 - 11 after 10000ms)'`

Ich habe das Gefühl ihr Dimmkurve des zigfred ist sehr linear. Hier währe eine logarithmische Dimmkurve viel schöner. Gerade beim Einsatz von Retrofit LED ist es sehr schwierig diese im unteren Dimmbereich noch sauber zu dimmen. 
Es geschieht sehr schnell, dass die LED einfach aus geht. Dies könnte mit einer angepassten Dimmkurve besser abgefangen werden. 

In ihrer Software fehlen mir noch einige Basic Features. z.B. Minimum Dimmerwert, maximum Dimmwert, Ein/Aus Übergangszeit usw. 
Diese Werte sollten auch über Zigbee zur Verfügung stehen. Diese drei Werte gehören zu den Basic Attributen des OnOff Cluster und sollte auch zur Verfügung stehen. 
Gerade bei Retrofit LED ist der Minimum Dimmwert sehr wichtig, da viele LED's im untersten Dimmbereich einfach ausschalten. 

Ich finde es sehr schön, dass ihr die Actions der Taster auf dem Endpoint 5 zur Verfügung stellt. 
Toll währe aber, wenn man die Taster direkt mit anderen Leuchtmitteln verbinden könnte. So wie es bei Zigbee eigentlich über das "Binding" vorgesehen währe. 
Dazu müssten aber alle vier Taster einen eigenen Endpoint haben. Toll währe, wenn man in der App die Funktion des Tasters auswählen könnte (Toggle, DimmUp, DimmDown, On, Off, DimmUp and Down usw.) und anschliessend über die Zigbee Software das "Binding" vornehmen könnte. 
Somit ist es auch möglich bei einem Ausfall des Coordinator(Gateway) die verbunden Lampen weiter zu bedienen und dies sollte ja eigentlich das Ziel sein. 
Zusätzlich die Actions noch über den Endpoint 5 zur Verfügung zu stellen finde ich trotzdem sehr hilfreich. So können zusätzliche Automatisierungen realisiert werden.

Was meinst du zu diesen Punkten?
Ist da etwas in der Art in Planung?

Vielen Dank für die Antwort und eure Arbeit am Zigfred. 

PS: Ausserdem wurde mir mal gesagt, dass die Taster noch getauscht werden können weil sie nicht dem Weiss von Feller entsprechen. Wie sieht es da aus?

Grüsse aus Thun
:smile:

closed 2022-10-31 by github-actions[bot]