Koenkk/zigbee2mqtt #14125: Jasco Products 43095 not reporting onOff state

gbourcier, 2022-09-21:
### What happened?

Jasco Products 43095 smart plug is not reporting onOff state when manually turned on or off. Manually reading value of genOnOff provide the correct result.

### What did you expect to happen?

When manually turning the switch on or off, it should report it state in Z2M.

### Zigbee2MQTT version

1.26.0

### Adapter firmware version

20210708

### Adapter

sonoff zigbee 3.0 adapter

gbourcier, 2022-09-30:
Hello, 

To add some context, when manually turning on the device, it will report the following : 
[image]

Manually reading the onOff state give the correct value, however the state is incorrectly still showing Off.

Koenkk, 2022-09-30:
Can you try to reconfigure the device via the z2m frontend -> device -> yellow refresh button, does it give you a notification "Configured successfully"?

gbourcier, 2022-09-30:
Hello,

thanks for your reply. 

The UI does says « configured successfully »

[image]

Koenkk, 2022-10-01:
can you provide the data/database.db entry of this device?

gbourcier, 2022-10-03:
Hello, 

Please find the requested information here : 

{"id":2,"type":"Router","ieeeAddr":"0x588e81fffe040afe","nwkAddr":40795,"manufId":4388,"manufName":"Jasco Products","powerSource":"Mains (single phase)","modelId":"43095","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,1794,2821],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"43095","manufacturerName":"Jasco Products","powerSource":1,"zclVersion":3,"appVersion":18,"stackVersion":101,"hwVersion":2}},"seMetering":{"attributes":{"multiplier":1,"divisor":10000,"instantaneousDemand":0,"currentSummDelivered":[0,0]}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c173ac","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c173ac","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":1794,"attrId":1024,"minRepIntval":5,"maxRepIntval":3600,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":259,"inClusterList":[0,3,2821],"outClusterList":[3,6],"clusters":{"genBasic":{"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":18,"stackVersion":101,"hwVersion":2,"zclVersion":3,"interviewCompleted":true,"meta":{"configured":114811697},"lastSeen":1664802387834,"defaultSendRequestWhen":"immediate"}

Thank you for your support!

Koenkk, 2022-10-04:
It looks that the on/off reporting has been set up correctly so it seems the device does not properly support this.

@GoldbergAlexander does this work for you?

gbourcier, 2022-10-04:
Hello!

The device is reporting properly when using another bridge with proprietary software.

Koenkk, 2022-10-04:
Could you maybe sniff the traffic when pairing it to that and triggering the state updates? https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

closed 2022-12-14 by github-actions[bot]