Koenkk/zigbee2mqtt #14128: [New device support]: Gledopto Smart Light Kit

guichd, 2022-09-21:
### Link

https://www.amazon.com/dp/B09SZ9V2TQ?ref=ppx_yo2ov_dt_b_product_details&th=1

### Database entry

{"id":35,"type":"Router","ieeeAddr":"0xa4c1380b561cdfb8","nwkAddr":29568,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-004P","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}},"genBasic":{"attributes":{"modelId":"GL-C-004P"}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":2,"dateCode":"20210916","swBuildId":"10151203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":735417325},"lastSeen":1663785937758,"defaultSendRequestWhen":"immediate"}

### Comments

I picked up a few of these cheap three button Gledopto RGB+CCT controllers and trying to add them to the supported devices but the physical buttons do not update the state. 
There is a button for on/off, one for DIM+ with long press for CCT+ and one for DIM- with long press for CCT-. 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const ota = require('zigbee-herdsman-converters/lib/ota');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['GL-C-004P'],
    model: 'GL-C-004P',
    vendor: 'Gledopto',
    description: 'Zigbee LED Strip Light Kit',
    extend: extend.light_onoff_brightness_colortemp_color(),
};

module.exports = definition;
```

### Supported color modes

color temperature, color

### Color temperature range

{"colorTempPhysicalMin":158, "colorTempPhyiscalMax":495}

guichd, 2022-09-27:
After digging into this a little bit more I was able to add reporting to the device for the buttons and it works, but I am unsure if this is the correct way to do it. Is there a way to add reporting to the converter?

[Endpoint, Cluster, Attribute]
11, OnOff, onOff
11, LevelCtrl, CurrentLevel
11, LColorCtrl, colorTemperature

I have tried variations on this but so far it is not working:

`configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);`

Koenkk, 2022-09-28:
`device.getEndpoint(1)` should be `device.getEndpoint(11)`, I've added the device now so it should work out-of-the-box.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-11-05 by github-actions[bot]