Koenkk/zigbee2mqtt #14130: Aqara fp1 reset not working anymore

MattL0, 2022-09-21:
### What happened?

Since a little while aqara fp1 reset event presence Is not working. Even when send empty string or reset string.

### Zigbee2MQTT version

Dev

### Adapter firmware version

20220528

### Adapter

2652p

### Debug log

[image]
[image]
[image]

MattL0, 2022-09-27:
Sorry, The unit was at fault

closed 2022-09-27 by MattL0

jakemauer, 2022-11-04:
@MattL0 Sorry to resurrect a closed issue but I'm having issues with my previous fully working FP1's and I'm curious to know how you determined your unit was at fault? My FP1's all decided to no longer reset when holding down the button.

MattL0, 2022-11-04:
From my experience , I think this was a power supply issue.  The unit was continually rebooting , and the internals were heating more and more because of this.

I switched the powersupply to another one, put the unit not connected for a night, and next day it was working perfectly.

jakemauer, 2022-11-05:
Thanks for the reply @MattL0. 

I figured out my issue as well. I had to listen to the coil whine of my USB brick while holding down the reset button. Once the coil whine stopped, after about 10 seconds, I let go and pressed the button a single time. That seemed to snap the unit out of it and return it to pairing mode. 

I got no visual feedback until that second single button press, the rest was listening and guessing. Posting that here for anyone else Googling like I did.

andys73, 2023-02-19:
thank you all for your comments. Managed somehow to bring my FP1 back to live using the same method like @jakemauer

MattL0, 2023-06-01:
> Thanks for the reply @MattL0.
>
> I figured out my issue as well. I had to listen to the coil whine of my USB brick while holding down the reset button. Once the coil whine stopped, after about 10 [...]

This worked for me two!!

VincentSC, 2024-12-15:
> For anyone struggling to make this work, I just ran into this problem again and had to just press the button rapidly and repeatedly like you're playing a button masher video game until the blue light would finally blink twice as if to say "OKAY OKAY".

That worked!! Here it's written out in detailed steps:
- Put the device in the power source. LED should flash once. If quickly after that, it flashes twice, then there is no problem with it.
- Keep pressing the reset-button for about 12 seconds. Between 9 en 12 seconds it should flash twice, and you hit jackpot! If not, repeat.
- Put it into pairing mode immediately after that, to make sure it stays unstuck.

After trying multiple methods I found on the internet, this is the only fix that helped unstuck both of my failing FP1's.
One I got unstuck by pressing the button after about 10 seconds, and I hit the jackpot the first try - no long press before was needed. The second I only could fix by pressing the button continuously, and it got unstuck the second try.

The devices got stuck in the first place due to updating&restarting Z2M!