Koenkk/zigbee2mqtt #14135: Interview fails after deleting and re-pairing the same Zigbee device

Element2, 2022-09-22:
Did some more testing. It happens with all previously paired/interviewed devices, regardless of type device.
 
Hi,
Zigbee2mqtt newbie her, long time Linux and home automation software user.
Hope this is in the right category.
Decided to try out Zigbee2mqtt because Deconz still does not support Lidl Silvercrest thermostat 368308_2010.

Installed Zigbee2mqtts (v1.27.2) on Ubunto v20.04 and used a hacked Lidl gateway over TCP (coordinator EZsp v8 6.7.8.0 build 373).
This all worked as expected. I can pair different supported devices, they get interviewed and can be controlled from the front end web interface.
I paired and used a few Slivercrest thermostats without problems. Then deleted one from the interface, checked the database.db and state.json files that it was gone.

Reset the thermostat to default factory settings, pulled the battery and tried to re-pair it again. Zigbee2mqtt discovered it again, but now fails on interview. Tried countless (forced) deleting and re-paring, reconfigure, rebooting server and gateway, new batteries, put the thermostat next to the gateway, etc. Outcome always the same.
Paired this thermostat with other Zigbee software and it paired and worked. 

Then tried this with another previously paired and working Silvercrest thermostat, with the same result, failed on interview.
Now I tried to pair a new, not paired with Zigbee2mqtt before Silvercrest thermostat, and it paired and interviewed with success.
So it looks like a problem specific with the repair/interview process and/or this device and this software setup.

Do I miss anything here?

Logfiles of re-pair:
ep 21 14:16:53 Bob npm[3753]: Zigbee2MQTT:info  2022-09-21 14:16:53: Device '0x0c4314fffe62fbe7' joined
Sep 21 14:16:54 Bob npm[3753]: Zigbee2MQTT:info  2022-09-21 14:16:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe62fbe7","ieee_address":"0x0c4314fffe62fbe7"},"type":"device_joined"}'
Sep 21 14:16:54 Bob npm[3753]: Zigbee2MQTT:info  2022-09-21 14:16:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x0c4314fffe62fbe7"},"type":"device_connected"}'
Sep 21 14:16:54 Bob npm[3753]: Zigbee2MQTT:info  2022-09-21 14:16:54: Starting interview of '0x0c4314fffe62fbe7'
Sep 21 14:16:54 Bob npm[3753]: Zigbee2MQTT:info  2022-09-21 14:16:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe62fbe7","ieee_address":"0x0c4314fffe62fbe7","status":"started"},"type":"device_interview"}'
Sep 21 14:16:54 Bob npm[3753]: Zigbee2MQTT:info  2022-09-21 14:16:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x0c4314fffe62fbe7"},"type":"pairing"}'
Sep 21 14:19:54 Bob npm[3753]: Zigbee2MQTT:error 2022-09-21 14:19:54: Failed to interview '0x0c4314fffe62fbe7', device has not successfully been paired
Sep 21 14:19:54 Bob npm[3753]: Zigbee2MQTT:info  2022-09-21 14:19:54: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0c4314fffe62fbe7","ieee_address":"0x0c4314fffe62fbe7","status":"failed"},"type":"device_interview"}'
Sep 21 14:19:54 Bob npm[3753]: Zigbee2MQTT:info  2022-09-21 14:19:54: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x0c4314fffe62fbe7"},"type":"pairing"}'

_Originally posted by @Element2 in https://github.com/Koenkk/zigbee2mqtt/discussions/14121_

samuelthng, 2022-09-22:
I am having the same issue with 3 different Tuya devices.

When I do try to remove, it will eventually fail without a force remove as well.
Pairing worked the first time, but after device reset, z2m sees the device and shows pairing but does not seem to fetch data from the device. Eventually pairing times out and fails with "device has not successfully been paired"

Let me know how I can help with this investigation! :)

samuelthng, 2022-09-22:
Just tested a few devices on my network.
Brand new bulbs from IKEA and Tuya pair without any issues.

Paired bulbs that were reset and/or force removed:
Devices join but unable to complete interview process.
Even the image of the devices are not shown.

Seems only Aqara devices can pair again effortlessly after a reset.

Screenshot: trying to pair an IKEA TRÅDFRI GU10 bulb after resetting it for the 5th time.
Logs do not show anything very useful, just pairing failed error message after timeout.

[image]

samuelthng, 2022-09-22:
Seems to be related: https://github.com/Koenkk/zigbee2mqtt/issues/13964

Also EZSP v6.7.*

Koenkk, 2022-09-22:
Please test with the latest dev branch, @kirovilya added an improvement for pairing with the EZSP adapter (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Element2, 2022-09-22:
Thanks, let's see what that brings.

Element2, 2022-09-22:
Tried to pair two different re branded Tuya devices. Those two were paired and working before and were deleted.
When I tried to pair them with latest dev just now, nothing happened. No popups in front end, nothing in logs.

samuelthng, 2022-09-23:
Can confirm same behaviour for me - on master it shows up but can't pair. On dev doesn't show up.

TTR6214, 2022-09-25:
I think I have the same problem.
I sucessfully paired brand new MS-108ZR out of the box (so Tuya device).
Then I force-removed this device and try to pair it again.
Zigbee2mqtt cannot recognize the manufacturer, model etc ... and cannot finish the pairing process.

I also use EZSP 6.7.8.0 (Hacked silvercrest gateway).

t112013, 2022-09-26:
I have experienced a similar issue and i almost becase homeless after a mad wife had no lights :P 
Well, it all started when i lost one Elko thermostat (connected but no state change,nothing), i tried repairing it and it never worked,  it never came any message, almost like the device never talked to the coordinator.
So tried another dongle, i have also the sonoff dongle (the new version). and it didnt work.  Then i made a new network. tried repairing all devices for the 15th time and this time some remotes and motion sensor where never coming on. Tried then ZHA and there they worked almost on the second so i though it had to be something with ZigbeetoMqtt. Then i even reinstalled home assistant to try a blank installation (though it had something to do with HA) .
I also tried now the edge branch where it actually works, but sometimes its not very responsive to pair, i have stopped and stated the add on many times, some times mqtt broker becomes unresponsive too, i have one ikea motion sensor that was very difficult to pair, i even changed batties but, sometimes its not repporting the status, it works again when i restart the whole box.
I have tried adding a develco smoke sensor, which in the DB says its supported but it also fails to interview. And a few Ikea lightbulbs that dont even show up when pairing.
Also one Namrom 4 buttons remote never shows up, but when i factory reset the remote it show device left the network ( i have two more of those, both paired with no issue) 

Iam bit lost but i can patiantly wait too.
If i can somehow help, please let me know.

Btw, if it helps. for those that have no response when pairing in the edge version, remember to remove the offical add on, eventually restart the add on a few times and pair a bulb or something on mains before adding end-devices
Cheers

samuelthng, 2022-09-26:
Hello, wanted to check if anyone else with pairing issues faced this:
- Repeated trustCenterJoin frame
- No action taken by z2m after

Not sure if it's related to the pairing issue but wanted to get some opinions and eyes on this.
https://github.com/Koenkk/zigbee-herdsman/blame/85c277f82204be6d7419df396896645da549def6/src/adapter/ezsp/driver/driver.ts#L298

```
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,6,0): 9600010500
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:uart --> [66d421a9512ab85c7e]
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart <-- [67d4a1a9512a20c47e]
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,7,0): 67d4a1a9512a20c47e
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 9680010500
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false}
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:uart <-- [77d4b1a9702a9900fa7d387d38a5126d53ed9d4e27ab59f77e]
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,7,0): 77d4b1a9702a9900fa1818a5126d53ed9d4e27ab59f77e0000
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (0)
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:uart --> [8070787e]
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 96900124008cb2a38c5280b838c1a401000000b439
2022-09-26T15:06:02.610Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_id_":36,"_cls_":"trustCenterJoinHandler","_isRequest_":false,"newNodeId":45708,"newNodeEui64":[164,193,56,184,128,82,140,163],"status":1,"policyDecision":0,"parentOfNewNodeId":0}
2022-09-26T15:06:06.196Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2022-09-26T15:06:06.196Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2022-09-26T15:06:06.197Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2022-09-26T15:06:06.197Z zigbee-herdsman:adapter:ezsp:uart --> DATA (7,0,0): 9700010500
2022-09-26T15:06:06.197Z zigbee-herdsman:adapter:ezsp:uart --> [70d521a9512a85687e]
2022-09-26T15:06:06.197Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (0)
2022-09-26T15:06:06.201Z zigbee-herdsman:adapter:ezsp:uart <-- [00d5a1a9512a1dcc7e]
2022-09-26T15:06:06.201Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (0,0,0): 00d5a1a9512a1dcc7e
2022-09-26T15:06:06.201Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (1)
2022-09-26T15:06:06.201Z zigbee-herdsman:adapter:ezsp:uart --> [8160597e]
2022-09-26T15:06:06.202Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 9780010500
2022-09-26T15:06:06.202Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false}
2022-09-26T15:06:06.202Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (0) success
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:uart <-- [10d5b1a9702a9900fa7d387d38a5126d53ed9d4e27ab643a7e]
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (1,0,0): 10d5b1a9702a9900fa1818a5126d53ed9d4e27ab643a7e0000
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 97900124008cb2a38c5280b838c1a40100000089f4
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_id_":36,"_cls_":"trustCenterJoinHandler","_isRequest_":false,"newNodeId":45708,"newNodeEui64":[164,193,56,184,128,82,140,163],"status":1,"policyDecision":0,"parentOfNewNodeId":0}2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:ezsp Time to watchdog ... 0
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:ezsp ==> nop: null
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:ezsp ==> {"_cls_":"nop","_id_":5,"_isRequest_":true}
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:uart --> DATA (6,6,0): 9600010500
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:uart --> [66d421a9512ab85c7e]
2022-09-26T15:05:56.197Z zigbee-herdsman:adapter:ezsp:uart -?- waiting (7)
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart <-- [67d4a1a9512a20c47e]
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (6,7,0): 67d4a1a9512a20c47e
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (7)
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart --> [87009f7e]
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 9680010500
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x5: {"_id_":5,"_cls_":"nop","_isRequest_":false}
2022-09-26T15:05:56.202Z zigbee-herdsman:adapter:ezsp:uart -+- waiting (7) success
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:uart <-- [77d4b1a9702a9900fa7d387d38a5126d53ed9d4e27ab59f77e]
2022-09-26T15:06:02.609Z zigbee-herdsman:adapter:ezsp:uart <-- DATA (7,7,0): 77d4b1a9702a9900fa1818a5126d53ed9d4e27ab59f77e0000
[... cut ...]
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:uart --> ACK  (2)
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:uart --> [82503a7e]
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:ezsp <== Frame: 97900124008cb2a38c5280b838c1a40100000089f4
2022-09-26T15:06:10.177Z zigbee-herdsman:adapter:ezsp:ezsp <== 0x24: {"_id_":36,"_cls_":"trustCenterJoinHandler","_isRequest_":false,"newNodeId":45708,"newNodeEui64":[164,193,56,184,128,82,140,163],"status":1,"policyDecision":0,"parentOfNewNodeId":0}
```

y8s, 2022-09-28:
I recently added a Sonoff ZDongle-E (using `adapter: ezsp`)and was doing fine but wanted to add some range and picked up a couple Sonoff S40LiteZB and they both paired fine at first but upon re pair of one, it fails interview and sits in the device list as unsupported.

I did try the docker `latest-dev` tag today but then z2m wouldn't see it at all.

Current non dev setup:
Zigbee2MQTT version
1.27.2 commit: 7dc48fb
Coordinator type
EZSP v8
Coordinator revision
6.10.3.0 build 297

FabiJa, 2022-09-28:
I am having the exact same problem after deleting a device. Also, with the current dev version, the device is not found at all.

I have a smabit Bitron USB dongle (EM3587) flashed to 6.7.10.

Element2, 2022-09-29:
The latest dev modifications (Update Herdsman to 0.14.61) fixed the problem for me. The dev now pairs and interviews successful.
And after a forced remove again pairs and interviews successful.

y8s, 2022-09-29:
I will try to get a herdsman log on the docker latest-dev sometime today.

Element2, 2022-09-30:
Closed as this is fixed in latest dev with Herdsman 0.14.61.
See also #13964

closed 2022-09-30 by Element2

sdalu, 2022-12-30:
It seems I have the same problem, though it supposed to be fixed.
Device: FLSZB-110
zigbee-herdsman: 0.14.76
zigbee2mqtt: 1.28.4
adapter: sonoff with zstack CC1352P2_CC2652P_launchpad_coordinator_20220219

Device has been force removed, but "left the network" notification only appearing when trying to join
~~~
Zigbee2MQTT:warn  2022-12-30 14:15:31: Device '0x0015bc00330069cd' left the network
Zigbee2MQTT:info  2022-12-30 14:15:31: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00330069cd","ieee_address":"0x0015bc00330069cd"},"type":"device_leave"}'
Zigbee2MQTT:info  2022-12-30 14:15:41: Device '0x0015bc00330069cd' joined
Zigbee2MQTT:info  2022-12-30 14:15:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00330069cd","ieee_address":"0x0015bc00330069cd"},"type":"device_joined"}'
Zigbee2MQTT:info  2022-12-30 14:15:41: MQTT publish: topic 'zigbee2mqtt/0x0015bc00330069cd', payload '{"last_seen":1672406141098,"linkquality":193}'
Zigbee2MQTT:info  2022-12-30 14:15:41: Starting interview of '0x0015bc00330069cd'
Zigbee2MQTT:info  2022-12-30 14:15:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00330069cd","ieee_address":"0x0015bc00330069cd","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-12-30 14:15:41: MQTT publish: topic 'zigbee2mqtt/0x0015bc00330069cd', payload '{"last_seen":1672406141483,"linkquality":193}'
Zigbee2MQTT:info  2022-12-30 14:15:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00330069cd","ieee_address":"0x0015bc00330069cd"},"type":"device_announce"}'
Zigbee2MQTT:info  2022-12-30 14:15:45: MQTT publish: topic 'zigbee2mqtt/0x0015bc00330069cd', payload '{"last_seen":1672406145105,"linkquality":102}'
Zigbee2MQTT:error 2022-12-30 14:15:51: Failed to interview '0x0015bc00330069cd', device has not successfully been paired
Zigbee2MQTT:info  2022-12-30 14:15:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x0015bc00330069cd","ieee_address":"0x0015bc00330069cd","status":"failed"},"type":"device_interview"}'
Zigbee2MQTT:info  2022-12-30 14:15:52: MQTT publish: topic 'zigbee2mqtt/0x0015bc00330069cd', payload '{"battery_low":false,"last_seen":1672406152749,"linkquality":98,"tamper":false,"water_leak":false}'
Zigbee2MQTT:info  2022-12-30 14:15:54: MQTT publish: topic 'zigbee2mqtt/0x0015bc00330069cd', payload '{"battery_low":false,"last_seen":1672406154965,"linkquality":102,"tamper":false,"water_leak":false}'
~~~

carbogit, 2023-04-01:
just had similar troubles with Ikea Styrbar and Sonoff Dongle-E. Manual force-remove followed by stopping and starting zigbee2mqtt solved it. 

Zigbee2MQTT:info  2023-04-01 16:56:16: Starting Zigbee2MQTT version 1.30.2 (commit #cdf62ea)
Zigbee2MQTT:info  2023-04-01 16:56:16: Starting zigbee-herdsman (0.14.96)

Zigbee2MQTT:info  2023-04-01 16:56:43: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2023-04-01 16:56:43: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"a21qr-3"}'
Zigbee2MQTT:warn  2023-04-01 16:56:48: Device 'styrbar' left the network
Zigbee2MQTT:info  2023-04-01 16:56:48: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"styrbar","ieee_address":"XXXXXXXXXXXXX"},"type":"device_leave"}'

Zigbee2MQTT:info  2023-04-01 16:56:59: Device 'styrbar' joined
Zigbee2MQTT:info  2023-04-01 16:56:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"styrbar","ieee_address":"XXXXXXXXXXXXX"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-04-01 16:56:59: Starting interview of 'styrbar'
Zigbee2MQTT:info  2023-04-01 16:56:59: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"styrbar","ieee_address":"XXXXXXXXXXXXX","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:error 2023-04-01 16:57:19: Failed to interview 'styrbar', device has not successfully been paired
Zigbee2MQTT:info  2023-04-01 16:57:19: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"styrbar","ieee_address":"XXXXXXXXXXXXX","status":"failed"},"type":"device_interview"}'

sveinse, 2023-12-26:
Running 1.34.0 and I observe the same problems repairing to IKEA devices all the time (and many times in a row). @carbogit fix by force removing the device and the restarting z2m seems to work (or I've been incredibly lucky every time).

bassho, 2023-12-28:
Same issue.
Failed to interview '', device has not successfully been paired

richimaint, 2023-12-29:
Same issue.
Failed to interview '', device has not successfully been paired

theimo1221, 2024-01-06:
Had it 4 times today with devices from different Brands (illuminize and Ubisys) running 1.35

richimaint, 2024-01-06:
I installed a new battery and was able to pair it successfully.

inteller, 2024-01-07:
Same issue here (SLZB-06, different devices from IKEA and SunRicher; all versions default up-to-date HA).

sveinse, 2024-01-08:
Perhaps a new issue should be opened with reference to this? I'm not sure beating an old closed issue will be productive.

inteller, 2024-01-08:
> Same issue here (SLZB-06, different devices from IKEA and SunRicher; all versions default up-to-date HA). 

FWIW, I have been able to resolve the issue by changing the Zigbee channel and *deleting /config/zigbee2mqtt/database.db* 

Maybe this helps

theimo1221, 2024-01-08:
> > Same issue here (SLZB-06, different devices from IKEA and SunRicher; all versions default up-to-date HA).
> 
> FWIW, I have been able to resolve the issue by changing the Zigbee channel and _deleting /config/zigbee2mqtt/database.db_
> 
> Maybe this helps

Sorry but this drastic measure shouldn't have to be necessary, as repairing 100/200+ devices is very time consuming....

TonKro, 2024-01-13:
Having the same problem:

Error 2024-01-13 10:36:38Failed to interview '0x5cc7c1fffeee996e', device has not successfully been paired
Info 2024-01-13 10:36:39MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x5cc7c1fffeee996e","ieee_address":"0x5cc7c1fffeee996e","status":"failed"},"type":"device_interview"}'
Info 2024-01-13 10:36:39MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x5cc7c1fffeee996e"},"type":"pairing"}'
Debug 2024-01-13 10:37:23Received Zigbee message from '0x5cc7c1fffeee996e', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":33816581,"imageType":4555,"manufacturerCode":4476}' from endpoint 1 with groupID null
Debug 2024-01-13 10:37:26Received Zigbee message from '0x5cc7c1fffeee996e', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":33816581,"imageType":4555,"manufacturerCode":4476}' from endpoint 1 with groupID null
Debug 2024-01-13 10:37:29Received Zigbee message from '0x5cc7c1fffeee996e', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":33816581,"imageType":4555,"manufacturerCode":4476}' from endpoint 1 with groupID null
Debug 2024-01-13 10:37:32Received Zigbee message from '0x5cc7c1fffeee996e', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":33816581,"imageType":4555,"manufacturerCode":4476}' from endpoint 1 with groupID null

itainoam, 2024-01-24:
Force removing the device and the restarting z2m container seems to work for me as well.

chrzanu180, 2024-01-29:
I tried all above solutions, but only reflashing of my zigbee dongle helped with the issue and all seems to be working fine now

richimaint, 2024-01-29:
Which dongle do you have?

richimaint

lfilho, 2024-02-14:
Just happened to me with a [Enbrighten outdoor switch](https://www.zigbee2mqtt.io/devices/43100.html) too. Force removing + restarting z2m addon on home assistant worked for me.

drublin, 2024-03-04:
what worked was updating my SONOFF 3.0 P dongle to 20230507 firmware (from 20210708), remove devices and finally pair again successfully.

SONOFF 3.0 CC2652P update steps here: https://forum.iobroker.net/post/535605
Most of youtube guides are obsolete because Texas Instruments flash tool is EOS.